G6EJD/ESP32-e-Paper-Weather-Display #2: error: stray '\302' in program

schroepfer, 2018-05-27:
Hello,
compiling your code gave the following errors.
Pleas can you help me?
BR
Juergen

In file included from /Users/Juergen-S/Desktop/ESP32-e-Paper-Weather-Display/ESP32-e-Paper-Weather-Display.ino:20:0:
credentials.h:13: error: stray '\302' in program
 const char * host    = "api.wunderground.com";
 ^
credentials.h:13: error: stray '\240' in program
credentials.h:13: error: stray '\302' in program
credentials.h:13: error: stray '\240' in program
credentials.h:13: error: stray '\302' in program
credentials.h:13: error: stray '\240' in program
credentials.h:14: error: stray '\302' in program
 String      City    = "Kassel";             // Your home city
 ^
credentials.h:14: error: stray '\240' in program
credentials.h:14: error: stray '\302' in program
credentials.h:14: error: stray '\240' in program
credentials.h:14: error: stray '\302' in program
...
...
..

G6EJD, 2018-05-27:
How have you downloaded the code? Ideally get the zip version, it looks to me that these are copy/paste errors. Usually if you retype the errant lines it fixes the problem. Sometimes selecting the text and then pasting it into say word or an email body as text only removes these unseen escape characters.

schroepfer, 2018-05-27:
I downloadet your zip.
I folloed exactly your readme.
I also retyped this lines...

G6EJD, 2018-05-27:
I just download the latest 4.2" version and it compiles correctly. What your getting is hidden characters. See here for an example: https://arduino.stackexchange.com/questions/30581/stray-302-in-program-error-when-compiling

1. Download the zip.
2. Uncompress
3. Copy the files to your IDE location.

4. Open the Sketch with the IDE

G6EJD, 2018-05-27:
usually it will be the " symbol that need to be deleted and replaced with a simple quote.
e.g. 
const char * host    = api.wunderground.com;

Then:
const char * host    = "api.wunderground.com";

schroepfer, 2018-05-27:
The credentials.h in your ZIP doesn't work!
I copy the raw-text from the source on github and it works now!

kaylahkramer, 2020-06-01:
#include <CapacitiveSensor.h>

CapacitiveSensor capSensor = CapacitiveSensor(4,2);// 1M resistor between pins 4 & 2, pin 2 is sensor pin, add a wire and or foil

int   threshold = 100;//this is an important number which is a preset capacitance reading. Check the serial monitor when touching the foil or hovering the hand over the foil. You need to determine what is the ACTUAL number the sensor is reading, then change this threshold to that number in order to make the sensor to work appropriately.

const int ledPin = 12;

//define notes' frequency
const int c = 261;//C4
const int d = 294;//D4
const int e = 329;//E4
const int f = 349;//F4
const int g = 391;//G4
const int gS = 415;//G4#
const int a = 440;//A4
const int aS = 455;//A4#
const int b = 466;//B4
const int cH = 523;//C5
const int cSH = 554;//C5#
const int dH = 587;//D5
const int dSH = 622;//D5#
const int eH = 659;//E5
const int fH = 698;//F5
const int fSH = 740;//F5#
const int gH = 784;//G5
const int gSH = 830;//G5#
const int aH = 880;//A5

//digital pin 8 is the output pin 
const int buzzerPin = 8;

//define note lengths
int Q = 500;
int H = 2*Q;
int E = 0.5*Q;
int S = 0.25*Q;
int ED = 0.75*Q;

void setup()                    

{

   Serial.begin(9600);

   pinMode(ledPin,OUTPUT);

   //Setup pin modes
  pinMode(buzzerPin, OUTPUT); 

}

void loop()  //create a variable type long to hold the sensor's value.                  

{

  delay(1+H);//additional rest between repeats
  
 long sensorValue = capSensor.capacitiveSensor(30); //30 samples

 Serial.println(sensorValue); // print sensor output 

   if(sensorValue > threshold)

   {

    digitalWrite(ledPin,HIGH);
    //duration is defined as quarter note 500, delay at least 1ms for each note so you can hear it
  tone(buzzerPin, a, Q);
  delay(1+Q);
  tone(buzzerPin, a, Q); 
  delay(1+Q);   
  tone(buzzerPin, a, Q);
  delay(1+Q);
  tone(buzzerPin, f, ED);
  delay(1+ED);
  tone(buzzerPin, cH, S); 
  delay(1+S); 
  tone(buzzerPin, a, Q);
  delay(1+Q);
  tone(buzzerPin, f, ED);
  delay(1+ED);
  tone(buzzerPin, cH, S);
  delay(1+S);
  tone(buzzerPin, a, H);
  delay(1+H);
  
  tone(buzzerPin, eH, Q);
  delay(1+Q);
  tone(buzzerPin, eH, Q);
  delay(1+Q);
  tone(buzzerPin, eH, Q);  
  delay(1+Q);
  tone(buzzerPin, fH, ED);
  delay(1+ED);
  tone(buzzerPin, cH, S);
  delay(1+S);
  tone(buzzerPin, gS, Q);
  delay(1+Q);
  tone(buzzerPin, f, ED);
  delay(1+ED);
  tone(buzzerPin, cH, S);
  delay(1+S);
  tone(buzzerPin, a, H);
  delay(1+H);

  delay(1+H);//additional rest between repeats

   }

   else{

    digitalWrite(ledPin,LOW);

   }  

   delay(10);

 
}

kaylahkramer, 2020-06-01:
this isn't working do you know why?

G6EJD, 2020-06-01:
You aren’t referring g to my code, but your error is due to The use of copy and paste code from usually a web page. You have ASCII codes copied that add spaces, etc to your code. Go through each identified line and remove any extra spaces at the beginning and end of any identified line. Then, (Arduino IDE) go to TOOLS, Auto Format.