G6EJD/ESP32-e-Paper-Weather-Display #4: deep sleep

schroepfer, 2018-05-27:
'esp_sleep_enable_timer_wakeup' was not declared in this scope:

_/Users/Juergen-S/Desktop/ESP32-e-Paper-Weather-Display/ESP32-e-Paper-Weather-Display.ino: In function 'void setup()':
ESP32-e-Paper-Weather-Display:127: error: 'esp_sleep_enable_timer_wakeup' was not declared in this scope
   esp_sleep_enable_timer_wakeup(UpdateInterval);_

G6EJD, 2018-05-27:
You have to be using IDE 1.8.5 and ESP Core 2.4.1

closed 2018-05-27 by G6EJD

schroepfer, 2018-06-15:
Hello,
(i use IDE 1.8.5 and ESP Core 2.4.1 and Board Wemos Lolin32)

after a break i want to compile your new code: "ESP32_OWM_Current_Forecast_75_epaper_v3.ino"
and i get back this error:

_exit status 1
'esp_sleep_enable_timer_wakeup' was not declared in this scope_

schroepfer, 2018-06-15:
It's ok now! 
I  completely erased my Arduino and installed it new...