G6EJD/ESP32-e-Paper-Weather-Display #5: stuck on "Creating object..."

schroepfer, 2018-05-27:
...the display shows only "Getting data from Weather Onderground". Nothing more...

_.....WiFi connected at: 1286777024
17:04:19
Connecting to api.wunderground.com
Requesting URL: http://api.wunderground.com/api/b3d60ab21601c97e/conditions/q/DE/Kassel.json
Receiving API weather data.................
Closing connection
Bytes received: 981
Creating object...
Connecting to api.wunderground.com
Requesting URL: http://api.wunderground.com/api/b3d60ab21601c97e/forecast/astronomy/q/DE/Kassel.json
Receiving API weather data.................
Closing connection
Bytes received: 1000
Creating object..._

G6EJD, 2018-05-27:
This is because WU do not provide astronomy data for every location in the world, you will need to find a location in your vicinity maybe continent where they do provide moon rise/set times. So-far few locations are supported in the Sothern hemisphere, ask WU why? 

Please test the code first using my locations to validate that the code is working as expected, then it is between you and WU to resolve.

There are no code issues.

closed 2018-05-27 by G6EJD

schroepfer, 2018-05-27:
It doesn't work with your original locations.

const char* host    = "api.wunderground.com";
String      City    = "Melksham";             // Your home city
String      Country = "UK";                   // Your country

G6EJD, 2018-05-27:
http://api.wunderground.com/api/b3d60ab21601c97e/astronomy/q/UK/Melksham.json 

Click on the link to see it's working, then repeat with your location:

http://api.wunderground.com/api/b3d60ab21601c97e/astronomy/q/DE/Kassel.json

G6EJD, 2018-05-27:
It's because there are multiple locations with the name Kassel !

You have to use the list of registered station names by doing this as in the README:
4.Adjust your City name, Country and Language and add your free WU Key.   Find your country here: https://www.wunderground.com/weather-by-country.asp Click on your Contry to find ytour nearest City/Location that is supported by WU. In the USA use your State as Country.

G6EJD, 2018-05-27:
My displays are all running here, the problem is because the astronomy data is missing for Kessel try clicking on the links above for Melksham and kessel to see the difference in the response

G6EJD, 2018-05-27:
I’ve looked at the api documentation and it does not say how to resolve multiple locations like yours the return shows 4 locations with the same name. 

Which version are you using and I’ll try to resolve it 2.9” or 4.2” ?

schroepfer, 2018-05-27:
2.9‘‘. I also test with 7.5‘‘

G6EJD, 2018-05-27:
I have the 2.9" running here with:
// Set your location according to WU locations
String City       = "Calden";                     // Your home city
String Country    = "DE";                          // Your country  

The problem with Kassel is there are multiple location responses for both the forecast and Astronomy calls.

G6EJD, 2018-05-27:
// Set your location according to WU locations
String City       = "Vellmar";                     // Your home city
String Country    = "DE";                          // Your country 

Is working that's close to Kassel

G6EJD, 2018-05-27:
http://api.wunderground.com/api/b3d60ab21601c97e/forecast/astronomy/q/autoip.json

This call detects your location

G6EJD, 2018-05-27:
http://api.wunderground.com/api/b3d60ab21601c97e/forecast/astronomy/q/51.3,9.5.json
http://api.wunderground.com/api/b3d60ab21601c97e/forecast/astronomy/q/lat,lon.json

Works, but the results are all imperial so °F etc !

G6EJD, 2018-05-27:
Vellmar seems your best match

schroepfer, 2018-05-28:
(Das Mistding will nicht!)
Tested it with:
_const char* host    = "api.wunderground.com";
String      City    = "Vellmar";             // Your home city
String      Country = "DE";                   // Your country_  

The ESP stuck in "Creating object..."
_...WiFi connected at: 1728620736
09:40:08
Connecting to api.wunderground.com
Requesting URL: http://api.wunderground.com/api/b3d60ab21601c97e/conditions/q/DE/Vellmar.json
Receiving API weather data.................
Closing connection
Bytes received: 2564
Creating object...
Connecting to api.wunderground.com
Requesting URL: http://api.wunderground.com/api/b3d60ab21601c97e/forecast/astronomy/q/DE/Vellmar.json
Receiving API weather data.................
Closing connection
Bytes received: 7589
Creating object..._

[https://photos.app.goo.gl/DDEJKiTy8cHvssSr2](url)

G6EJD, 2018-05-28:
The response sizes are correct and it's working on my 2.9" display OK. So what's different?

Are you using IDE 1.8.5?
Is your ESP Core Version 2.4.1 ? Tools, Boards, Board Manager, then type in ESP to check

G6EJD, 2018-05-28:
I checked the version 5a last night.

schroepfer, 2018-05-28:
Yes and yes.
Now i compiled your ESP32_OWM_Current_Forecast_29_epaper_v5.ino.
And it works with OWM on my 7.5'' Display!! WU not. I don't know why.

G6EJD, 2018-05-28:
I've given up with WU, and OWM works OK. I'm just updating the 4.2 version to get all 5-days forecast data for what it's worth.

Good to hear it's working.

G6EJD, 2018-05-28:
I've just uploaded a v6 of the OWM, now it receives all 40 periods from OWM and I made some overall improvements. Only the .ino has changed.

schroepfer, 2018-05-29:
Thank you, but your Version 6 doesn't work on my board

.......WiFi connected at: 1728620736
08:11:20
Received OK...
Creating object...and Decoding weather data
Lon:
Lat:
Guru Meditation Error: Core  1 panic'ed (LoadProhibited)
. Exception was unhandled.
Register dump:
PC      : 0x400013f9  PS      : 0x00060630  A0      : 0x800dbd0e  A1      : 0x3ffdb910  
A2      : 0x3ffd3808  A3      : 0x00000000  A4      : 0x000000ff  A5      : 0x0000ff00  
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x00000000  A9      : 0x3ffdb8c0  
A10     : 0x3ffd3808  A11     : 0x00000000  A12     : 0x3ffe59a0  A13     : 0x00000000  
A14     : 0x00000000  A15     : 0xff000000  SAR     : 0x00000019  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x400013f9  LEND    : 0x4000140d  LCOUNT  : 0xffffffff  

Backtrace: 0x400013f9:0x3ffdb910 0x400dbd0b:0x3ffdb920 0x400dbd76:0x3ffdb940 0x400d58d5:0x3ffdb960 0x400d880b:0x3ffdb9d0 0x4011a6db:0x3ffdba10

Rebooting...
ets Jun  8 2016 00:22:57

G6EJD, 2018-05-29:
On the face of the stack dump, the problem occurred at line 1642:
      if (spacing < yticks) gfx.drawHorizontalLine((x_pos + 1 + j * gwidth / number_of_dashes), y_pos + (gheight * spacing / yticks), gwidth / (2 * number_of_dashes));
 
And as it's an exception unhandled this suggests a divide by 0 error, maybe yticks has been set to 0? or number of dashes to 0 ?

The programme works OK for me, I just checked it again.

Please click on the weather link in the owm_credentials tab and see what the OWM Server responds with, there should be a lat, lon for your location the above trace indicates they were both blank. Maybe copy the json response for weather here please.

schroepfer, 2018-05-29:
I changed nothing in your code, only  for my "EPD_WaveShare_75.h",
EPD_WaveShare75 epd(5, 16, 17, 4);, and SCREEN_WIDTH  384, SCREEN_HEIGHT
640.

{"cod":"200","message":0.0028,"cnt":2,"list":[{"dt":1527584400,"main":{"temp":18.69,"temp_min":18.26,"temp_max":18.69,"pressure":1020.43,"sea_level":1031.38,"grnd_level":1020.43,"humidity":83,"temp_kf":0.43},"weather":[{"id":801,"main":"Clouds","description":"few
clouds","icon":"02d"}],"clouds":{"all":24},"wind":{"speed":6.91,"deg":41.5067},"sys":{"pod":"d"},"dt_txt":"2018-05-29
09:00:00"},{"dt":1527595200,"main":{"temp":21.43,"temp_min":21.11,"temp_max":21.43,"pressure":1018.82,"sea_level":1029.59,"grnd_level":1018.82,"humidity":80,"temp_kf":0.32},"weather":[{"id":801,"main":"Clouds","description":"few
clouds","icon":"02d"}],"clouds":{"all":24},"wind":{"speed":7.08,"deg":46.0002},"sys":{"pod":"d"},"dt_txt":"2018-05-29
12:00:00"}],"city":{"id":2642786,"name":"Melksham","coord":{"lat":51.3728,"lon":-2.1401},"country":"GB","population":14692}}

G6EJD, 2018-05-29:
I can get a similar result if I disconnect my display, is the display/pins working OK with another code?

G6EJD, 2018-05-29:
Does it work on a 4.2" display. I've just been looking at the EPD display driver and for a 7.5" display it allocates 640/8*384 bytes= 31KB whereas the 4.2" variant requires 15KB, so I think your ESP32 has run out of memory with the 7.5" display. you could try reducing the number of periods from OWM from 40 to 20 and reducing the memory allocation from :
DynamicJsonBuffer jsonBuffer (50 * 1024);

to 

DynamicJsonBuffer jsonBuffer (35 * 1024);

On line 481this will give you a bit more memory.

schroepfer, 2018-05-29:
Yes, it works with the code from yesterday.
Compiling the unchanged code: ESP32_OWM_Current_Forecast_42_epaper_v6.ino
crashes.

G6EJD, 2018-05-29:
Try reducing the memory requirements down with that line 50*1024 to 35*1024, I've just checked the JSON requirement and it's 31K, so 35K is a minimum.

Today I'll look at how some memory can be saved.

G6EJD, 2018-05-29:
This is a problem with larger displays.

G6EJD, 2018-05-29:
There is now a version 7, I have not tested the graph display yet, but it uses 32K less RAM than the full 5-day/3-hr version, so this one is 3-day/3-hrs.
I think it should work OK.

schroepfer, 2018-05-29:
V7:
on my MAC i get now a compiler-error:

/Users/Juergen-S/Documents/Arduino/ESP32-e-Paper-Weather-OWM_V7/ESP32-e-Paper-Weather-OWM_V7.ino: In function 'void setup()':
ESP32-e-Paper-Weather-OWM_V7:121: error: 'esp_sleep_enable_timer_wakeup' was not declared in this scope
   esp_sleep_enable_timer_wakeup(UpdateInterval);

G6EJD, 2018-05-29:
I've download v7 and it all compiles OK. Not sure why you get that error, Usually a typo or added character or hidden characters,