G6EJD/ESP32-e-Paper-Weather-Display #9: error in compile

lrtal, 2018-06-24:
hello
i followed your instructions.
after compiling ..
Error compiling for board ESP32 Dev Module

G6EJD, 2018-06-24:
How can I help if you provide no information, like what is the error? You might have the wrong version of the IDE (not the latest) you might have the wrong version of the ESP Core (not the latest). Please copy the compilation error message and post here so I can see it.

lrtal, 2018-06-24:
Arduino: 1.8.5 (Windows 10), Board: "ESP32 Dev Module, Default, QIO, 80MHz, 4MB (32Mb), 115200, None"

C:\Users\liort\OneDrive\Documents\Arduino\sketch_weather_display_29\sketch_weather_display_29.ino:22:73: fatal error: ArduinoJson.h: No such file or directory

compilation terminated.

exit status 1
Error compiling for board ESP32 Dev Module.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

wilhelmzeuschner, 2018-06-24:
You need to install the ArduinoJSON library

G6EJD, 2018-06-24:
You need v 5.13.2

closed 2018-06-24 by G6EJD

lrtal, 2018-06-24:
thank you.  installed.
now new errors..

----------------------------------------------------------------------------------------------
Arduino: 1.8.5 (Windows 10), Board: "ESP32 Dev Module, Default, QIO, 80MHz, 4MB (32Mb), 115200, None"

In file included from C:\Users\liort\OneDrive\Documents\Arduino\sketch_weather_display_29\sketch_weather_display_29.ino:20:0:

moon_icons.h:30: error: stray '\302' in program

   <title>ESP32-e-Paper-Weather-Display/moon_icons.h at master ֲ· G6EJD/ESP32-e-Paper-Weather-Display</title>

   ^

moon_icons.h:30: error: stray '\267' in program

In file included from C:\Users\liort\OneDrive\Documents\Arduino\sketch_weather_display_29\sketch_weather_display_29.ino:20:0:

moon_icons.h:143: error: missing terminating ' character

     </option></form><form class="js-site-search-form" data-scope-type="Repository" data-scope-id="129303357" data-scoped-search-url="/G6EJD/ESP32-e-Paper-Weather-Display/search" data-unscoped-search-url="/search" action="/G6EJD/ESP32-e-Paper-Weather-Display/search" accept-charset="UTF-8" method="get"><input name="utf8" type="hidden" value="&#x2713;" />

     ^

In file included from C:\Users\liort\OneDrive\Documents\Arduino\sketch_weather_display_29\sketch_weather_display_29.ino:20:0:

moon_icons.h:180: error: stray '\342' in program

                       <span aria-hidden="true" class="d-inline-block ml-1 v-align-middle">ג†µ</span>

                       ^

moon_icons.h:180: error: stray '\206' in program

moon_icons.h:180: error: stray '\265' in program

moon_icons.h:185: error: stray '\342' in program

                       <span class="d-inline-block ml-1 v-align-middle">ג†µ</span>

                       ^

moon_icons.h:185: error: stray '\206' in program

moon_icons.h:185: error: stray '\265' in program

In file included from C:\Users\liort\OneDrive\Documents\Arduino\sketch_weather_display_29\sketch_weather_display_29.ino:20:0:

moon_icons.h:320: error: missing terminating ' character

         <li></option></form><form class="logout-form" action="/logout" accept-charset="UTF-8" method="post"><input name="utf8" type="hidden" value="&#x2713;" /><input type="hidden" name="authenticity_token" value="GnLnnZU/kl7vy5yVT1MnvPE9uZFqvvaaImUsXGmDRk92Pu80q5mKU3djW9KI6TT6uuXtF3yrPok462A6olYDbA==" />

         ^

moon_icons.h:332: error: missing terminating ' character

         </option></form><form class="sr-only right-0" action="/logout" accept-charset="UTF-8" method="post"><input name="utf8" type="hidden" value="&#x2713;" /><input type="hidden" name="authenticity_token" value="xv5rtCsHeM1P7NJhhfex+QRh7sW1ToXt/P1M0W+kJmiqsmMdFaFgwNdEFSZCTaK/T7m6Q6NbTf7mcwC3pHFjSw==" />

         ^

moon_icons.h:368: error: missing terminating ' character

         </option></form><form data-autosubmit="true" data-remote="true" class="js-social-container" action="/notifications/subscribe" accept-charset="UTF-8" method="post"><input name="utf8" type="hidden" value="&#x2713;" /><input type="hidden" name="authenticity_token" value="L0TlKG4hYQvYxkwMbcEJYcMxeyzY3Y/BzQqSowdxT/v0dG+fqEdP3SxGfvR2tbx8Z/QdgtTqPXtLG5rY32UFcA==" />      <input type="hidden" name="repository_id" id="repository_id" value="129303357" class="form-control" />

         ^

In file included from C:\Users\liort\OneDrive\Documents\Arduino\sketch_weather_display_29\sketch_weather_display_29.ino:20:0:

moon_icons.h:403: error: stray '@' in program

[... 497 lines cut ...]

         <td id="LC15" class="blob-code blob-code-inner js-file-line">String      Country = <span class="pl-s"><span class="pl-pds">&quot;</span>UK<span class="pl-pds">&quot;</span></span>;                   <span class="pl-c"><span class="pl-c">//</span> Your country</span></td>

                                                                                                                                                                                                                ^

credentials.h:832: error: expected unqualified-id before numeric constant

       <li class="mr-3">&copy; 2018 <span title="0.29573s from unicorn-7fd7cb5d45-rxqhh">GitHub</span>, Inc.</li>

                               ^

In file included from C:\Users\liort\OneDrive\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.hpp:9:0,

                 from C:\Users\liort\OneDrive\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.h:9,

                 from C:\Users\liort\OneDrive\Documents\Arduino\sketch_weather_display_29\sketch_weather_display_29.ino:22:

C:\Users\liort\OneDrive\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/DynamicJsonBuffer.hpp:16:28: error: expected declaration before end of line

exit status 1
stray '\302' in program

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

G6EJD, 2018-06-24:
This usually happens when you copy from the raw view of the code in github and different browser languages. Try downloading the code in the zip file. There are embedded characters in the code.

lrtal, 2018-06-24:
yess, much better..

now..
Arduino: 1.8.5 (Windows 10), Board: "ESP32 Dev Module, Default, QIO, 80MHz, 4MB (32Mb), 115200, None"

sketch_weather_display_29:89: error: redefinition of 'const char* host'

 const char* host = "api.wunderground.com";

             ^

In file included from C:\Users\liort\OneDrive\Documents\Arduino\sketch_weather_display_29\sketch_weather_display_29.ino:21:0:

sketch\credentials.h:13:13: note: 'const char* host' previously defined here

 const char* host    = "api.wunderground.com";

             ^

exit status 1
redefinition of 'const char* host'

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

G6EJD, 2018-06-24:
Let's recap on the process.
1. Create a folder called 'ESP32_OWM_Current_Forecast_29_epaper_v6'
2. Copy ESP32_OWM_Current_Forecast_29_epaper_v6.ino into the folder
3. Copy 'owm_credentiasl.h' into the folder
4. Copy 'ArialRounded.h ' into the folder

Now the code should compile OK?

Next go to the owm_credentials tab in your IDE and edit your SSID and PASSWORD and OWM key, leave the location as mine until your prove it's working then change your location and country, etc.

Try to compile again, OK, then upload.

lrtal, 2018-06-24:
Arduino: 1.8.5 (Windows 10), Board: "ESP32 Dev Module, Default, QIO, 80MHz, 4MB (32Mb), 115200, None"

ESP32_OWM_Current_Forecast_29_epaper_v6:61: error: redefinition of 'String Units'

 String Units    =  "M"; // M for Metric or I for Imperial

        ^

In file included from C:\Users\liort\OneDrive\Documents\Arduino\ESP32_OWM_Current_Forecast_29_epaper_v6\ESP32_OWM_Current_Forecast_29_epaper_v6.ino:21:0:

sketch\owm_credentials.h:16:8: note: 'String Units' previously declared here

 String Units         = "M";                             // Use M for Metric or I for Imperial 

        ^

exit status 1
redefinition of 'String Units'

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

G6EJD, 2018-06-24:
You have already defined the Units in the main sketch body even though they are already defined in the owm_credentials file !
Do you see the tabs at the top of your IDE if your using the Arduino IDE? You select the owm_credentials.h file which gets included at compilation so no need to redefine anything.

Out of the zip file and with the 3 files I listed above it will compile OK.

G6EJD, 2018-06-24:
delete the line 21 in ESP32_OWM_Current_Forecast_29_epaper_v6.ino
Now try

lrtal, 2018-06-24:
ok, now the compile is good thank you very much!
will connect the display later to the esp32 and try it out, will let you know how it works
Regards
Lior

G6EJD, 2018-06-24:
If you monitor the serial port you should see all the received data.

lrtal, 2018-06-24:
in the monitor

⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮*⸮⸮⸮⸮⸮n⸮⸮!⸮Y⸮ǽ⸮⸮⸮⸮⸮⸮⸮X⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮	⸮⸮B⸮⸮lV(⸮⸮	-⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮X+⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮	⸮⸮t⸮\⸮⸮̆7⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮V%⸮⸮⸮⸮⸮

G6EJD, 2018-06-24:
Speed correct 115200baud?
OWM key correct?

lrtal, 2018-06-24:
my esp32 is form banggood
https://www.banggood.com/ESP32-Development-Board-WiFiBluetooth-Ultra-Low-Power-Consumption-Dual-Cores-ESP-32-ESP-32S-Board-p-1109512.html?rmmds=myorder&cur_warehouse=CN

and the pins are different from your diagram
can you help with that please?

lrtal, 2018-06-24:
https://drive.google.com/file/d/1T9bGPgnU4MZE333Dga5NV4SAUjW5YWYh/view?usp=sharing

lrtal, 2018-06-24:
Connecting to: home net
...WiFi connected at: 134217738
17:18:50
Received OK...
{"cod":401, "message": "Invalid API key. Please see http://openweathermap.org/faq#error401 for more info."}

lrtal, 2018-06-24:
serial monitor is good, but nothing on the display

G6EJD, 2018-06-24:
Which board type are you choosing in the IDE?
You should use: Node32s
Wire the display literally using pin numbers on the ESP32 all these boards are the same, so the wiring should be the same as that in the code.
static const uint8_t SS       = 5;
static const uint8_t MOSI  = 23;
static const uint8_t MISO  = 19;
static const uint8_t SCK     = 18;

G6EJD, 2018-06-24:
If you give me your country and city name I will look up in the OWM list of cities to see if it is covered or what is the nearest for you.

lrtal, 2018-06-24:
Israel
Ramat Gan  OR   Bnei Brak
thanks..
i just subscribed to your YouTube Chanel

lrtal, 2018-06-24:
where do i choose the board type?  i'm new at this.. only a few days

G6EJD, 2018-06-24:
Tools Board

lrtal, 2018-06-24:
i use ESP32 Dev Module board type 
i connected the pins and the display not responding

G6EJD, 2018-06-24:
Your country code is "IL" for Israel 
Nearest Station might be "Kefar Aharon"
or
"Reáº–ovot"

G6EJD, 2018-06-24:
please tell me what pins your connecting to the display
Display     Board
3.3            3.3
Gnd          Gnd
DIN          ?
CLK          ?
CS            ?
DC           ?
RST          ?
BUSY       ?

Make sure your cables are connecting, Sometime they make a loose connection. Sometimes turning the square connector as you push it onto the pins makes a difference.
How long are your board pins? in mm

G6EJD, 2018-06-24:
or 
"Ramat HaSharon"
or
"Tel Litwinsky"
or
"Rosh HaAyin"
or
"Givat Shemuel"
or
"Petah Tikva"
or
"Kefar Shemaryahu" I think this is the closest

lrtal, 2018-06-24:
this is my board image 
https://drive.google.com/file/d/1T9bGPgnU4MZE333Dga5NV4SAUjW5YWYh/view?usp=sharing

3.3 3.3
Gnd Gnd
DIN   23
CLK   clk
CS   5
DC   17
RST 16
BUSY 19

the pins are 5mm
and Petah Tikva is very good!

lrtal, 2018-06-24:
can you please mark the pins by my board image..?

3.3 3.3
Gnd Gnd
DIN ?
CLK ?
CS ?
DC ?
RST ?
BUSY ?
DC 
RST 
BUSY

lrtal, 2018-06-24:
should i use the serial pins on this diagram?
https://cdn.instructables.com/FQM/7X6B/J7GGGD9O/FQM7X6BJ7GGGD9O.LARGE.jpg

G6EJD, 2018-06-24:
CLK to 18

G6EJD, 2018-06-24:
41 static const uint8_t EPD_BUSY  = 4; 
42 static const uint8_t EPD_SS    = 5; 
43 static const uint8_t EPD_RST   = 16; 
44 static const uint8_t EPD_DC    = 17; 
45 static const uint8_t EPD_SCK   = 18; 
46 static const uint8_t EPD_MISO  = 19; // Not used 
47 static const uint8_t EPD_MOSI  = 23; 

You have pin 4 missing on your list? you have connected busy to 19 not 4....

G6EJD, 2018-06-24:
OK try rewiring it all again, this time use GPIO numbers not the arbitrary pin numbers of the board.
41 static const uint8_t EPD_BUSY = GPIO-4;
 42 static const uint8_t EPD_SS = GPIO-5;
 43 static const uint8_t EPD_RST = GPIO-16;
 44 static const uint8_t EPD_DC = GPIO-17;
 45 static const uint8_t EPD_SCK = GPIO-18;
 46 static const uint8_t EPD_MISO = GPIO-19; // Not used
 47 static const uint8_t EPD_MOSI = GPIO-23;

If your board diagram is connect (like most boards are the same) then you have all the pins you need. When pin numbers are quoted they are always General Purpose Input Output pins (GPIO then the port number

lrtal, 2018-06-24:
now on the serial monitor..

rst:0x1 (POWERON_RESET),boot:0x3 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))
waiting for download

G6EJD, 2018-06-24:
Is it a black and white display?

Does it still have the manufacturers image on screen?

Have you ever had the screen working?

lrtal, 2018-06-24:
yes black and white
still have the manufacturers image
tried 2 new ones..

lrtal, 2018-06-24:
maybe i will order a different esp32 ..? if you can recommend on one

lrtal, 2018-06-24:
https://drive.google.com/open?id=1x2vxD867FbbaSCKkFclzEh04EIKbInWU

lrtal, 2018-06-25:
https://drive.google.com/open?id=1MC5waD3EdnOgJLPRCe-BLxyIU64pADTg
https://drive.google.com/open?id=197g3iRxO1sjGtQhm9ut5EQfKQLb6kyQH
https://drive.google.com/open?id=1FuxSO3ilJvrmOqkEAYTsnwowAKqAZJkx

lrtal, 2018-06-25:
hi, do you have a link to a esp32 with normal pin layout?  i will buy a new one..

G6EJD, 2018-06-25:
Your wiring is wrong, see attached wiring diagram.
[image]

lrtal, 2018-06-25:
oh wow tnx, will try it soon and let you know

G6EJD, 2018-06-25:
You had the BUSY connection missing, follow my diagram verbatim and it should work OK. Note all I have done is to follow the pin numbers:
41 static const uint8_t EPD_BUSY = GPIO-4; (D4 to Violet)
 42 static const uint8_t EPD_SS = GPIO-5; (D5 to Orange)
 43 static const uint8_t EPD_RST = GPIO-16; (RX2 to White)
 44 static const uint8_t EPD_DC = GPIO-17; (TX2 to Green)
 45 static const uint8_t EPD_SCK = GPIO-18; (D18 to Yellow)
 46 static const uint8_t EPD_MISO = GPIO-19; // Not used
 47 static const uint8_t EPD_MOSI = GPIO-23; (D23 to Blue)

The only unknown now is RX2 and TX2 by try them initially if it does not work try two other spare pins.

 43 static const uint8_t EPD_RST = 2; //(D2 to White)
 44 static const uint8_t EPD_DC = 15; // D15 to Green)

lrtal, 2018-06-25:
mmm not working.  maybe i should give external 3.3v ?

G6EJD, 2018-06-25:
Are you sure it's wired correctly? You had a missing connection on the pictures you posted (BUSY was missing). I know it's a pain, but have you accounted for all 6 connections.
3.3v supply is more than adequate the display takes little power.
Post a picture please.

G6EJD, 2018-06-25:
You are using a standard e-paper connection?

lrtal, 2018-06-25:
https://drive.google.com/open?id=1xZ-RCqVJ_bHfNpmT6MgYuc9WzPl0GW2Q
https://drive.google.com/open?id=1DzTU7WKiwsN9R5rRcexZJavEkxXcexZP

yes the standard e paper connection

G6EJD, 2018-06-25:
OK, very strange. It does complete the compilation successfully?

You have selected Node32s board type?

Now let's try:
41 static const uint8_t EPD_BUSY = 4; (D4 to Violet)
42 static const uint8_t EPD_SS      = 5; (D5 to Orange)
43 static const uint8_t EPD_RST    = 2; //(D2 to White) ***
44 static const uint8_t EPD_DC     = 15; // D15 to Green) ***
45 static const uint8_t EPD_SCK    = 18; (D18 to Yellow)
46 static const uint8_t EPD_MISO = 19; // Not used
47 static const uint8_t EPD_MOSI = 23; (D23 to Blue)

Can you change the source code for lines 43 and 44 so that RST is on pin 2 and DC is on pin 15 and move the cables accordingly.

Try again.

G6EJD, 2018-06-25:
try this short programme:

void reset(){
  Serial.begin(115200);
  Serial.println(MOSI);   // should be 23
  Serial.println(MISO);   // Should be 19 
  Serial.println(SCK);     // Should be 18
  Serial.println(SS);       // Should be 5
}

void loop(){
}

Copy into a blank sketch, compile and upload and then check serial monitor, may need to reset the board, it should print out the numbers as shown, this tells us the pins are correct.

lrtal, 2018-06-25:
23
19
18
5

lrtal, 2018-06-25:
after i change the code :  i get error ...
// Display connectiopins_arduino.h, e.g. LOLIN32 LITE
static const uint8_t EPD_BUSY  = 4;
static const uint8_t EPD_SS    = 5;
static const uint8_t EPD_RST   = 2;
static const uint8_t EPD_DC    = 15;
static const uint8_t EPD_SCK   = 18;
static const uint8_t EPD_MISO  = 19; // Not used
static const uint8_t EPD_MOSI  = 23;
----------------------------------------------------------
Arduino: 1.8.5 (Windows 10), Board: "Node32s, 80MHz, 921600"

Archiving built core (caching) in: C:\Users\liort\AppData\Local\Temp\arduino_cache_414179\core\core_espressif_esp32_node32s_FlashFreq_80,UploadSpeed_921600_7f8e94b54a37fa76465d6bd3bba002de.a
Sketch uses 636350 bytes (48%) of program storage space. Maximum is 1310720 bytes.
Global variables use 41164 bytes (13%) of dynamic memory, leaving 253748 bytes for local variables. Maximum is 294912 bytes.
esptool.py v2.3.1
Traceback (most recent call last):
  File "esptool.py", line 2637, in <module>
  File "esptool.py", line 2630, in _main
  File "esptool.py", line 2355, in main
  File "esptool.py", line 193, in __init__
  File "site-packages\serial\__init__.py", line 88, in serial_for_url
  File "site-packages\serial\serialwin32.py", line 62, in open
serial.serialutil.SerialException: could not open port 'COM3': WindowsError(5, 'Access is denied.')
Failed to execute script esptool
the selected serial port Failed to execute script esptool
 does not exist or your board is not connected

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

G6EJD, 2018-06-25:
Ok all in order. Do you have a meter to check continuity between display pins and ESP32 pins, so leave everything connected, remove power and check from display pcb soldered connector points to the top of the ESP32 pins.
Have you tried change the pins rx2 and tx2 over.
There is nothing wrong with the code and now it must be the display but the image on the epaper does not look normal, where did you get them from.

G6EJD, 2018-06-25:
When you got to boards and port what number shows up

G6EJD, 2018-06-25:
Unplug your board then go boards port check port numbers then come out of the menu and plug the board in check port again the board should show up. Select that port. Try again.

lrtal, 2018-06-25:
check with the meter is ok.
tried to change the pins .. yes
the epaper is from here
https://www.ebay.com/itm/2-9-inch-E-Ink-Display-e-Paper-Module-for-Raspberry-Pi3-Arduino-SPI-interface-SG/112543054022?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2057872.m2749.l2649

lrtal, 2018-06-25:
can you send me a link to  esp32, i will buy a new one  please

G6EJD, 2018-06-25:
The best one is Lolin d32 from AliExpress

G6EJD, 2018-06-25:
https://m.aliexpress.com/item/32808551116.html?pid=808_0000_0131&spm=a2g0n.search-amp.list.32808551116&aff_trace_key=637b012f1d2f44a08d51b606c5bb8636-1523725859317-02134-UneMJZVf&aff_short_key=UneMJZVf&aff_platform=msite

I think your board is on though

G6EJD, 2018-06-25:
Can you send me your ino file please and I’ll try it on my board like yours davidlhbird@gmail.com

[15 more replies]