G6EJD/ESP32-e-Paper-Weather-Display #15: Json?

schroepfer, 2018-06-27:
Hello,
i get this error. ArdoinoJon 6.0.1.a is installed.
Please, can you help?
BR
Juergen

ESP32_WU_Current_Forecast_29_epaper_v5a:351: error: 'DynamicJsonBuffer' was not declared in this scope
   DynamicJsonBuffer jsonBuffer(10*1024);
   ^

anpfeiffer56, 2018-06-27:
This version requires *** Arduino JSON v6 or above!

schroepfer, 2018-06-27:
When you read my issue, you saw, that i wrote, that i installed ArduinoJson 6.0.1.a !

G6EJD, 2018-06-27:
Ok downgrade the arduinojson library to v5.13.2 via sketch include manage search ‘json’ then select v 5.13.2 this will solve the problem.

schroepfer, 2018-06-27:
Thank you! This works!

Please change your README: "This version requires *** Arduino JSON v6 or above *** ..."

G6EJD, 2018-06-27:
Ok

closed 2018-06-27 by G6EJD