G6EJD/ESP32-e-Paper-Weather-Display #16: Forecasted Weather

Will711990, 2018-10-10:
Hi G6EJD !!

Thank you for your amazing work !!
I've got an issue with the forecasted weather... My graphs stay empty as hourly forecast:

[image]

Do you have a solution for this ?

Thank you !!

G6EJD, 2018-10-10:
What are the diagnostic prints saying on the serial monitor? Copy and paste please.

Is your location supported by OW? Is it on their list of stations?

I can see it’s receiving the first set of data OK, but now I’m wondering why not the second set, now this usually happens when there is insufficient RAM to hold the message as it’s very big in relative terms, are you using the ESP32?

G6EJD, 2018-10-11:
I've just checked the code with 'Fontaine, FR' and it's working OK, it may have been a server issue. Other than that, probably not enough memory? to decode the JSON data, there's a lot...

Will711990, 2018-10-11:
Hi,
thank you for your answer !

My serial monitor is:
```
Connecting to: TP-LINK_FDAF
....WiFi connected at: 1711384768
22:05:29
Received OK...
Creating object...and  Decoding weather data
4.77
46.85
Clear
ciel dégagé
01n

18.00
1015.00
77.00
18.00
18.00
2.60
180.00
0
10000
FR
1539237245
1539277203
01n

tota
```

You're right, the forecasts are not written !

My ESP32 has 32Mb of memory ! I also took a screenshot during the compilation:

[image]

Will711990, 2018-10-12:
I don't know why it doesn't work... I haven't modify the code, but effectively the hadn't saw the French was not supported ! I tried with EN, but no more results !
Do you have a free account on OWM ?

May be should I use another esp !?
Your code does work with ESP8266 ?

G6EJD, 2018-10-12:
Let’s check the code version your using?

The code can’t run on an ESP8266 as the JSON size is too big and needs a lot of RAM.

Yes I’m using a free account.

What happens when you use e.g. Paris or London?

OWM does not translate anything other than the weather description.

G6EJD, 2018-10-12:
I tried your location on a nodeMCU, Firebeetle, D32, D32 Pro all work.

What version of the IDE are you using? Version of the ESP extension 2.4.2?

G6EJD, 2018-10-12:
I just downloaded the latest version, used Fontaine,FR all OK. What do you get, OK no French translations, but easily fixed.

Will711990, 2018-10-12:
It's really strange...

I tried to use exactly your file, and the modifications I done are : wifi pwd, and the apikey.
After I tried to settled arduino IDE (1.8.7) on different cards.

But unfortunately , no more results...

In every cases, thank you for your implication to help me, I really appreciate that !!!

G6EJD, 2018-10-13:
I’ll send you a working code

G6EJD, 2018-10-13:
Email address?

Will711990, 2018-10-13:
m_262@hotmail.fr

Will711990, 2018-10-13:
Wow thank you very much !!! 
It works like a charm... I don't know why it didn't but it's fixed !!!

Just for information, how long is the battery life?

Thank you so much for your patience and your implication !!!

G6EJD, 2018-10-13:
Good to hear.

I get about 4-6 months from a 3800mAhr lipo.

You can get more battery life by reducing the delay values, spends less time awake, but you might find the display does not get fully updated before the ESP32 sleeps, lower temperatures means slower displays.

closed 2018-10-13 by G6EJD