G6EJD/ESP32-e-Paper-Weather-Display #17: Using ESP8266 instead of ESP32

Darkmelman, 2018-11-05:
Hello,

is it possible to use an ESP8266 instead of an ESP32? I ask because there is an ESP8266 E-Paper Driver Board from Waveshare and this would be perfect for use with the 7.5" Display.

Which changes must be do on the code to run it on the ESP8266?

I changed `#include <Wifi.h>` to `#include <ESP8266Wifi.h>` and commend out the lines

`esp_sleep_enable_timer_wakeup(UpdateInterval);`

and 

`esp_deep_sleep_start();         // Sleep for e.g. 30 minutes`

also the function UpdateLocalTime(); 

```
void UpdateLocalTime() {
 struct tm timeinfo;
  while (!getLocalTime(&timeinfo)) {
    Serial.println(F("Failed to obtain time"));
  }
  //See http://www.cplusplus.com/reference/ctime/strftime/
  //Serial.println(&timeinfo, "%a %b %d %Y   %H:%M:%S");      // Displays: Saturday, June 24 2017 14:05:49
  Serial.println(&timeinfo, "%H:%M:%S");                      // Displays: 14:05:49
  char output[30], day_output[30];
  if (Units == "M") {
    strftime(day_output, 30, "%a  %d-%b-%y", &timeinfo);      // Displays: Sat 24/Jun/17
    strftime(output, 30, "( Updated: %H:%M:%S )", &timeinfo); // Creates: '@ 14:05:49'
  }
  else {
    strftime(day_output, 30, "%a  %b-%d-%y", &timeinfo);      // Creates: Sat Jun/24/17
    strftime(output, 30, "( Updated: %r )", &timeinfo);       // Creates: '@ 2:05:49pm'
  }
  Day_time_str = day_output;
  time_str     = output;
}
```

and the call of the function:

`UpdateLocalTime();`

Pin- Setting changed to:

```
static const uint8_t EPD_BUSY = 16;
static const uint8_t EPD_SS   = 15;
static const uint8_t EPD_RST  = 5;
static const uint8_t EPD_DC   = 4;
static const uint8_t EPD_SCK  = 18;
static const uint8_t EPD_MISO = 19; // Master-In Slave-Out not used, as no data from display
static const uint8_t EPD_MOSI = 23;
```

The Display is updating but without any data... 

I would be nice if anyone can help me to get it working!

Thanks a lot, Regards

Darkmelman

G6EJD, 2018-11-05:
The ESP8266 has no where near enough memory to draw the 7.5” image before it’s transferred and it needs a lot of memory to receive the JSON formatted data from OWM. You could try out the streaming version that uses considerably less memory for the JSON decoding but there’s still the display driver issue. Try out the streaming version first and do what you’ve done for WiFi etc and see how you get on.

Darkmelman, 2018-11-06:
Thanks for the fast answer!

I will try the streaming version later!

G6EJD, 2018-11-21:
The GxEPD library can display on an 7.5" e-paper using an ESP8266 and the streaming version uses a lot less memory too. But it will require some minor amendments (the basic graphic constructs are near identical).

closed 2018-11-21 by G6EJD