G6EJD/ESP32-e-Paper-Weather-Display #19: not working with Waveshare ESP32 board

matphillips, 2019-01-02:
Waveshare have a rather nice driver:
https://www.waveshare.com/wiki/E-Paper_ESP32_Driver_Board

So I've got one of those and their 7.5 mono display. They have demos code that allows sending an image to the display using a web browser and I've ran that and it works fine, so the board and connections are good.

The Weather Display code does nothing though. I've got the pin assignments set to the same values as their demo code, the output in the serial console seems fine (gets the data from web service, calls update screen code) but no data is reaching the screen.

Any ideas on the cause or how I can investigate?
Thanks.

G6EJD, 2019-01-02:
This is almost certainly due to incorrect pin assignments for that board, what pins have you defined in my code please?
static const uint8_t EPD_BUSY = 4; 
static const uint8_t EPD_SS      = 5; 
static const uint8_t EPD_RST    = 16; 
static const uint8_t EPD_DC     = 17; 
static const uint8_t EPD_SCK    = 18; 
static const uint8_t EPD_MOSI  = 23; 

The Waveshare documentation is (as usual) poor and uses non-standard terms so it's difficult to ascertain what pins they are using for the display/SPI, certainly non-standard usage for sure.

matphillips, 2019-01-02:
This is what I have in your code:
// Waveshare ESP32 ePaper board
static const uint8_t EPD_SCK  = 13;
static const uint8_t EPD_MOSI = 14;
static const uint8_t EPD_SS   = 15;
static const uint8_t EPD_BUSY = 25;
static const uint8_t EPD_RST  = 26;
static const uint8_t EPD_DC   = 27;

Those are the right assignments according to the example code for the board.
I think the EPD_SCK and ESP_MOSI values are not actually being used? I edited the pin definition file for a random board (there's no built in board type for this board) and I edited the SCK and MOSI pins in there, and communication happens. So that's good.

Don't know if this is something you can fix, it may be a limitation of the driver?

There seems to be an issue with clearing the screen however, in that, the screen flashes but the old image is not erased.

G6EJD, 2019-01-02:
OK according to the schematic all pin assignments are correct.  But you don't need a new driver, the code v8 as published with the ThingPulse epd driver works well with the display, just set the pins assignments as you have above. 
Select the ESp32 Dev Board so there is no pin remapping and it should all work. What does it say on the serial port, is it receiving weather data and displaying it and then updating before going to sleep?
You may need to add a delay(2000); just before it goes to sleep to allow the screen to catch up.

G6EJD, 2019-01-02:
You need to be at ESP core 1.0 and ESP8266 2.4.2 and ideally 1.8.7 or later for the IDE if using Arduino IDE.

Boards, Board Manager, search for ESP to check.

G6EJD, 2019-01-02:
BTW SCK is essential as its the SPI clock and MOSI is master out data so essential for the display to work.

I'm using the same waveshare boards up to the ribbon cable,

matphillips, 2019-01-02:
Thanks,
Let's add some data to the record,
- Arduino 1.8.8
- ESP32 version 1.0.0 (in the boards manager)
- using sketch: ESP32_OWM_Current_Forecast_75_epaper_v8.ino
- ESP32 Dev Module selected as the board in use

In this configuration, no data reaches the screen. If I try adding the following lines, to redefine the pins:
static const uint8_t SCK  = 13;
static const uint8_t MOSI = 14;

Then of course the IDE complains about the const being redefined.
a quick edit to ~/user/Library/Arduino15/packages/esp32/hardware/esp32/1.0.0/variants/esp32/pins_arduino.h and the display works again. I really don't think the EPD_SCK variable is being used anywhere.

I then added a ten second delay before the sleep, but there's still the problem with the screen not being cleared.
I can record a demo video if that may help?

matphillips, 2019-01-02:
recorded a demo anyway, hope this explains it better :)
https://www.dropbox.com/s/tscdli8ow6eatgm/epaper%20issue.mov?dl=0

G6EJD, 2019-01-02:
Ok now I see what’s happening and I presume your display is fairly new m, which makes me think the display has been updated with a new command for clear screen.

I suggest you try out the GxEPD driver as I know that author gets the latest displays from waveshare and so should work.

In his library there are lots of examples and setting the pins for an ESP32 is done in that section, note this bit can be tricky. Worth a try.

If that works the code needs updating a little but both use the GFX baseline.

matphillips, 2019-01-02:
Yes, screen arrived this morning ;)
GxEPD2_Example seems to work correctly, I'll try to get the Weather code working with this library.

G6EJD, 2019-01-02:
The conversion is quite straightforward only one command to limit width of the weather forecast won’t work but could be coded appropriately, I’ll have a go later.

matphillips, 2019-01-02:
Then I'm doing something wrong! Hitting issues with .drawString .setFont .setTextAlignment.......
I'll take a step back and work on something else for a bit!
Thanks for all your attention with this.

G6EJD, 2019-01-02:
There is no standardisation in this world 😊the parameters are different and perhaps a little more efficient in the GxEPD library I should have it converted later this afternoon in the UK.

G6EJD, 2019-01-02:
I have converted a version for GxEPD, but there are issues:
1. the smallest font is 9pt, it needs to be at least 6pt and I can't find any.
2. There is no ° symbol in the fonts.
3. I have not yet found any smaller fonts.
Uploaded as an example.

matphillips, 2019-01-02:
Awesome!
I left the panel in the office but will check this out tomorrow, even 'not right' it's better than not working!
I will attempt to find a solution to those font issues too and submit a PR if I manage to solve it.

Huge thanks!

AWSW2017, 2019-04-28:
Hello,

I hope you can help me with the E-Paper ESP32 Driver Board in combination with the WaveShare 7.5” ePaper display?
https://www.waveshare.com/wiki/E-Paper_ESP32_Driver_Board
https://www.waveshare.com/wiki/7.5inch_e-Paper_HAT

I think I have a similar problem as here described.

What do I need to change to make the code compatible to the Waveshare ESP32 Driver Board ? 

I am actually using the following German sketch with the only changes done for the WiFi access, location and the api key for the openweathermap access. 
https://github.com/G6EJD/ESP32-e-Paper-Weather-Display/blob/master/German/ESP32_OWM_Current_Forecast_Streaming_75_epaper_v3.ino
In the serial monitor I can see that the download of the weather data works but the 7.5”  screen is not updated at all. I tried to perform the changes as described here, but this did not work so far.

Thank you in advance :)

Kind regards

G6EJD, 2019-04-29:
I know what the problem is and hopefully you have now solved it as the issue is closed.

The thingpulse display driver starts the spi bus with spi.begin(); and assumes the default spi connections, whereas the Waveshare board for some strange reason, probably for ease of pcb layout has chosen different pins, I can list these out for you, but the solution is to modify the spi.begin statement to include PIN numbers e.g. spi.begin(mosi,miso,clk,dc); also CS will be different and that needs modifying. 

Have you solved this now. Looking at the Waveshare schematic I note it is not capable of really low power consumption unlike the lolin d32 can.

AWSW2017, 2019-04-29:
Hello G6EJD,
that would be a great help if you can help me with this problem. I could not solve it so far. 
I just did not want to open another topic to the same problem. 
Thank you in advance.

svwhisper, 2019-04-29:
> Have you solved this now. Looking at the Waveshare schematic I note it is not capable of really low power consumption unlike the lolin d32 can.

What have you found on the schematic that indicates this won't be able to sleep with low current draw?  That would be disappointing, for sure.

G6EJD, 2019-04-29:
These are the pins for the Waveshare module:
tatic const uint8_t EPD_SCK = 13;
static const uint8_t EPD_MOSI = 14;
static const uint8_t EPD_SS = 15;
static const uint8_t EPD_BUSY = 25;
static const uint8_t EPD_RST = 26;
static const uint8_t EPD_DC = 27;

Next you need to modify the display driver and change SPI.begin(); to SPI.begin(CLK,MISO,MOSI,SS);

or for your board SPI.begin(13,19,14,15);

Then in the main programme body:
static const uint8_t EPD_SCK = 13;
static const uint8_t EPD_MOSI = 14;
static const uint8_t EPD_SS = 15;
static const uint8_t EPD_BUSY = 25;
static const uint8_t EPD_RST = 26;
static const uint8_t EPD_DC = 27;

G6EJD, 2019-04-29:
Now find here's the source code https://github.com/ThingPulse/minigrafx/blob/master/src/EPD_WaveShare_75.cpp

So, in your local library folder go to the minigrafx folder then edit EPD_WaveShare_75.cpp and find the SPI.begin line (at line 294) and change it to SPI.begin(13,19,14,15);

Save it. Compile and try it out.

AWSW2017, 2019-04-29:
You made my day!
That really works well.
Thanks a lot and have a nice day.

G6EJD, 2019-04-29:
Happy to help. Enjoy.

closed 2019-04-29 by G6EJD

Joppl, 2020-07-03:
An AWSW2017
Ich wollte auch das Projekt nachmachen (ganz neuer Anfänger)
Habe auch folgende Hardware:
https://www.waveshare.com/wiki/E-Paper_ESP32_Driver_Board
https://www.waveshare.com/wiki/7.5inch_e-Paper_HAT V.2
Ich bekomme es einfach nicht hin
Gibt es eine Anfängerlösung?

Danke Josef

G6EJD, 2020-07-03:
What is your hardware please?
Have you connected the display?
Can you see weather data on the serial port?
Let me know and I’ll post details of how to fix it for you

Joppl, 2020-07-03:
Hello, thank you and sorry for my english, i will try

Hardware

Display Waveshare 7.5inch e-Paper Hat   SKU: 13504

e-paper ESP32 Driver Board

 

Can you see weather data on the serial port?    no idea how that works, I am a Beginner

 

please also note my photo in the appendix

 

thanks Josef

 

 

Gesendet: Freitag, 03. Juli 2020 um 13:27 Uhr

G6EJD, 2020-07-03:
OK, do this:
1. Download the programme and edit the version for your display board, that's the T7 type:
https://github.com/G6EJD/ESP32-e-Paper-Weather-Display/tree/master/examples/Waveshare_7_5_T7
2. Now make sure the display driver pins look like this:

Joppl, 2020-07-03:
I get an error message

Multiple libraries were found for "WiFi.h"
In file included from C:\Users\josef\AppData\Local\Temp\arduino_modified_sketch_273428\Waveshare_7_5_T7.ino:26:0:
 Used: C:\Users\josef\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\WiFi
C:\Users\josef\Documents\Arduino\libraries\GxEPD2-master\src/GxEPD2_BW.h:29:26: fatal error: Adafruit_GFX.h: No such file or directory
 Not used: C:\Users\josef\Desktop\arduino-1.8.13\libraries\WiFi
compilation terminated.
exit status 1
Error compiling for board ESP32 Dev Module.

Joppl, 2020-07-03:
at 4.Now upload the sketch

G6EJD, 2020-07-03:
You need to install the libraries Adafruit GFX, foto sketch manage in the top right type GFX it will find the library install it. Same for any other libraries required in the reader the duplicate WiFi.h does not matter

Joppl, 2020-07-03:
Yes Yes Yes you are the best i hav e picture

Joppl, 2020-07-03:
Please can you tell me how i can change to german and my City

Joppl, 2020-07-03:
I have found everything and can change only under temperature and air pressure is Clouds (few clouds) in English

thanks for your help i love your project

G6EJD, 2020-07-03:
Sorry for the delay, been working.
Uncomment the German include at the start, comment out the English. Then edit the German language.h to your requirement, the German is already done anyway so no work to do just include it.

Joppl, 2020-07-03:
Everything is ok. Thanks for your Time
Thank you very much, wish you a quiet working day

G6EJD, 2020-07-03:
Good to hear all is working now.

AWSW2017, 2020-07-03:
> An AWSW2017
> Ich wollte auch das Projekt nachmachen (ganz neuer Anfänger)
> Habe auch folgende Hardware:
> https://www.waveshare.com/wiki/E-Paper_ESP32_Driver_Board
> https://www.waveshare.com/wiki/7.5inch_e-Paper_HAT V.2
> Ich bekomme es einfach nicht hin
> Gibt es eine Anfängerlösung?
> 
> Danke Josef

Sorry, I was to busy today with work... Just saw your request. 
Glad you got it running. 

Thanks to G6EJD again for this awesome support. 😊

Joppl, 2020-07-05:
Hello
I have another question:
I now operate the weather display with a Mi Powerbank 10,000 mAh and everything works as usual, unfortunately I am not shown the battery level in% and V. The battery field is empty
My ESP32 is a Waveshare Driver Board Power is fed in via USB.
Is there a way to change something?
Many thanks in advance Josef

AWSW2017, 2020-07-05:
Hi,
there is a way with 2 resistors described here: https://github.com/G6EJD/ESP32-e-Paper-Weather-Display/issues/22 which i used at first to. But the deep sleep battery usage performance did not impress me to much with the WaveShare ESP32. 

Therefore i switched to the LOLIN D32 board which has a special battery connector for LiPo batteries (Akkus) and charges and monitors in all of my ePaper projects the battery as you request to be shown in the display section...

I could not find the LOLIN D32 on f.e. Amazon but on AliExpress. You then need to setup the default cable setting to use the board as default set in the sketch here.  

I use a 5000mA battery in a frame with the old ePaper (640x384) with the LOLIN D32 and it was charged on 1st of January this year and has today (5th of July 2020) still 85% remaining which I think is pretty nuts therefore that it updates every 30 minutes during 6am and 23pm...

On a other frame I use the new V2 ePaper display now about 2.5 months with a 10000mA battery and still runs at 99% left...

The LiPo batteries can be found at Amazon or AliExpress as well... To connect them you need a small connector cable named “Micro JST PH 2.0 2-Pin“ wich needs to be soldered to the battery...

Happy updating & viel Spaß 😉

Joppl, 2020-07-05:
Ok thanks for the info, then there will soon be a lolin board
Thank you Josef

Joppl, 2020-07-05:
Hello I do not get the data below correctly in lang_gr.h, so in my weather display there are clouds (few clouds) but I would like ein paar Wolken displayed better.
Thanks Josef
I copied the following into lang_gr.h it but it doesn't work

/*String Translate_EN_DE(String text) {
  if (text == "clear")            return "klar";
  if (text == "sunny")            return "sonnig";
  if (text == "mist")             return "Nebel";
  if (text == "fog")              return "Nebel";
  if (text == "rain")             return "Regen";
  if (text == "shower")           return "Regenschauer";
  if (text == "cloudy")           return "wolkig";
  if (text == "clouds")           return "Wolken";
  if (text == "drizzle")          return "Nieselregen";
  if (text == "snow")             return "Schnee";
  if (text == "thunderstorm")     return "Gewitter";
  if (text == "light")            return "leichter";
  if (text == "heavy")            return "schwer";
  if (text == "mostly cloudy")    return "größtenteils bewölkt";
  if (text == "overcast clouds")  return "überwiegend bewölkt";
  if (text == "scattered clouds") return "aufgelockerte Bewölkung";
  if (text == "few clouds")       return "ein paar Wolken";
  if (text == "clear sky")        return "klarer Himmel";
  if (text == "broken clouds")    return "aufgerissene Bewölkung";
  if (text == "light rain")       return "leichter Regen";
  return text;
  }
*/

G6EJD, 2020-07-05:
Put a serial.println(text); in the code to see what is being received, maybe “Few clouds” which is not “few clouds” solution
Text.tolowercase();

G6EJD, 2020-07-05:
I thought OWM did provide translations of weather text, clearly not! Do you have the language set correctly?

Joppl, 2020-07-05:
yes  
//#include "lang.h"                     // Localisation (English)
#include "lang_gr.h"                  // Localisation (German)
and in owm_credentials.h
String Country          = "AT";                           
String Language         = "AT";         

I think there is already an approach
https://github.com/G6EJD/ESP32-e-Paper-Weather-Display/issues/105

Joppl, 2020-07-05:
ok now i set the 
String Language         = "DE";    
an it shows Clouds ( Ein paar Wolken) 
perfect, I can live with that

Joppl, 2020-07-05:
Thank you again, Mr. G6EJD, for letting me recreate and use your project and for being patient with a beginner.
Thanks Josef from Austria

G6EJD, 2020-07-05:
AT ? And not DE ?

G6EJD, 2020-07-05:
You could create an AT language file by copying the DE version and fill in the translation to your requirement, when complete put the file the sketch folder then change your #include to your new AT file.