G6EJD/ESP32-e-Paper-Weather-Display #22: Battery level not correct or showing

not4unco, 2019-03-15:
Good evening Dave,

I'm currently using your Streaming V1 ino and i'm having issues with displaying the correct battery voltage. When the default ino file is loaded there isn't any battery reading at all even when my battery voltage is reading 4.03v. At times it will read the battery and will only show 0.15V on the screen as I changed the if (voltage >1) to 0.000. The battery i'm currently using is a 3.7v 10000mAh.

G6EJD, 2019-03-15:
I need to know what type of development board your using?

From your description of the readings it’s likely your board does not have any battery monitoring capability, only the Lolin D32 has the capability to do it, but you can add external resistors to achieve the same result.

not4unco, 2019-03-16:
Well that's a big duh! on my part then. I have the Login 32 Lite. Thanks Dave....Do you still recommend the 220k&100k voltage dividers for battery monitoring?

G6EJD, 2019-03-16:
Yes, no vbat monitor on the Lite board, so add a voltage divider as you describe using 100K resistors. Providing the code is correct and I just checked it is, then the sunset time issue could be a OWM issue, what location are you using?

not4unco, 2019-03-16:
I switched the location city to location I'd as in the US there are a few locations here with the same name. The Unix time is different between the two looking at the serial output from the Json data.

G6EJD, 2019-03-16:
Maybe a display problem if the codes been modified for a smaller/different display. I think the issue can be closed now.

closed 2019-03-16 by G6EJD

not4unco, 2019-03-16:
Dave, my apologies....i just checked the unix time code online and it is correct. Yesterday and today's unix time stamp is the same as far as time goes when it comes to hour and minutes. Only thing missing is the am/pm time indicator. Everything works great though..thanks for your help and patience.

G6EJD, 2019-03-16:
You could add a am/pm function that draws in superscript style just after the time like the weather symbols. Time should already be be correctly formatted for you using the posix format, but it can be tricky to know when it’s pm as a simple time > 1100 is no good but you could pick up the am or pm text in the time String to determine which.

Thoky81, 2020-08-18:
Sorry for opening the thread again but I would like to ask more about the wiring of voltage regulator for **Lolin32 Lite**.
I have connected 100k from 3.3V and 100k from ground to pin 35. The voltage I am getting (with voltmeter) is 1.57-1.58V on the pin 35. I am doing something wrong or what should I change in code in order to get proper battery readings? Is this voltage regulator I just mention correct?
I have also try to solder the power from battery connector and ground from G pin on the board. The display was showing 100% battery and it was running for approx 2-3 days but then the battery was drained. I am using small 1000mAh LiPo.

Thanks in advance

G6EJD, 2020-08-18:
Hello, the reading your getting after adding a /2 voltage divider is correct and within tolerance so 3.3/2 = 1.65 so you can calibrate your ADC Reading by adding a small offset to compensate.
Voltage = Voltage * 2.1 
You have to multiply the result by x2 because you divided the input by 2.

The ESP32 lite contrary to what it’s name might suggest is not low power and the regulator and UART run continuously which is why it consumes power and discharges your 1000mahr battery so quickly 1000/48=20mA which sounds like the ESP32 is not in sleep mode.
The best board to use is the genuine Lolin D32 board only available at the Lolin store on AliExpress about £4 or equivalent

Thoky81, 2020-08-18:
Hello and thanks for reply.
So I guess it should be like this?
`Serial.println("Voltage * 2.1 " + String(voltage));`

Regarding the board I have this one:
https://a.aliexpress.com/_dXI2qNW

And I also purchase the pro version which is still on the way:
https://a.aliexpress.com/_d6Du9vi

So dont know if those are the correct ones.

Plus there is a version of Lolin D32 Pro with TFT port. They also sell cable to connect epaper display. Do you know if this works?
https://a.aliexpress.com/_dYzk2oc

Thank you

Thoky81, 2020-08-18:
... just found your video https://youtu.be/JaNAH2MHlAE 😀

G6EJD, 2020-08-19:
It looks like the Lolin D32 is currently unavailable, I suspect there is no stock: https://lolin.aliexpress.com/store/1331105?spm=a2g0o.detail.1000061.1.cbd92e52A6f5gNhttps://lolin.aliexpress.com/store/1331105?spm=a2g0o.detail.1000061.1.cbd92e52A6f5gN

The TFT port can be used on the V2 Pro, but personally I don't like the battery performance of the Pro board and it's very difficult (impossible) to get the SD Card working when using the TFT display port.