G6EJD/ESP32-e-Paper-Weather-Display #26: e-Paper Display - contrast

hoffmakl1961, 2019-05-22:
I use the Waveshare 7,5" e-Paper Display with e-Paper HAT (SPI) as Interface to the ESP32 and the software package ESP32-e-Paper-Weather-Display/German/ changed by me. 

At the beginning I have a very good black and white contrast. After a few days, the contrast will always be less and less until it's still "white". 
I changed the e-Paper Display and e-Paper HAT interface PCB, but the result was the same thing.  

If I switch off for same days and switch on  I've also a very good black and white contrast. After a few days, the contras
[image]
t will always be less and less

Any idea's what could be the reason for this problem?

G6EJD, 2019-05-22:
This is either due to a faulty display or driver.

I've been using my 7.5" for about a year now and occasionally I get a similar effect, usually when it is in a cold environment, but usually it recovers. I had contemplated contacting Waveshare as others have reported similar problems and they have replaced them.
It could also be a display driver problem, you could try the GxEPD version to see if this makes a difference.

You are running the display from a 3.3v supply? They are very intolerant of 5v, whilst they do work in the short-term.

It's not a software issue per-se, that is more likely a driver issue.

hoffmakl1961, 2019-05-22:
Thank you for your quick answer. 
The temperature was relativly constant about 23°C. For the display I use the 3.3V power supply from the ESP32 and the power for the ESP32 will be provided by the USB (5V). 

I'd exclude a hardware problem because I had the same problem with 2 different displays and 2 different e-Paper HAT interface PCBs. 

My suspicion is also that I have a driver problem. 

It's not a software issue.

closed 2019-05-22 by hoffmakl1961

G6EJD, 2019-05-22:
Try the GxEPD variant in the same folder.

hoffmakl1961, 2019-05-24:
I tested with the new verion: ESP32_OWM_Current_Forecast_75_epaper_v10.ino and  lib: GxEPD2 Version 1.1.7. 
The result was the same. 

[image]

G6EJD, 2019-05-24:
You beat me to it, I was going to let you know I’d converted the code to GxEPD2 and to try it.
Thanks for letting me know, I have 3 7.5@ displays and they are all working OK, one for over a year now. 
Yours are an interesting problem and thanks for letting me know, as I rather hoped the revised and better driver would help solve your problem.
Did you buy them from Waveshare? If so, they are very good at customer service as I’ve had two replaced so-far, one developed vertical bars and they said one of the column drivers had failed, the other a 2.9” 3-colour gained an overall red tinge.
Interested by your problem I have been searching for people reporting display fading, which I have experienced, but have found no reports, which by itself is interesting. Did you buy them through AliExpress?
There is limited documentation about running the display (by Waveshare) on 5v for a very short time as a way of clearing the panel, but I have not tried that.
So in your case, the panel is being polarised with the image because we can see it with low contrast, so when the display driver charges the panel to release the black pigment, that is not happening, well only to s limited extent and in your panels just s fee ink particles are separated. This tells us something.
So my next question is have you tried a different board to drive them, and what type of ESP32 board are you using?  Some boards have a limited capacity regulator that can barely provide enough power to the ESP32 let alone an e-paper display, but I think you said you tried powering it separately from the ESP32?
Assuming your panels are OK then something has to be common if mine are ok with the same code.

G6EJD, 2019-05-24:
When my display did fade, interestingly it was similar to yours with progressively less contrast from left to right. I don’t know why but it’s common.

G6EJD, 2019-05-24:
Do you have the e-paper hat set to 4-wire?

hoffmakl1961, 2019-05-24:
The display I bought at EXP GmbH - Saarbrücken (Germany).
The Board is a NodeMCU ESP32. I haven't tested with an other ESP board.
The e-paper hat interface switch is set to 0 = 4-wire SPI. 

Power consumption of the ESP is very high if the WiFi is launched and runs. That could be a problem
for Display. I'll test with a separate power supply. 

Test without WiFi >> same Hardware

lib: GxEPD2 Version 1.1.7 and with the GxEPD2_Example from the lib 
no problem - a very good contrast

G6EJD, 2019-05-24:
I’ll try the GxEPD2 examples later, so that could be a hint, as and although I can see no implementation differences, in my code there is hardly any time / delay before it goes to sleep maybe before the screen update has completed, I’ll also check the GxEPD2 library to see if the BUSY line is used. By the way the pin out of the NodeMCU and Lolin D32 are different so I’d be interested to know which pins you are using just to be sure.

G6EJD, 2019-05-24:
I've just uploaded v10a which does not use Paged Mode and does a full screen update, that might help.

Zinngg does discuss power supply issues e.g. the Wemos D1 Mini is in-capable of providing enough power to update the display correctly, this could be similar to your problem. I'll check the nodeMCU-32 schematic to see what regulator they are using - AMS1117-3.3 capable of 800mA, but not a LDO type at 1.1v difference required, so should not be a problem, 450mA peak for the ESP32 and 38mW or 12mA for the display during update, so no issues there.
Ideally try another microUSB cable to eliminate that too.

hoffmakl1961, 2019-05-25:
G6EJD - thanks for your support. 
// busy pin 
static const uint8_t EPD_BUSY = 4; 
regarding power:  I'll use an separate, external power supply (3.3 V). with a lab power supply. 

I don't have time right now, but I'll test asap.

G6EJD, 2019-05-25:
OK, thanks for confirming BUSY pin.
Separate power might help, as I don't use NodeMCU32, then there's a difference in test setups.
When you get a chance, v10a uses the GxEPD2 revised screen update and would be worth a try.

hoffmakl1961, 2019-05-25:
My Test with the external power supply was unsuccessful. 

1. I interrupt the 3.3V pin from the ESP. 
2. connect a switched off lab power supply. 
3. switched on the ESP32, all pins from e-paper hat are connected to the ESP expect the 3.3V power. 
4. switched on the e-paper hat 3.3V. 

nothing happens only a gray screen

On the seriell monitor I see that the weather date will be downloaded and go in deep-sleep.

I removed the external power supply and reconnect the power from the ESP. 
nothing happens only a gray screen

I'd asssume that the e-paper - hat is broken. 

After the ESP switch on, same pins have a high level but the electronic of the e-paper hat was switched off.

G6EJD, 2019-05-25:
It does sound like the HAT may be faulty, did you say the GxEPD2 examples worked OK?

G6EJD, 2019-05-25:
Did you have a common ground between ESP32 and external PSU?

hoffmakl1961, 2019-05-26:
as described above - the GxEPD2_example worked OK. 
after my power supply test. - nothing happens only a gray screen
Yes - a common ground between ESP32, e-paper HAT and external PSU is present.

G6EJD, 2019-05-26:
For the avoidance of doubt, the GxEPD2 examples are also showing dim results?
The only common item is the HAT but it’s passive, so an Interestingly bizarre fault. Unfortunately only a replacement will enable it to be illuminated easily.

hoffmakl1961, 2019-05-26:
For the avoidance of doubt, the GxEPD2 examples are also showing dim results?  No. 

Sorry my english is not so good. .. the HAT but it’s passive - What does it mean? 
The HAT have also same active components.

hoffmakl1961, 2019-05-26:
I'll change the Hat. but it's not so easy. 
[image]

G6EJD, 2019-05-26:
The photo very useful and I see why changing won't be easy. Actually looking at the HAT schematic there seems no point in changing, it is as I thought passive (no active components apart from it's PSU).
What do we know:
1. Using the GxEPD2 examples gives a display that is clear and high contrast - yes.
2. Using my weather code gives a display that is low contrast and dim - yes.
3. Now we must check the software connections for each test, in GxPED2 the connections for an ESP32 are as follows:
SS=5
DC=17
RST=16
BUSY=4
All other SPI are default MOSI = 23, MISO= 19, CLK = 18
4. Looking at your wiring I can see GPIO4 but I don't see a wirte to it?
5. Looking at the HAT, I see a grey wire on Busy and that goes to the 4th pin up from the bottom-left, which GPIO port is that?

hoffmakl1961, 2019-05-26:
I checked the connections between ESP and HAT.  
What for a stupid mistake. 
For Busy I defined in the sketch pin 4 and use GPIO 15. 
I'm getting old. Thanks for the tip. 

After my test with the power supply, the screen was gray - with the GxEPD2_Example and weather sketch. 
Now I changed the HAT. but the screen remains gray.
After that, I changed the display and upload the GxEPD2_Example sketch. And it works!
After that, I uploaded the German weather sketch (with EPD_BUSY = 15;) - and I have a very nice contrast. 

The test with V10a was unsuccessful. 
1. I've got some warnings e.g. 
iteration 23u invokes undefined behavior [-Waggressive-loop-optimizations]    
temperature_readings[r] = WxForecast[r].Temperature; 

note: containing loop
    for (int r = 1; r <= max_readings; r++) {

and here a picture from the screen with V10a. 
[image]

In the next few days I will test the contrast  with the "old" Version. 

Back to the HAT
I haven't a schematic from my HAT, but I find some active components. 
[image]

Tanks again and a nice Sunday evening.

G6EJD, 2019-05-26:
Your HAT is completely different from mine, Imine just has series resistors to the display, interesting.

 

 

To use the v.10a just change static const uint8_t EPD_BUSY = 4; to static const uint8_t EPD_BUSY = 15; to match your wiring, this explains why the programme stopped updating the screen (dim) before going to sleep, now it will wait until the screen is complete.

 

Those warnings: iteration 23u invokes undefined behavior [-Waggressive-loop-optimizations] are completely normal, what it’s saying is there is nothing to stop an excursion beyond 23 , but there is, nothing to worry about.

Regards

 

 <https://user-images.githubusercontent.com/42253824/58384469-14370f80-7fe2-11e9-8e4d-c0b4db5848c4.jpg> 

 <https://user-images.githubusercontent.com/42253824/58382766-b2b97580-7fce-11e9-9c41-80bd0b94cd56.jpg>

hoffmakl1961, 2019-05-28:
Before I tested with v.10a I changed BUSY to 15.
(It's also defined in the GxEPD2_BW<GxEPD2_750, GxEPD2_750::HEIGHT> ... line)

As a first look, all works good. 

In v.10a you changed the fonts from ArialRounded.h to FreeMonoBold12pt7b.
The disadvantage for us in Germany is that the umlauts aren't available. 

I've created a test file lang_gr.h 

Some word's or word groups are too long in German.

//Sonne
const String TXT_SUNRISE  = "Sunrise"; // Sonnenaufgang - word is to lang
const String TXT_SUNSET   = "Sunset";  // Sonnenuntergang - word is to lang

//Moon
const String TXT_MOON_NEW             = "Neumond";
const String TXT_MOON_WAXING_CRESCENT = "zunehmende Mondsichel";
const String TXT_MOON_FIRST_QUARTER   = "zunehmender Halbmond";
const String TXT_MOON_WAXING_GIBBOUS  = "zunehmender Mond";
const String TXT_MOON_FULL            = "Vollmond";
const String TXT_MOON_WANING_GIBBOUS  = "abnehmender Mond";
const String TXT_MOON_THIRD_QUARTER   = "abnehmender Halbmond";
const String TXT_MOON_WANING_CRESCENT = "abnehmende Mondsichel";

For measurement units I would use [] and not (). 

At the moment, the contrast is very good.
[lang_gr.zip](https://github.com/G6EJD/ESP32-e-Paper-Weather-Display/files/3227542/lang_gr.zip)

[image]

G6EJD, 2019-05-28:
Good to hear all is working now with a contrast the way it should be. 
The problem with the GxEPD screen library is it uses 7-but fonts so nothing above 127 where the accented characters are normally held. I’ll have a look for a suitable font.

hoffmakl1961, 2019-05-29:
The datasheet of the display defined a range from 0 to ~50°C. 
Inside the room, when the sun shines on the display >> the contrast will be bed, but  it recovers in the shadow. 
conclusion: The display can only be used in the shadow.

G6EJD, 2019-05-29:
I've just uploaded an improved v11. Corrected a few errors in text width, and forecast text, which can be quite long, now wraps. New fonts too.
Yes, the displays are not perfect, but 50° is quite hot for most things.

timo619, 2021-12-14:
I have maybe the same issue with another setup (raspberry pi --> waveshare HAT --> 7.5inch EInk display) but cannot find the problem where it is coming from. What did you change in the software v10 so that it works?

[https://stackoverflow.com/questions/70318231/e-ink-display-shows-weird-behaviour-with-horizontal-lines-no-contrast](https://stackoverflow.com/questions/70318231/e-ink-display-shows-weird-behaviour-with-horizontal-lines-no-contrast)

Kind regards,
timo

G6EJD, 2021-12-14:
History - the display.init function failed when Waveshare changed the reset circuit of the epaper to suit the RPi, so the init function was modified to solve that for non RPi devices and there have been further changes for non RPi devices so it may well be that the library is no longer suitable for RPi usage.
There is no issue with this software.