G6EJD/ESP32-e-Paper-Weather-Display #37: forecast_record.h is missing

JonasDaWi, 2019-05-29:
Great project, thank you!
It seems as if, in the last commit, you forgot to commit forecast_record.h, because it is missing when i clone & try to compile the project.

Greetings from Berlin!

televator, 2019-05-29:
I will take this, because the last commit were done by me.
The forecast_record.h is in the src folder. You could copy it from there.
However, I recommend a different approach: just clone the repository to your Arduino libraries directory, open Arduino IDE, check the file->examples dialog and choose your Sketch of interest.

Unfortunately, the Readme does not reflect this new approach as I am currently in discussion with the library maintainer which is the best approach. Sorry about that.

JonasDaWi, 2019-05-29:
ah ok! I'll do it like this! Thank you

closed 2019-05-29 by G6EJD