G6EJD/ESP32-e-Paper-Weather-Display #52: WxConditions change to uppercase letter

hoffmakl1961, 2019-06-05:
In the DisplayForecastTextSection function you use for WxConditions the toLowerCase() function.

e.g. the OWM download for WxConditions  "Klarer Himmel"  (if String Language = "DE";) that is correct and Himmel have a uppercase letter.

G6EJD, 2019-06-05:
SO the issue here is in English 'Klarer Himmel' means 'Clear skies' and the toLowerCase() changes the text to 'klarer himmel' then displays the result as 'Klarer himmel'.
The problem is. does every German word have a capitalised start?  
You need to remove the line that changes all text to Lowwercase, viz:

Delete these lines:
212  WxConditions[0].Main0.toLowerCase();
213  WxConditions[0].Forecast0.toLowerCase();
214  WxConditions[0].Forecast1.toLowerCase();
215  WxConditions[0].Forecast2.toLowerCase();
And change this line:
220 drawStringMaxWidth(x, y-1, 22, TitleCase(Wx_Description),LEFT); // Limit text display to the 22 character wide section, then wrap
To:
220 drawStringMaxWidth(x, y-1, 22,Wx_Description,LEFT); // Limit text display to the 22 character wide section, then wrap

hoffmakl1961, 2019-06-05:
It Works.

closed 2019-06-05 by hoffmakl1961