G6EJD/ESP32-e-Paper-Weather-Display #53: connection lost with last of ..._75_epaper_v11

hoffmakl1961, 2019-06-05:
I downloaded the last release of ESP32_OWM_Current_Forecast_75_epaper_v11 and get a connection lost. 
It works without a "connection lost" >> with an older release of ..._75_epaper_v11 >> downloaded 30-May-2019 

**output serial monitor.**  
Connecting to: WLAN_xxxxxx
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START
.[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 4 - STA_CONNECTED
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:385] _eventCallback(): STA IP: 192.168.xxx.xxx, MASK: 255.255.255.0, GW: 192.168.xxx.1
.WiFi connected at: 2365630656
19Creating object...and  Decoding weather data
13.72
52.56
Clear
Klarer Himmel
01d

30.89
1007.00
33.00
28.89
32.22
6.70
90.00
0
10000
DE
1559702731
1559762513
**connection failed, error: connection lost**
_PowerOff : 3
Awake for : 6911mS
Starting deep-sleep period...

televator, 2019-06-05:
Does this happen sporadically or always?

hoffmakl1961, 2019-06-05:
3x uploaded and tested and 3x conncetion lost

televator, 2019-06-05:
Just to make sure: In the old setup you never had problems?

televator, 2019-06-05:
The error message is produced by the HTTPClient - looking at the source this should really only be produced if the connection was interrupted. 
If possible, you can revert the possibly offending commit with 
```bash
git revert 982b59e390cfac45fb4e55f5a0ff0a5239714366
```

G6EJD, 2019-06-05:
I’m going to test it in a few minutes time to see what’s going on.

hoffmakl1961, 2019-06-05:
> Just to make sure: In the old setup you never had problems?

with the 
old German Version - never (long time test); 
v10 - never (but, only a short time test);
v10a - never (but, only a short time test);
v11 - some times (old release) 
v11 - new release - as described

G6EJD, 2019-06-05:
I've just tested version 11 and see no errors. However, I have seen many times that the OWM server delivers the 'weather' element always but often fails to deliver the 24-hours worth of data in the 'forecast' section, I can see from your copy of the serial port that this is what your seeing, no forecast data.
In code terms when this happens (connection is lost) it is not being handled correctly. Or it is and there will be no data for the forecast. It usually gets cleared after a short time and normal service is resumed.
At the moment I don't have an answer for this issue as it emanates from OWM.

G6EJD, 2019-06-05:
Hang on I'm confused there is no German version in this repository?

televator, 2019-06-05:
From what I understood: he is using the current "international version", but formally used the german version when it still existed.

Note: I used to have "strange" errors with the old versions as well, but there was no proper error message printed to serial. I just saw that the device went to sleep without an update every now and then...
So, I am wondering whether the current version is indeed more problematic than the old ones or whether it just reports more errors...

G6EJD, 2019-06-05:
It is now reporting an error message, where previously it did not, so that's good. The drop out of the 'forecast' section is difficult to replicate, but it does happen a lot. We can see from the logic that it failed to make an HTTP Connection before decoding and then it prints out the error code from the server/HTTPClient so unless the connection was dropped due to time-out (unlikely) it does suggests OWM Server issues.
@hoffmakl1961 if you press RESET on your board and watch the serial port, does it happen every time or occasionally?

G6EJD, 2019-06-05:
I think I'll produce a version12, that includes a revised Battery Estimator; improved font utilisation and importantly adds a message on the top-line if there was a connection issue when fetching the 'forecast' data. I've never seen an error from fetching 'weather'.

G6EJD, 2019-06-05:
@televator I'm sure there was a lang_de.h before?

televator, 2019-06-05:
> @televator I'm sure there was a lang_de.h before?

Afaik, there was only the the "lang_gr.h" that someone posted in a different issue. There is my - pretty untested - [geman branch](https://github.com/televator/ESP32-e-Paper-Weather-Display/tree/german_translations_file) and there used to be the German ino file which we removed as it was stuck on version 3 or so...

G6EJD, 2019-06-05:
I think I know what the issue is, a connection lost occurs when the server takes a few seconds to respond and the connection times-out, the solution is to use HTTP/1.1 and in earlier versions I was using HTTP/1.1 in some latter versions it was changed back to HTTP/1.0.

Reason for HTTP/1.1:  Keep-alive connections are enabled by default in HTTP/1.1 while not in HTTP/1.0. HTTP/1.0 

The problem now is what HTTP version is HTTPClient using or how can be set to 1.1?

televator, 2019-06-05:
HTTPClient uses Http/1.1 as default, see https://github.com/espressif/arduino-esp32/blob/7d7824701fe5e22f08555d3e1ce3180a922b2151/libraries/HTTPClient/src/HTTPClient.h#L222
You can change to v1.0 with useHTTP10()

Small remark, seems there was a commut just 9h ago "fixing" reuse of the connection; hopefully that is not a regression.

G6EJD, 2019-06-05:
I was inspecting the same file - and you beat me to it :) 

So we have established it's probably not a keep-alive issue. Could be an internet connection issue and nothing it wrong with the code at all - that's life sort of thing.

I can go for days without any issues, then I get a spate of them. My theory is the OWM system periodically updates it's forecasts and during that update the forecast data is either not available and this results in a lost connection., it should not as HTTP/1.1 should keep the connection live until the data is available. 

I've added some response code printing to my version here and I'll see if I can capture the code returned, well I know it's -5 but there might be some more information around it.

Thanks for your assistance with this issue.

hoffmakl1961, 2019-06-06:
I added also a line and can capture the http return code. 

`Serial.printf("\rhttp request to OWM with RC: %d\n\r", httpCode);`

**the result is**
http request to OWM with RC: -5

I can't find a decription of a http return code -5. 
The same result after reset.

G6EJD, 2019-06-06:
HTTPC_ERROR_CONNECTION_LOST     (-5)

My version is now logging error codes but for the last ~12 hours each response has been OK or 200. So, we must establish what’s different.
Can you go to Boards, Boards Manager, enter ESP in the search box, is your ESP32 core 1.2 or above?
Does OWM provide a forecast for your location? Please advise where you are or what location your using for the request? OWM does not provide a forecast for locations outside its list of locations and I can test the response from here. Thank you.

hoffmakl1961, 2019-06-06:
String City          = "Altlandsberg";   
String Country       = "DE";             
String Language      = "DE";         

esp32 by Espressif Systems Version 1.0.1 INSTALLED  (max. is 1.0.2, 1.2 is not available

G6EJD, 2019-06-06:
Thank you. Response from OWM is instant with those parameters, I'm running a 1-min update test, no failures reported so-far:
"cod":"200","message":0.0069,"cnt":24,"list":[{"dt":1559822400,"main":{"temp":29.95,"temp_min":29.95,"temp_max":29.95,"pressure":1007.49,"sea_level":1007.49,"grnd_level":1001.27,"humidity":39,"temp_kf":0},"weather":[{"id":800,"main":"Clear","description":"clear sky","icon":"01d"}],"clouds":{"all":1},"wind":{"speed":4.1,"deg":205.427},"sys":{"pod":"d"},"dt_txt":"2019-06-06 12:00:00"},{"dt":1559833200,"main":{"temp":23.25,"temp_min":23.25,"temp_max":23.25,"pressure":1009.66,"sea_level":1009.66,"grnd_level":1004.25,"humidity":65,"temp_kf":0},"weather":[{"id":501,"main":"Rain","description":"moderate rain","icon":"10d"}],"clouds":{"all":71},"wind":{"speed":7.11,"deg":293.936},"rain":{"3h":3.25},"sys":{"pod":"d"},"dt_txt":"2019-06-06 15:00:00"},{"dt":1559844000,"main":{"temp":19.85,"temp_min":19.85,"temp_max":19.85,"pressure":1011.81,"sea_level":1011.81,"grnd_level":1006.44,"humidity":70,"temp_kf":0},"weather":[{"id":500,"main":"Rain","description":"light rain","icon":"10d"}],"clouds":{"all":86},"wind":{"speed":5.7,"deg":309.665},"rain":{"3h":0.938},"sys":{"pod":"d"},"dt_txt":"2019-06-06 18:00:00"},{"dt":1559854800,"main":{"temp":17.45,"temp_min":17.45,"temp_max":17.45,"pressure":1013.94,"sea_level":1013.94,"grnd_level":1008.57,"humidity":70,"temp_kf":0},"weather":[{"id":500,"main":"Rain","description":"light rain","icon":"10n"}],"clouds":{"all":100},"wind":{"speed":4.99,"deg":324.38},"rain":{"3h":0.375},"sys":{"pod":"n"},"dt_txt":"2019-06-06 21:00:00"},{"dt":1559865600,"main":{"temp":15.85,"temp_min":15.85,"temp_max":15.85,"pressure":1015.61,"sea_level":1015.61,"grnd_level":1010.28,"humidity":70,"temp_kf":0},"weather":[{"id":500,"main":"Rain","description":"light rain","icon":"10n"}],"clouds":{"all":100},"wind":{"speed":3.79,"deg":317.484},"rain":{"3h":0.25},"sys":{"pod":"n"},"dt_txt":"2019-06-07 00:00:00"},{"dt":1559876400,"main":{"temp":14.35,"temp_min":14.35,"temp_max":14.35,"pressure":1016.52,"sea_level":1016.52,"grnd_level":1010.66,"humidity":71,"temp_kf":0},"weather":[{"id":804,"main":"Clouds","description":"overcast clouds","icon":"04d"}],"clouds":{"all":100},"wind":{"speed":3.05,"deg":332.059},"sys":{"pod":"d"},"dt_txt":"2019-06-07 03:00:00"},{"dt":1559887200,"main":{"temp":15.64,"temp_min":15.64,"temp_max":15.64,"pressure":1017.21,"sea_level":1017.21,"grnd_level":1011.43,"humidity":65,"temp_kf":0},"weather":[{"id":803,"main":"Clouds","description":"broken clouds","icon":"04d"}],"clouds":{"all":66},"wind":{"speed":2.2,"deg":336.873},"sys":{"pod":"d"},"dt_txt":"2019-06-07 06:00:00"},{"dt":1559898000,"main":{"temp":21.25,"temp_min":21.25,"temp_max":21.25,"pressure":1016.51,"sea_level":1016.51,"grnd_level":1010.78,"humidity":43,"temp_kf":0},"weather":[{"id":800,"main":"Clear","description":"clear sky","icon":"01d"}],"clouds":{"all":0},"wind":{"speed":1.2,"deg":6.293},"sys":{"pod":"d"},"dt_txt":"2019-06-07 09:00:00"},{"dt":1559908800,"main":{"temp":26.05,"temp_min":26.05,"temp_max":26.05,"pressure":1015.27,"sea_level":1015.27,"grnd_level":1009.52,"humidity":30,"temp_kf":0},"weather":[{"id":800,"main":"Clear","description":"clear sky","icon":"01d"}],"clouds":{"all":0},"wind":{"speed":1.11,"deg":86.533},"sys":{"pod":"d"},"dt_txt":"2019-06-07 12:00:00"},{"dt":1559919600,"main":{"temp":26.75,"temp_min":26.75,"temp_max":26.75,"pressure":1014.2,"sea_level":1014.2,"grnd_level":1008.01,"humidity":30,"temp_kf":0},"weather":[{"id":800,"main":"Clear","description":"clear sky","icon":"01d"}],"clouds":{"all":0},"wind":{"speed":2.12,"deg":111.721},"sys":{"pod":"d"},"dt_txt":"2019-06-07 15:00:00"},{"dt":1559930400,"main":{"temp":25.05,"temp_min":25.05,"temp_max":25.05,"pressure":1013.21,"sea_level":1013.21,"grnd_level":1007.48,"humidity":40,"temp_kf":0},"weather":[{"id":800,"main":"Clear","description":"clear sky","icon":"01d"}],"clouds":{"all":0},"wind":{"speed":3.43,"deg":106.276},"sys":{"pod":"d"},"dt_txt":"2019-06-07 18:00:00"},{"dt":1559941200,"main":{"temp":21.45,"temp_min":21.45,"temp_max":21.45,"pressure":1014.35,"sea_level":1014.35,"grnd_level":1008.78,"humidity":57,"temp_kf":0},"weather":[{"id":800,"main":"Clear","description":"clear sky","icon":"01n"}],"clouds":{"all":0},"wind":{"speed":3.63,"deg":98.938},"sys":{"pod":"n"},"dt_txt":"2019-06-07 21:00:00"},{"dt":1559952000,"main":{"temp":20.15,"temp_min":20.15,"temp_max":20.15,"pressure":1015.52,"sea_level":1015.52,"grnd_level":1009.36,"humidity":64,"temp_kf":0},"weather":[{"id":802,"main":"Clouds","description":"scattered clouds","icon":"03n"}],"clouds":{"all":38},"wind":{"speed":3.24,"deg":161.286},"sys":{"pod":"n"},"dt_txt":"2019-06-08 00:00:00"},{"dt":1559962800,"main":{"temp":16.75,"temp_min":16.75,"temp_max":16.75,"pressure":1017.66,"sea_level":1017.66,"grnd_level":1012.48,"humidity":64,"temp_kf":0},"weather":[{"id":500,"main":"Rain","description":"light rain","icon":"10d"}],"clouds":{"all":9},"wind":{"speed":6.07,"deg":282.703},"rain":{"3h":0.188},"sys":{"pod":"d"},"dt_txt":"2019-06-08 03:00:00"},{"dt":1559973600,"main":{"temp":16.38,"temp_min":16.38,"temp_max":16.38,"pressure":1020.66,"sea_level":1020.66,"grnd_level":1015.46,"humidity":61,"temp_kf":0},"weather":[{"id":801,"main":"Clouds","description":"few clouds","icon":"02d"}],"clouds":{"all":24},"wind":{"speed":4.88,"deg":266.77},"sys":{"pod":"d"},"dt_txt":"2019-06-08 06:00:00"},{"dt":1559984400,"main":{"temp":17.15,"temp_min":17.15,"temp_max":17.15,"pressure":1022.75," [... cut ...] 19-06-09 06:00:00"},{"dt":1560070800,"main":{"temp":23.55,"temp_min":23.55,"temp_max":23.55,"pressure":1026.2,"sea_level":1026.2,"grnd_level":1020.69,"humidity":34,"temp_kf":0},"weather":[{"id":802,"main":"Clouds","description":"scattered clouds","icon":"03d"}],"clouds":{"all":33},"wind":{"speed":4.34,"deg":150.478},"sys":{"pod":"d"},"dt_txt":"2019-06-09 09:00:00"}],"city":{"id":2957051,"name":"Altlandsberg","coord":{"lat":52.5615,"lon":13.7246},"country":"DE","population":8804,"timezone":7200}}

hoffmakl1961, 2019-06-06:
that is the output from a Web-browser. and with V.11?

G6EJD, 2019-06-06:
Yes, output from using a web browser and v11, I used:

http://api.openweathermap.org/data/2.5/forecast?q=Altlandsberg,DE&APPID=your_OWM_API_key&mode=json&units=metric&cnt=24

Put your key in and you can simulate the connection using your browser address line.

So-far after 23-mins of updates no errors.
It's starting to look like a connection issue from your location to the OWM servers; an internet  service provider issue, data size is comparably low though.  I'll keep mine running for now.

hoffmakl1961, 2019-06-06:
The output in firefox is ok.

cod | "200"
-- | --
message | 0.0072
cnt | 24
list | […]
city | {…}

no problems
-----------------------------------------------------------------------------
I 've reinstall the V11 and tested,  >> connection lost. 
if I use v.10a - no problem. 
-----------------------------------------------------------------------------
Let's stop here.

G6EJD, 2019-06-06:
HTTPClient then, needs further investigation.

televator, 2019-06-06:
Currently v10a and v11 are using the same implementation with HTTPClient... Question is whether @hoffmakl1961 talks about a recent v10a or an older one...

hoffmakl1961, 2019-06-06:
10a

televator, 2019-06-06:
Recently downloaded?

hoffmakl1961, 2019-06-06:
download from 26-05-2019

televator, 2019-06-06:
Just to be sure that this is not a weird side effect: can you give the 10a ino of the downloaded zip where the problematic 11 was in a try?

G6EJD, 2019-06-06:
I’ve been monitoring the Altlandsberg weather all day now using v11 and no errors to report. If more locally you are getting drop-outs you should see the effect using a browser request where you can repeat that quickly and where the error should be evident.
You you let us know the board type your using?
What sort of power source are you using? the reason is, it’s a long burst of high current activity by the ESP32 and the PSU May not be sufficient.

hoffmakl1961, 2019-06-06:
PSU = by PC USB;
Board =  NodeMCU ESP32; 
V10a - download from 26-05-2019 runs without a connection lost since 4 hours. 
In the monitor - all required receiving forecast periods are available. 

> can you give the 10a ino of the downloaded zip where the problematic 11 was in a 

> try?

???

G6EJD, 2019-06-06:
OK, I’ll give 10a a trial period

televator, 2019-06-06:
I doubt this will enlighten us; the 10a is not too different.
@hoffmakl1961 compares the 10a of 26-05 to the 11 of 30-05. In this period the HTTPClient was introduced, so this seems to be problematic with @hoffmakl1961's setup (whatever could be different from our setup). I suggest to close this as "not reproducable" for the time being and wait if someone else comes up with a similar observation...

G6EJD, 2019-06-06:
Agreed, but I’ll leave my 7.5” 10a setup running overnight and if there are problems I’ll report back.

closed 2019-06-06 by G6EJD

hoffmakl1961, 2019-06-07:
**a long term test with Version 10a**
after 13 deep sleeps , no OMW data will be downloaded. 
**monitor output**
07:35:16.527 -> rst:0x5 (DEEPSLEEP_RESET),boot:0x16 (SPI_FAST_FLASH_BOOT)
07:35:16.527 -> configsip: 0, SPIWP:0xee
07:35:16.527 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
07:35:16.527 -> mode:DIO, clock div:1
07:35:16.527 -> load:0x3fff0018,len:4
07:35:16.527 -> load:0x3fff001c,len:1100
07:35:16.527 -> load:0x40078000,len:10088
07:35:16.527 -> load:0x40080400,len:6380
07:35:16.527 -> entry 0x400806a4
07:35:16.809 -> 
07:35:16.809 -> Connecting to: WLAN
07:35:16.856 -> [D][WiFiGeneric.cpp:342] _eventCallback(): Event: 0 - WIFI_READY
07:35:16.856 -> [D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START
07:35:17.371 -> .[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 4 - STA_CONNECTED
07:35:17.793 -> [D][WiFiGeneric.cpp:342] _eventCallback(): Event: 7 - STA_GOT_IP
07:35:17.793 -> [D][WiFiGeneric.cpp:385] _eventCallback(): STA IP: 192.168.xxx.xxx, MASK: 255.255.255.0, GW: 192.168.xxx.1
07:35:17.840 -> .WiFi connected at: 2365630656
07:35:17.887 -> 07_PowerOff : 2
07:35:18.158 -> Awake for : 1368mS
07:35:18.192 -> Starting deep-sleep period...

Test with firefox >> 1 sec later, download of OWM data >> ok. 

If the problem only occurs to me, it's probably a hardware problem
Right now I  don't have other hardware (ESP) to test. 

I would like to thank you for your support and close the issue.

televator, 2019-06-07:
Now I am confused. Was your long time test with the 10a downloaded at 26-05-2019 which used to be working? (Looking at the output: yes, it is the old version; there is now proper error message)

So "after 13 deep sleeps" means: it worked for 13 times but then started to discontinue operation for at least one or more turns? 

That would be something I experienced myself as well - random connection issues with OWM. Not sure where they come from (owm, wifi power spikes causing resets, timeouts somewhere...)

G6EJD, 2019-06-07:
@hoffmakl1961 it is likely that v10a includes a time fence that does not refresh the data outside the hours of 08:00 and 23:00, if you check the code at the early stages of void reset() you may see: 
if wakeup_hour >= 8 || wakeup _hour <= 23
I added this to save battery power. Please check your code version, I think this is the cause of no connection, which is by design.

hoffmakl1961, 2019-06-07:
if (wakeup_hour >= 8 && wakeup_hour <= 23) {
    Received_WxData_OK = (obtain_wx_data("weather") && obtain_wx_data("forecast"));
  }
works as designed.

G6EJD, 2019-06-07:
Because your recent log showed the time as < 08:00 and the total on-tine was less than 2-secs then your ESP32 woke up, then immediately went back to sleep which was correct and there would have been no update. You could either remove those lines or add a test to the line if (>=8 and <=23 || forcedWake) and make forcedWake = true.

hoffmakl1961, 2019-06-07:
Sorry, one addtional question to V12: 
if I compile the V12, I get the following message: 'client' was not declared in this scope

ESP32_OWM_Current_Forecast_75_epaper_V12:98:25: error: 'client' was not declared in this scope
     if ((obtain_wx_data(client, "weather") && obtain_wx_data(client, "forecast"))) {
                         ^

G6EJD, 2019-06-07:
My mistake, I moved: WiFiClient client; // wifi client object to the common.h file

I've amended v12, you can add this to say line88:
WiFiClient client; // wifi client object

This resolves that error.

G6EJD, 2019-06-10:
in common.h I've added http.end() art the end of the file;

  http.end();
  return true;
}

This has improved reliability because after the first http request for weather it was not closed, then when the second attempt for 'forecast' was attempted, it failed with 'connection lost' now after 2-days of running since modification there have been no failures, which I have been getting.

hoffmakl1961, 2019-06-10:
I changed the ESP32 and add the http.end(); line, but the result is the same
20:46:40.809 -> connection failed, error: connection lost
20:46:50.799 -> Busy Timeout!
20:46:50.799 -> _PowerOff : 10001023
20:46:50.799 -> Awake for : 17442mS

The problem was after the code separation in common.xxx. 

The following was the last code without connection lost.
[ESP32_OWM_Current_Forecast_75_epaper_v10a.zip](https://github.com/G6EJD/ESP32-e-Paper-Weather-Display/files/3273411/ESP32_OWM_Current_Forecast_75_epaper_v10a.zip)

I'm probably a problem with my IDE environment.

televator, 2019-06-10:
@G6EJD concerning http.end(): did you experience any connection issues as well? I am wondering because you mentioned that this improves reliability, and it would be great for debugging if you saw the same connection dropouts...
Having client.stop() at the beginning

G6EJD, 2019-06-10:
**@televator**, yes I have been seeing a few failures of the 'forecast' section, but never weather. So, I read the Espressif documentation on HTTPClient and they recommend http.end(); to close an http client and so-far over the last 3-days now, I have not seen a failed forecast section. I was thinking that when the second attempt is made (for the forecast) the current httpclient session is still open and this might be causing the error.
I'm using ESP32 1.0.3-rc1 and 1.8.9 IDE

G6EJD, 2019-06-10:
**@hoffmakl1961** I forgot to say, your system is awake for a comparatively long time, my v21 updates take about 4-sesc. Yours at 17.4-secs are long, this suggests there is a connection issue.
20:46:50.799 -> Awake for : 17442mS

What IDE and ESP32 core are you using (Board, Board Manager ESP32 in search box)

G6EJD, 2019-06-11:
I’ve had another day without connection issues since adding http.end(); 
Interestingly, I’ve had two emails today from users of version 12 showing a serial port output of connection lost for the forecast, so I’ve advised them to move to 1.0.2 or ideally 1.0.3-rc1

hoffmakl1961, 2019-06-11:
I‘ll give you the informationen asap.
In the webbrowser, the answer to the http request was faster.

hoffmakl1961, 2019-06-14:
I'm using the arduino IDE 1.8.9 and esp32 version 1.0.2.
I'll not invest time to change my IDE Arduino environment. 
In the future I want to change to platformIO. 

via webbrowser, I get the OWM data very fast about 1 sec. 
http://api.openweathermap.org/data/2.5/forecast?q=Altlandsberg,DE&APPID=my_OWM_API_key&mode=json&units=metric&cnt=24

serial monitor:  with version 13
....
21:02:34.326 -> Period-23--------------
21:02:34.326 -> Temp: 25.15
21:02:34.326 -> TLow: 25.15
21:02:34.326 -> THig: 25.15
21:02:34.326 -> Pres: 1019.97
21:02:34.326 -> Humi: 45.00
21:02:34.326 -> For0: Clouds
21:02:34.326 -> For1: 
21:02:34.326 -> For2: 
21:02:34.326 -> Icon: 04d
21:02:34.326 -> Desc: Überwiegend bewölkt
21:02:34.326 -> CCov: 60
21:02:34.326 -> WSpd: 3.48
21:02:34.326 -> WDir: 88.07
21:02:34.326 -> Rain: 0.00
21:02:34.326 -> Snow: 0.00
21:02:34.326 -> Peri: 2019-06-17 18:00:00
21:02:34.373 -> Icon name: 01d
21:02:34.373 -> Icon name: 10n
21:02:34.373 -> Icon name: 04n
21:02:34.373 -> Icon name: 04d
21:02:34.373 -> Icon name: 04d
21:02:34.373 -> Icon name: 04d
21:02:34.373 -> Icon name: 03d
21:02:34.373 -> Icon name: 10d
21:02:34.373 -> Icon name: 04d
**21:02:44.401 -> Busy Timeout!**
21:02:44.401 -> _PowerOn : 10001028
**21:02:56.767 -> Busy Timeout!**
21:02:56.767 -> _Update_Full : 10001016
**21:03:06.809 -> Busy Timeout!**
21:03:06.809 -> _PowerOff : 10001015
21:03:06.809 -> Awake for : 34.961-secs
21:03:06.809 -> Starting deep-sleep period...

I get 3x busy timeout.

G6EJD, 2019-06-14:
Are you getting ### failed to receive forecast data ### in the centre of the screen.

Recall you had some wiring issues, well it looks like busy is not connected or on another pin, that’s why the time-out is occurring. If you get a DVM on continuity mode (resistance) and test between screen BUSY and ESP busy pin being used there should be 0-ohms / continuity

hoffmakl1961, 2019-06-14:
The test was without a display.
That is the reason for the timeouts.

G6EJD, 2019-06-14:
I’ve spent some time working on connection reliability and timing, improved the WiFi connection and time setup function, plus I’ve added a few features like cloud cover and RH%, plus now if there’s a connection problem it reports it.

hoffmakl1961, 2019-06-14:
no connection lost, great job!!!
thank you.

G6EJD, 2019-06-14:
Time will tell😀

I noticed at 1510 your time today, that the OWM servers were unusually providing forecast data but no weather data, the changes now show those event on screen in the weather forecast area, which drops to a smaller font when text gets too long to fit. It no longer changes to lowercase too when language is DE. You will find the weather forecast text is in 4 groups ‘Main forecast (forecast0 & forecast 1 & forecast2) because OWM gives an area forecast so for me it’s not unusual today to see Rain (light rain & drizzle & fog) this reflects the conditions across my town. Ironically it’s my weather station that provides the main forecast 😳 for this area.