G6EJD/ESP32-e-Paper-Weather-Display #54: Any method to make it work on colour 7.5 inch display with waveshare esp32 board,?plz

drvjs, 2019-06-11:
I am wondering if your project can work on my new 7.5 inch e ink display with waveshare esp32 board.plz help me.

G6EJD, 2019-06-11:
Yes it will, just change the connections to like this:
// E-Paper pins to ESP32 GPIO pins e.g. Waveshare ESP32 driver board:
static const uint8_t EPD_BUSY  = 25;
static const uint8_t EPD_CS      = 15;
static const uint8_t EPD_RST    = 26; 
static const uint8_t EPD_DC     = 27;
static const uint8_t EPD_SCK    = 13;
static const uint8_t EPD_MISO = 19; // Master-In Slave-Out not used, as no data from display
static const uint8_t EPD_MOSI = 14;

closed 2019-06-11 by G6EJD

drvjs, 2019-06-11:
Is there any need to  edit EPD_WaveShare_75.cpp as you mentioned earlier? Because I m not getting any output and my serial monitor keep getting connection closed error in any sketch from v 10a to 12

G6EJD, 2019-06-11:
Have you downloaded the ZIP? Then installed the library, then modified the credentials tab with your WiFi details and API key from OpenWeatherMap (it’s free).
What is the serial port reporting?

G6EJD, 2019-06-11:
You need to modify the v12 code where the pins are defined as I replied earlier.

drvjs, 2019-06-11:
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1100
load:0x40078000,len:9232
load:0x40080400,len:6400
entry 0x400806a8

Connecting to: Dr.Vikramjit Singh
...WiFi connected at: 3154225344

drvjs, 2019-06-11:
Creating object...and  Decoding weather data
74.75
30.67
Clear
clear sky
01n

42.05
990.44
8.00
42.05
42.05
4.12
212.43
0
0
IN
1560211069
1560261807
connection failed, error: connection lost
_PowerOff : 3
Awake for : 8305mS
Starting deep-sleep period...

drvjs, 2019-06-11:
above are my serial port reporting .plz help me ,i m not able to get any display output.

drvjs, 2019-06-11:
gxepd2 ws esp32 driver example work fine on my display

G6EJD, 2019-06-11:
Which version of the code are you using?
So what’s happening is the first stage is complete that’s called the ‘weather’ request then it attempts the second stage called ‘forecast’ and the connection failed. Please ensure your using ESP32 core 1.0.2.

If your getting no display have you modified the connections st the top / start of the v12 code to match your board type. This code uses EPD2.

G6EJD, 2019-06-11:
Can you give me your city name please so I can test the OWM response, sometimes they don’t provide data for all locations

slavik0329, 2019-07-15:
Heres how!

Change pins to this: 

```
static const uint8_t EPD_BUSY = 25;
static const uint8_t EPD_CS   = 15;
static const uint8_t EPD_RST  = 26; // Lolin D32 Pro pin N/A, so suggest using 12
static const uint8_t EPD_DC   = 27; // Lolin D32 Pro pin N/A, so suggest using 13
static const uint8_t EPD_SCK  = 13;
static const uint8_t EPD_MISO = 19; // Master-In Slave-Out not used, as no data from display
static const uint8_t EPD_MOSI = 14;
```

Then change:
`GxEPD2_BW<GxEPD2_750, GxEPD2_750::HEIGHT> display(GxEPD2_750(/*CS=5*/ EPD_CS, /*DC=*/ EPD_DC, /*RST=*/ EPD_RST, /*BUSY=*/ EPD_BUSY));
`
To:
`GxEPD2_3C<GxEPD2_750, GxEPD2_750::HEIGHT> display(GxEPD2_750(/*CS=5*/ EPD_CS, /*DC=*/ EPD_DC, /*RST=*/ EPD_RST, /*BUSY=*/ EPD_BUSY));
`

Then in the `InitialiseDisplay` function right below `display.init(115200);` Insert:

```
SPI.end(); // release standard SPI pins, e.g. SCK(18), MISO(19), MOSI(23), SS(5)
SPI.begin(13, 12, 14, 15); // map and init SPI pins SCK(13), MISO(12), MOSI(14), SS(15)
```

Upload and it should work!

G6EJD, 2019-07-16:
Thank you, I have combined these connections into the code examples. No need to stop and start SPI, the driver does that.

slavik0329, 2019-07-16:
Turns out with the esp32 spi driver board you do need to stop and start spi
or it won’t work.

G6EJD, 2019-07-16:
I’ve just tried it with my Waveshare board and it works as expected with no need for SPI.begin(MOSI, MISO, etc