G6EJD/ESP32-e-Paper-Weather-Display #55: Some minor issues with version 13

AWSW2017, 2019-06-19:
Hello G6EJD,

first of all: Thank you very much for this project and all the hard work you put into it. :)

I updated yesterday from v8 to v13 and found some minor issues which I like to report to you, just to let you know:

- The "owm_credentials.h" file was missing in the v13 folder, so I put it there. Maybe it would be good to store this in the src-folder where the rest of the common files are?

- The file "ESP32_OWM_Current_Forecast_75_epaper_v13.ino" contains the wifi settings as "ssid" and "password". The "owm_credentials.h" file contains them as "ssid1" and "password1". So a small change here to perform.

- I could not find the file "lang_gr.h" for localization into German, so I will put you mine here, if you like to overtake it? I am still searching for the replacement chars to support the in German often used umlauts like "ä, ü, ö, ß". Actually I used "ae, ue, oe, ss" instead for now.
[lang_gr.h.zip](https://github.com/G6EJD/ESP32-e-Paper-Weather-Display/files/3305251/lang_gr.h.zip)

- If you don't mind I would like to integrate my version of this IOT installation method to add the wifi credentials on first startup called Basecamp. Link: https://github.com/merlinschumacher/Basecamp
I used and enhanced it for some other projects I use in the below showed frames. I want to give these frames away for some friends and family as present and then they can setup their wifi connection on first startup themselves. ;)
If you like I can give it a shot and you have a look afterwards if you like it of not?

Also I have a little question: Why is the condition text mixed as "Clear (Klarer Himmel)". I think this comes from open weather map data and is not translated there, right? I set all settings to "DE" in the "owm_credentials.h" file...
[image]

Thanks a lot and have a nice day. :) 8) 

AWSW

G6EJD, 2019-06-19:
Thanks for your help, I have just uploaded v14, which contains more enhancements, fixes and improvements in error detection. I also included the credentials file and removed the ssid1 errors.
I'll upload the German file later, that's for that, much appreciated.

G6EJD, 2019-06-19:
BTW, that's a really nice solution you have created and the frame looks really good.

closed 2019-06-19 by G6EJD

AWSW2017, 2019-06-19:
Thanks a lot and have a nice day :)

hoffmakl1961, 2019-06-19:
@ AWSW2017: You wrote that you currently used "ae, ue, oe, ss" etc. 
How do you change the OWM text information such as 
[image]

@G6EJD - The used lib GxEPD2 provide also an example:  GxEPD2_U8G2_Fonts_Example with 8-bit fonts.
[image]

Is it possible to change the fonts?

G6EJD, 2019-06-19:
OWM only partially translates weather text as you've seen, another quirk is that air pressure is not converted to inches when imperial format is requested. I can add these characters to the font if you tell me what the translation is. For example, I can add new characters to the font if there enough room, for example there are some characters that never get used *&^%$£, but what I don't understand is in a standard 127 character font how umlauts are treated / selected as there is a for example a 'e' and 'é' but the latter is usually an escaped character beyond 128 in that example 233 in the ASCII table. Do OWM send those characters?

As the range of forecast text is quite small, it would be fairly easy to translate them for example:
if (...forecatst0 == "Clear" ) ..Forecast0 = "klar";

My natural tendency would be to make that Klar but I understand that can change the meaning?

Unfortunately, the API does not list the forecast text, so the dictionary would need to be built up over time.

So-far we have:
Clear - klar
Sunny - sonnig
Mist - Nebel
Fog - Nebel
Rain - Regen
Shower - Dusche
Cloudy - Wolkig
Drizzle - Nieseln
Snow - Schnee
Thunder - Donner
Light - Licht
Heavy - Schwer

Do they work?

hoffmakl1961, 2019-06-19:
Umlauts are not included in a 7-bit acsii table.
It would be great, if you can add the following new characters to the font:

G6EJD, 2019-06-19:
I’ll see what can be achieved the Adafruit fonts have a particular format. May take a little while but it would be great to get this done.

hoffmakl1961, 2019-06-19:
**a forecast text dictionary**
Clear - klar
Sunny - sonnig
Mist - Nebel
Fog - Nebel
Rain - Regen
Shower - Regenschauer
Cloudy - wolkig
Drizzle - Nieselregen 
Snow - Schnee
Thunderstorm - Gewitter
Light - leichter
Heavy - schwer

**forecast text for the next 3 days** 
mostly cloudy - grösstenteils bewölkt
overcast clouds - überwiegend bewölkt
scattered clouds - aufgelockerte Bewölkung
few clouds - ein paar Wolken
clear sky - klarer Himmel
broken clouds - aufgerissene Bewölkung
light rain - leichter Regen

hoffmakl1961, 2019-06-19:
In the ForecastTextSection will be displayed the engl. and german text.
[image]

G6EJD, 2019-06-19:
I see so, the weather yet is in English and the forecast has been translated - how strange is that. The text is as received from OWM. So we only need to convert Weather text.

AWSW2017, 2019-06-19:
Good evening :) You guys are great! 
I'm looking forward how this can be solved. For the moment I can live without the umlauts very well.
The Forecast section looks a bit strange. Looks the same at the moment for me here "Rain (Regenschauer)"... Thanks in advance :)

G6EJD, 2019-06-19:
I have it working with ug82_for_Adafruit_gfx, just needs some suitable font selections.
I chnaged the text for Rain to Rügen just to see that it was working OK, it was. I'll upload it tomorrow.

[image]

hoffmakl1961, 2019-06-20:
really cool, thank you.
----------------------------------------------------
I didn't understand the meaning of the OWM data main: 
I would assume "main:" is only a gerneral keyword and the relevant information are in "description".
That could be why only description is being translated. 

here the current OWM data (via firefox) - filtered by main and description 

main	"Clouds"	description	"Überwiegend bewölkt"
main	"Clouds"	description	"Überwiegend bewölkt"
main	"Clouds"	description	"Überwiegend bewölkt"
main	"Rain"		description	"Leichter Regen"
main	"Rain"		description	"Leichter Regen"
main	"Clouds"	description	"Bedeckt"
main	"Clouds"	description	"Überwiegend bewölkt"
main	"Clouds"	description	"Überwiegend bewölkt"
main	"Clouds"	description	"Überwiegend bewölkt"
main	"Clouds"	description	"Überwiegend bewölkt"
main	"Clouds"	description	"Überwiegend bewölkt"
main	"Clouds"	description	"Überwiegend bewölkt"
main	"Clouds"	description	"Ein paar Wolken"
main	"Clear"		description	"Klarer Himmel"
main	"Clear"		description	"Klarer Himmel"
main	"Clear"		description	"Klarer Himmel"
main	"Clear"		description	"Klarer Himmel"
main	"Clear"		description	"Klarer Himmel"
main	"Clouds"	description	"Überwiegend bewölkt"
main	"Clouds"	description	"Bedeckt"
main	"Clouds"	description	"Bedeckt"
main	"Clouds"	description	"Bedeckt"
main	"Clouds"	description	"Bedeckt"
main	"Clouds"	description	"Überwiegend bewölkt"

In the ForecastTextSection function, I would remove the WxConditions[0].Main, "(" and ")".

G6EJD, 2019-06-20:
Thanks for the help, a good team effort. I nearly have the conversion complete. The font file is huge, so I’m copying selected fonts over.
The (problem with the) OWM weather description API returns a Main forecast say Rain then sometimes supplements this in up to 3 area forecasts (0,1,2) they do this to show different weather conditions in an area, examples:
Main0: Rain
Forecast0: Null
Forecast1: Null
Forecast2: Null

Main0: Rain
Forecast0: Light drizzle
Forecast1: Null
Forecast2: Null

Main0: Rain
Forecast0: Light drizzle
Forecast1: Mist
Forecast2: Null

So, it’s:

Weather (area report, area report, area report)

If we omit Main then more often than not Forecast0,1,2 are null and so there would be nothing displayed. Conversely why would OWM only translate Forecast0,1,2 and not Main, I’ve never understood why. Maybe we’ll try it as just forecast text and see what happens!

G6EJD, 2019-06-20:
V15 has been uploaded, it's working for me in DE, I'm seeing accented characters.
Today I've seen Main0 weather and no Forecast0,1,2, data, so it does need to translate.

You need to install Sketch,  Include, Manage then U8g2_for_Adafruit_GFX
NOTE: U8g2_for_Adafruit_GFX was updated today as there was an issue, now fixed when I tried to compile with 1.0.3rc-1 there was an uninitialized variable in the library, then failed compilation.

AWSW2017, 2019-06-20:
Hello you busy ones :)

Thanks a lot! I just updated to V15 and it looks good to me. 

I made some changes in the lang_gr.h file because the German texts are (usually) to long to fit into the boxes on the screen. The new font allows to change the texts to make them fit better. Great work. :)
Attached you will find the new file: 
[lang_gr.h.zip](https://github.com/G6EJD/ESP32-e-Paper-Weather-Display/files/3310440/lang_gr.h.zip)

I also made a small change in the "ESP32_OWM_Current_Forecast_75_epaper_v15.ino" because the now translated texts were all set to lowercase which in my tests did not fit the grammar all the time... The original lines are commented out to make it hopefully more easy to compare... So I made a change here:

`void DisplayForecastTextSection(int x, int y) {
  if (Language == "DE") { 
    /*WxConditions[0].Main0.toLowerCase();     WxConditions[0].Main0     = Translate_EN_DE(WxConditions[0].Main0);
    WxConditions[0].Forecast0.toLowerCase(); WxConditions[0].Forecast0 = Translate_EN_DE(WxConditions[0].Forecast0);
    WxConditions[0].Forecast1.toLowerCase(); WxConditions[0].Forecast1 = Translate_EN_DE(WxConditions[0].Forecast1);
    WxConditions[0].Forecast2.toLowerCase(); WxConditions[0].Forecast2 = Translate_EN_DE(WxConditions[0].Forecast2);*/
    WxConditions[0].Main0.toLowerCase();     WxConditions[0].Main0     = Translate_EN_DE(WxConditions[0].Main0);
    WxConditions[0].Forecast0; WxConditions[0].Forecast0 = Translate_EN_DE(WxConditions[0].Forecast0);
    WxConditions[0].Forecast1; WxConditions[0].Forecast1 = Translate_EN_DE(WxConditions[0].Forecast1);
    WxConditions[0].Forecast2; WxConditions[0].Forecast2 = Translate_EN_DE(WxConditions[0].Forecast2);
  }`

Thanks a lot for the changes and have a nice day.

G6EJD, 2019-06-20:
Yes, I’ve just been looking at that, because the OWM main0 text is not translated it needs lowercase on just that but not the translated forecast text, this will retain the grammar. I’m going to move the translated lines to the ‘lang_cc.h’ files as this then will allow more flexibility. The good thing is this change to u8g2 fonts enables all accented hard tees to now be displayed, so I’m converting the smaller screen versions too.

hoffmakl1961, 2019-06-20:
thank you for V15, I'll test it asap. 

I'm sorry I still don#t get it. 
If I look at the OWM data, each main have one description.
{"dt":1561042800,"main":{"temp":25.12,"temp_min":25.12,"temp_max":25.9,"pressure":1009.74,"sea_level":1009.74,"grnd_level":1003.9,"humidity":51,"temp_kf":-0.78},"weather":[{"id":500,**"main":"Rain","description":"Leichter Regen"**,"icon":"10d"}],"clouds":{"all":79},"wind":{"speed":4.31,"deg":279.862},"rain":{"3h":2.5},"sys":{"pod":"d"},"dt_txt":"2019-06-20 15:00:00"},

{"dt":1561053600,"main":{"temp":23.77,"temp_min":23.77,"temp_max":24.35,"pressure":1010.89,"sea_level":1010.89,"grnd_level":1004.71,"humidity":55,"temp_kf":-0.58},"weather":[{"id":500,**"main":"Rain","description":"Leichter Regen"**,"icon":"10d"}],"clouds":{"all":42},"wind":{"speed":3.56,"deg":284.462},"rain":{"3h":0.5},"sys":{"pod":"d"},"dt_txt":"2019-06-20 18:00:00"},{

"dt":1561064400,"main":{"temp":21.06,"temp_min":21.06,"temp_max":21.45,"pressure":1012.35,"sea_level":1012.35,"grnd_level":1006.36,"humidity":66,"temp_kf":-0.39},"weather":[{"id":803,"main":"Clouds","description":"Überwiegend bewölkt","icon":"04n"}],"clouds":{"all":68},"wind":{"speed":1.85,"deg":321.393},"sys":{"pod":"n"},"dt_txt":"2019-06-20 21:00:00"},{

"dt":1561075200,"main":{"temp":19.76,"temp_min":19.76,"temp_max":19.95,"pressure":1013.43,"sea_level":1013.43,"grnd_level":1007.46,"humidity":74,"temp_kf":-0.19},"weather":[{"id":803,"main":"Clouds","description":"Überwiegend bewölkt","icon":"04n"}],"clouds":{"all":67},"wind":{"speed":0.76,"deg":41.864},"sys":{"pod":"n"},"dt_txt":"2019-06-21 00:00:00"},{

G6EJD, 2019-06-20:
I find it confusing too, but when the conditions are correct you will see:

“main”:”description”:””,”icon”...

So description will be null/blank.

Have you been able to try the German on v15 yet, I’m confident it will work.

G6EJD, 2019-06-20:
There was an error on line 102, I just corrected it, it had || true to keep it awake all the time.

hoffmakl1961, 2019-06-20:
@AWSW2017 - same small changes in the lang_gr.h file. 
[lang_gr.zip](https://github.com/G6EJD/ESP32-e-Paper-Weather-Display/files/3310829/lang_gr.zip)

V15 - Great. 

[image]

G6EJD, 2019-06-20:
In the API it says:
weather.main Group of weather parameters (Rain, Snow, Extreme etc.)
◦weather.description Weather condition within the group
All explained here: https://openweathermap.org/weather-conditions 

Thanks for the upload, good to see it working.

hoffmakl1961, 2019-06-20:
Some proposals:
1. display only main if description is blank, 
2. I would change the display areas, like in the old Versions. The reason is there is more room to describe the moonphases and sunrise (Sonnenaufgang) sunset (Sonnenuntergang).  
3. the headline is snow / rain. It is not required e.g. 50 mm rain or 50mm snow, (only 50 mm).  if rain and snow is 0 then I would only output "none" 
[image]

AWSW2017, 2019-06-20:
Hello,

@hoffmakl1961 Thanks for the changes, looks great. See below. I already wanted to write that in my last file the umlauts were not set back again. I like your version even more. Well done.

[image]

@G6EJD Tried to remove change the named "|| true", but the LOLIN32's blue LED stays on too. I think in the previous used V8 it went off after going to sleep. Will try with your next version again. :)

@hoffmakl1961 The proposals are good. I already thought of 2 of them too. In number 3 I also think it would be better to have the amount of "mm" only without naming it to rain or snow. The user might get that anyway. At the moment it looks kind strange to see a line of snow when its about 26°C. ;) 

Thanks a lot in advance and have a nice day! :)

G6EJD, 2019-06-20:
**@hoffmakl1961** I like your proposals.
1. I have now removed 'snow' if none forecast, but I have seen both rain and snow forecast for this country (all precipitation). I could take out the words and use symbols a drop for rain and * for snow. 
2. I've added Visibility, but only if it is reported.  and moved the Cloudcover over, so both can fit in the same space.
3. I've removed the rectangle around sunrise/set so there's a lot more space.
4. Regarding the forecast text box, right now mine is saying 'Rain (light intensity drizzle rain & Rain)' so that would become 'light intensity drizzle rain & Rain.
4.a I used to remove duplicate words so that it would look like this: Rain (light intensity drizzle rain)'
4.b. Maybe just for German to do it that way round, it won't look correct in EN.
?

hoffmakl1961, 2019-06-21:
@G6EJD I'm  impressed how fast you implement ideas. 
remark to 1:
the use of drop and * would be a good solution, but is the value 0 then e.g. only "none"
For a better optics 
The rectangle (width and height) for temperature, pessure and Rain/Snow could now be the same size.  
remark to 3.
"Sonnenaufgang", "Sonnenuntergang" and "abnehmender Halbmond" are long. 
remark to 4.
for German
 - I not sure, but I would use 4b.
 - remove - all set to lowercase (as described above by AWSW2017). I think that was implemented in an earlierer version

hoffmakl1961, 2019-06-21:
@G6EJD 
With V15 I have a  problem with the deep-sleep process. 
Following some timestamps from the serial Monitor: 
12:29:50.870 -> Starting deep-sleep period...
...
12:59:39.682 -> Starting deep-sleep period...
...
12:59:57.054 -> Starting deep-sleep period...
...
13:00:08.777 -> Starting deep-sleep period...
...
13:29:52.934 -> Starting deep-sleep period...
...
13:59:42.708 -> Starting deep-sleep period...
...
13:59:58.153 -> Starting deep-sleep period...
...
14:00:08.649 -> Starting deep-sleep period...
...
14:29:52.636 -> Starting deep-sleep period...

[image]

G6EJD, 2019-06-21:
I've just uploaded a version that addresses nearly all that's been discussed here:
1. Added ß to translations, eventually that conversion can move to the lang_xx.h file
2. Spaced temperature, pressure and precipitation equally., suggest GE use 'niederschlag' for 'Rain/Snow'
3. It no-longer displays Rain or Snow unless there has been any.
4. The 'Rain suffix' has been replaced with two rain drops, simialry for 'Snow' two snow flakes, no words and '=Rain' and same for snow have gone.
5. Improved the Cloud Cover icon. Only shows if reported.
6. Added a Visibility icon and reported distance. Only shows if reported.
7. Fixed the occasional sleep time error resulting in constant restarts, occurred when updates took longer than expected.
8. Improved the smaller sun icon.
9. Added more space for the Sunrise/Sunset and moon phases when translated.
10. I need to see that Main Weather is never reported without a forecast weather, I'm sure it does no, next stage will be to remove the Main Weather category i.e. Cloudy (broken clouds) becomes 'Broken clouds'.

Hopefully this corrects. most of the observations and suggested improvements.

hoffmakl1961, 2019-06-21:
it's incredible how fast you implement changes. 
I'll install the code asap. 
to 1: I would move it to each lang_xx.h file,
to 2: please use a  large letter for  "**N**iederschlag". I have also used "Niederschlag" in the old Germany version.  
to 7: I also analyse the SleepTimer, but you was faster as I.
to 10: for German without set to lowercase. OWM data were all written corectly. And I would remore the "(" and ")".

AWSW2017, 2019-06-21:
WOW! That is really fast. Many thanks. You deserve a beer today evening. :)

Made some light changes to the German lang_gr.h maybe for a next version. The texts fit now much better on the screen.
[lang_gr.h.zip](https://github.com/G6EJD/ESP32-e-Paper-Weather-Display/files/3315386/lang_gr.h.zip)

Looks actually like this:
[image]

Thanks a lot. To me it is a big and good update.

AWSW2017, 2019-06-21:
One think I am wondering about... Maybe you can give me a hint... I am using the LOLIN32 and the blue LED still stays on after the display was updated. Is this usual or can I turn it off? I played around with the builtin LED command in the DeepSleep void but this does not to seem an effect...

Thanks in advance :)

G6EJD, 2019-06-21:
Are you selecting the D32 board type? If on 1.0.2 of the ESP32 you should see that board type.

Interesting problem with the LED. Do you know what GPIO pin it is on, usually GPIO5 which is used by the SPI bus as SS and why in the code I include a #ifdef LED_BUILTIN test and if defined I turn off GPIO5 to switch the LED OFF.
What you need to do is confirm what pin it is on, then do this:
  pinMode(YOUR_GPIO_PIN_for_LED, INPUT);     // If it's On, turn it off as some boards use GPIO-5 for SPI-SS
  digitalWrite(YOUR_GPIO_PIN_for_LED, HIGH);

G6EJD, 2019-06-21:
Thanks for the example photo, I appreciate that and also for taking the time to update the lang_gr.h file, which I have just uploaded.
Hope you solve your LED issue.

hoffmakl1961, 2019-06-21:
Can you please introduce a version and release management? 
major changes get a new version # and minor changes get a new release #. e.g. V15R1

G6EJD, 2019-06-21:
I’ll do that in the code currently v16 along the lines you suggest, now when you go to examples you will only get the latest. Plus I’ll improve the release notes so you can see what’s changed without using github.

hoffmakl1961, 2019-06-21:
In the lastest sketch (40 min - old) is the same problem with the SleepTimer. 

serial output
20:26:28.291 -> Awake for : **9.305-secs**
the next one is correct. 
20:30:07.340 -> Entering 1790.00-secs of sleep time

G6EJD, 2019-06-21:
It depends when you start it off it calculates the time to the next 30min boundary so if started at 20:55 then it will sleep for 5-mins then awake again and then sleep for 30-mins then go to sleep finally at 23:00 if sleeptime = 22; 00:00 if 23; then wake at 07:00 if wakeuptime = 7;

AWSW2017, 2019-06-21:
A very big "thank you" from me again too. 
I selected the board type "ESP32 Dev Module" instead of "LOLIN D32"... :(
Now the LED is sleeping too.

Thanks for the efforts to both of you and enjoy your weekend.   :)

G6EJD, 2019-06-21:
With a 3000mAHr battery, I use a Samsung 18650 you should easily get 6/months of operation, especially now the LED is off.

AWSW2017, 2019-06-21:
Sounds really good. In the frame i could only use a 1000mAh LiPo battery pack because the frame has not enough depth. I added a TP4056 to charge it up quickly so that should not be a problem even if it unloads faster. I will have a look at it. ;)

G6EJD, 2019-06-21:
Today I received s flat li-ion pack from AliExpress it’s 100mmx60mmx5mm with a 5000mAHr capacity, cost me €7 delivered, but took 4-weeks to arrive. It’s really compact.

AWSW2017, 2019-06-21:
That sounds interesting. Thanks. I just created 2 new projects for these frames which will need more power... I’ll have a look.

G6EJD, 2019-06-21:
http://s.aliexpress.com/jma2YNF7
Price has up since 4-weeks ago, still reasonable.

hoffmakl1961, 2019-06-22:
I agree with AWSW2017 that is really a very nice project. 
You invest a lot of time.
     - .... .- -. -.- -.-- --- ..-

Some ideas for the future
1. use of the E-Paper ESP32 Driver Board, supports various Waveshare SPI e-Paper raw panels. No additonal HAT PCB is required. 
	https://www.waveshare.com/wiki/E-Paper_ESP32_Driver_Board
2. It would be very helpful >> use of OTA. I think that is a memory problem. 
3. Change from the Arduino IDE environment to platformio
4. create a PCB layout that the PCB can be ordered by a CN - manufacture; The inside of the station is not really nice. 
[image]

5. I've also installed a BME280 connected via SPI for the indoor climate. I know  it is not suitable for battery operation.    
[image]

The plan is >> wake up the ESP32 every 3-5 min (for the BME and calculate a average value) and every 30 min download the OWM data.
	In the Specification of the 7.5inch e-Paper - Each update interval time should be minimum at 180 sec. 
6. I've also installed a Mikrowelle RCWL-0516 - e.g. the e-paper will only be updated  if someone nearby. (Not implemented in the sketch by me and not suitable for battery operation)
[image]
PS: I do not have much experience with c++ and object oriented programming. I need a lot of time for each change.

G6EJD, 2019-06-22:
I've just uploaded a new version that changes the weather forecast text from Category (forecast1, forecast2) to Forecats0, Forecast1, Forecast2, no need to translate the DE, OWM does it for us.
I see what you mean about the PCB. This is what I use, it has provision for BME280 on board. Costs about $5 from jlcpcb, you can see them in the public area and order if you wish. Or I can email you the so-called gerber files which you use to order.
[image]
I2C-1 takes a 'standard' BME Vcc-Gnd-SCL-SDA

hoffmakl1961, 2019-06-22:
The weather forecast works fine. Now I will make a long term Test.
[image]

The PCB is nice, but I'll use my existing device.

hoffmakl1961, 2019-06-24:
@G6EJD are you interested in code changes?  What I will change.