G6EJD/ESP32-e-Paper-Weather-Display #56: Release management

televator, 2019-06-20:
The current procedure for updates to this library is to copy one of the example directories, change the version number and ultimately delete the old example files.
This approach is pretty unique to this repo and comes with a couple a drawbacks, namely:
- The current version strings have no meaning, and are even not consistent over the different display sizes (e.g. v11 of 7.5" has no connection to v11 of 4.2")
- Updates to the "core" library don't result in a new version bump, so the date of download is highly relevant.
- It is unclear which old versions can be considered outdated, which need to be maintained and which can be deleted.
- There is no usable git history of the files.
- The user cannot simply choose his/her display size but also needs to choose among different versions, which have no meaning and the latest is always considered best anyways.
- and so on...

I opened this issue to start a discussion on how a better release management could look like and to hear what is important to you.

G6EJD, 2019-06-20:
I agree with what your saying, but this latest version 15 now includes u8g2 fonts and is still being developed for DE language translations so strictly speaking it’s done form of Beta but to be honest I don’t have the time to follow a strict version and release regime.

Yes my next task will be to refresh the 2.9 and 4.2 and soon 1.54 versions so they use the same font library, but getting the 7.5” version running has taken some considerable hours to do, now to start on the others.
That said the smaller sizes can now reasonably quickly be converted.
Ideally all the dimensions would be scalar but largely that’s not possible.
My next task is to address reliability of OWM connections are high are still occasionally failing for the forecast elements. 
The recent changes I made in v15 for wake up and sleep timings are an interesting puzzle as they always work for me but a friend in CZ reports it doesn’t, even when I use the same time server, so there’s a strange timing issue happening, so today I’ll devote some more hours on that issue.
Yes eventually it will all come together.
Finally looking at the translation elements this morning I realise these look up elements can go into the language file so that the translation can be in any language.
I think the move to accented characters is going to make a big difference to those easy of me in Europe.

televator, 2019-06-20:
Good to hear. But just to be clear, a (tbd) versioning scheme should make things easier for you and all users, but definitely shouldn't be a burden.
Therefore, I am thinking in the direction of a really lightweight solution like a dev-branch for you and other contributors to work on and a master-branch as a default download branch. Every now and then when you consider the dev-branch stable enough for general public use, you could merge the dev-branch into master.
Or even more lightweight: You just work on the master branch but override the old versions every time, instead of copying. In essence, this only means to drop the version number from the filenames.

G6EJD, 2019-06-20:
Yes I think the latter is the best solution, to drop version numbers and then that’s the latest for that screen size, the move to elements of the code in a separate include has definitely eased the workload. 
The other 4.2” weather girl icon version could be placed in this repository too as it shares much of the includes.
Currently I’m looking at the advantages of importing some u8g2 fonts into the font file here as the u8g2 font file is huge (~30MB) and it takes some time to compile. But then it all becomes non-standard plus u8g2 would not compile (does now) with the latest ESP32 build 1.0.3rc-1 due to increase syntax checking now being applied by gcc settings, un-initialised variables are now treated as an error previously a warning.
This is a full-time job, but it’s part-time / hobby time or should be!

televator, 2019-06-20:
In https://github.com/G6EJD/ESP32-e-Paper-Weather-Display/pull/57, I made a suggestion how dropped versions could look like. I called the examples "Waveshare_X_Y" as this is probably the most common use case and it looks pretty pithy in the Arduino IDE. However, I can change them to whatever you prefer.

And I fully respect all the effort you put into this project. I hope, I can offer some help in return. And I really appreciate your openness to changes to this project - I know this project is your "baby".

closed 2019-06-20 by televator