G6EJD/ESP32-e-Paper-Weather-Display #60: Accidental .ino.ino

televator, 2019-06-21:
Hi,

there is an accidentally committed Waveshare_7_5.ino.ino, I could clean that up, but in this case a rebase and force push would probably the better option.
Simply renaming and overwriting the .ino will delete my changes.

If you need help, you could also give me force-push rights for master for an hour or two.

G6EJD, 2019-06-21:
Yes, got to slow down a bit, I saved it over the top of another and the Arduino IDE did the usual trick of adding another .ino extension, so I removed it and replaced the correct file.

closed 2019-06-21 by G6EJD

televator, 2019-06-22:
Sorry, but as I predicted, you now overwrote my changes - unfortunately, not for the first time.
This is pretty frustrating to say the least as I offered you to do it professionally.
I played around how to reapply those patches... the constant change of formatting makes it really hard, though.

G6EJD, 2019-06-22:
Sorry, what changes did you make? in 7x5 version?

televator, 2019-06-22:
I reintroduced the changes made.

televator, 2019-06-22:
It was easier to make the changes "by hand" as the Diff is pretty messed up with all those formatting changes.
Have you considered using an auto-formatter? Not only does it help with consistent, nice-looking formatting, but it also helps for a consistent diff-able git history.
As far as I know you are using the Arduino IDE to make your coding. The Arduino IDE has a build-in Auto-formatter - it is definitely not state-of-the-art, but maybe an easy to use start. A professional solution would be clang-format, a command line tool. The latter could be easily setup as git pre-commit hook to avoid messing up formatting.

G6EJD, 2019-06-22:
Interestingly I always format the code before I upload it. What I have done which is what your referring to is remove excess spaces in some of the calculations around say ‘ + ‘ now ‘+’ and similarly for mathematical operators, that will make difference checking mask the real changes which were minimal.

televator, 2019-06-22:
I am just saying there is absolutely no need to do such things by hand and it will be consistent. Again, this should save you precious time, not be a burden; and I am happy to help you setting things. As a start, maybe give Ctrl+T a try in your Arduino IDE.

Just realized that you overwrote even more changes than I first saw. I'll fix those in a second.

G6EJD, 2019-06-23:
Observation: in the DrawMoon function the change to:
    double pW1x = (Xpos1 + number_of_lines) / number_of_lines * diameter + x
number_of_lines = 90;
Is degrees in the original formula as it calculates angles around the circle. Akin to sin(x+90) using Pythagoras (e.g. x^2 + y^2), so really 'quadrant_angle'
It can stay as it is, because I understand what it means.

televator, 2019-06-23:
I will changes this in the next PR or so... My intention is mainly to get rid off the magic numbers - I made a plot to see what's going on and thought `number_of_lines` would describe it okay-ish. Anyways, I already thought about rewritting the whole loop in terms of Bresenham's algorithm. Actually, I already have some code lying around. This would avoid a few artifacts which appear every now and then with the current solution... couldn't bother yet, though.

G6EJD, 2019-06-23:
Interesting, the Adafruit GFX library already does a really good job at aligning pixels to a drawn object, but most problems are aliasing between the desired line and screen resolution and that can never be solved in the equations. Actually the minigrafx circle drawing is not that good as it starts to approximate to a square rather than try to optimise the shape and pixel alignment. A move to the GFX solutions in u8g2 will help.
I don’t know what your education philosophy was/is but in the UK we are taught and I teach students not to over engineer solutions so that the maintainer and other students can understand how it works, not that drawing a moon is that easy to understand😊

televator, 2019-06-24:
just to let you know that you again overwrote my change uploading the 16.2 - I assume accidentally.
I think this is it.

btw: wcsftime would have been the one-liner solution to localization of dates.

G6EJD, 2019-06-24:
Andreas, yes of course nothing was deliberate, in-fact, by updating the function 'updatelocaltime' I was encouraging  and responding to the same encouragement you gave to @hoffmakl1961 this morning to propose changes, which he then did. Actually, in-hind sight, none of these changes were required as yes other ways could have been used. I guess I was struggling to get the code to conform to apparent German localisations that he said were the normal in Germany and wanted to introduce. Had I stood back from the changes, I should have seen that he could have done all that he wanted by just editing his own lang_gr.h file. 

Before upload I did check to see if there was anything pending and saw nothing. All other changes up to this morning (IP address display) were incorporated. 

What changes did you have planned? From my point of view I see nothing in progress to know whether I can make a change or not that might invalidate something your doing at your location.

televator, 2019-06-24:
> Before upload I did check to see if there was anything pending and saw nothing. All other changes up to this morning (IP address display) were incorporated.

That is not at all the problem - what you overwrote was no pull request or local change. You overwrote stuff that was already in the repo. See this commit https://github.com/G6EJD/ESP32-e-Paper-Weather-Display/commit/bbd98301419d3ddd6d7b39aa0c0d22854cfefe0b.