G6EJD/ESP32-e-Paper-Weather-Display #61: a minor issues with Waveshare_4_2 V11

hoffmakl1961, 2019-06-22:
Hello G6EJD,
with the current Waveshare_4_2 V11 sketch I get an compiler error.  

Waveshare_4_2:268:50: error: 'Sunrise' was not declared in this scope
Waveshare_4_2:269:50: error: 'Sunset' was not declared in this scope

after declaration of Sunrise, Sunset as int
`int wifisection, displaysection, Sunrise, Sunset;
`
the displayed values are 01:00.

[image]

televator, 2019-06-22:
Just hold on for a second, I am working on a fix.

televator, 2019-06-22:
PR is open, let's wait for a review: https://github.com/G6EJD/ESP32-e-Paper-Weather-Display/pull/62

closed 2019-06-22 by G6EJD

hoffmakl1961, 2019-06-22:
after the update; 
Sunset and Sunrise will be displayed with 1 o'clock. 

[image]

televator, 2019-06-22:
Thanks for the feedback.
Sorry, that was my mistake, https://github.com/G6EJD/ESP32-e-Paper-Weather-Display/pull/64 should fix it.
I hope that once @G6EJD finds the time to update the 4.2" and 2.7" to GxEPD2 all those functions can be unified among the display sizes and mistakes are easier to spot.

hoffmakl1961, 2019-06-22:
Lots of changes
You make a very good job.

G6EJD, 2019-06-22:
Thanks, I’m still having a few issues with timing for wake/up, it measures time to update and then subtracts that from the sleep duration so typically 30mins - 8secs except sometimes it reports longer than it took to update if that makes sense, so then the next update occurs too early on the 30-minute boundary programmed, I think it must be a data type problem.
The weather text will be tested when there are multiple entries in forecast 0, 1 and 2.