G6EJD/ESP32-e-Paper-Weather-Display #65: output error of WiFi.localIP in serial monitor

hoffmakl1961, 2019-06-24:
`Serial.println("WiFi connected at: " + String(WiFi.localIP()));`
create the following output
WiFi connected at: 2365630656

you could use the following line

    `Serial.printf("\r\nWiFi connected at: %s\n", WiFi.localIP().toString().c_str());`    
WiFi connected at: 192.168.0.141

televator, 2019-06-24:
Thanks for the bug report.

This fixes it
https://github.com/G6EJD/ESP32-e-Paper-Weather-Display/pull/66

Note that there is no need to use c arrays.

But maybe you can suggest such changes yourself by opening a pull request. It's really easy, you don't even need to use git. Just click the "Pencil" symbol when you have the file open in GitHub, make your changes, put a short description and open the pull request.

closed 2019-06-24 by hoffmakl1961

hoffmakl1961, 2019-06-24:
Thank you for the tip. (pull request)