G6EJD/ESP32-e-Paper-Weather-Display #68: Failed to obtain time

schroepfer, 2019-07-02:
Hello,
i use your V16.3 Waveshare_7_5 and get this error: (an 1 year old Version is running)

_Connecting to: FRITZ2
WiFi connected at: 192.168.8.110
Failed to obtain time
Entering 1800-secs of sleep time
Awake for : 9.110-secs
Starting deep-sleep period..._

Please can you tell me, what's wrong?

G6EJD, 2019-07-02:
It’s failing at line 502: and it looks like you need to change the time server ‘0.uk.pool.ntp.org’ to something closer to you, perhaps change line 502 to ‘0.europe.pool.ntp.org’ and see if that gets time.
On line 512 it waits up to 5000mS/5-secs for the time server to respond, it is possible your link is a little slow and 5-secs is not enough, you could try changing that to 10 or 15 secs (10000 or 15000).
16.3 has proved to be more reliable than earlier versions, and it includes more error reporting hence we now know what’s going wrong. It’s unusual to not not get time.

schroepfer, 2019-07-02:
Thank you for your fast help! 
The Time in line 512 was too short to sync with 5 sec. I changed it to 10000 and it worked!

And thank your for your fantastic work!
BR
Juergen

closed 2019-07-02 by schroepfer

G6EJD, 2019-07-02:
Good to hear it’s resolved, but please change your time server to ‘europe’ that will shorten the delay in getting g a valid time synchronisation. It measures round-trip time so from your location to the time/server and back again so it can gat the round trip delay and offset time readings accordingly, if too long it fails, so as short a time between your location and a local time server is best.
I’ll update the code today to allow for time servers to be set as part of the credentials file, so that will be 16.4.

schroepfer, 2019-07-02:
Yes, i set it to 
configTime(0, 0, "0.de.pool.ntp.org", "3.de.pool.ntp.org"); 
and 
const char* Timezone = "CET-1CEST,M3.5.0,M10.5.0/3";
now it is fine!

G6EJD, 2019-07-02:
Check out the changes in the new code I made some minor adjustments throughout