G6EJD/ESP32-e-Paper-Weather-Display #69: Variables not declared in scope

carjufi, 2019-07-02:
I have donwloaded the .zip, installed the library and while compiling there are errors of variables "not declared in scope". I´ve tried to fix it but I do not know the type of variables they are. Error:

Waveshare_2_9:398:3: error: 'CurrentHour' was not declared in this scope
   CurrentHour = timeinfo.tm_hour;
   ^
Waveshare_2_9:399:3: error: 'CurrentMin' was not declared in this scope
   CurrentMin  = timeinfo.tm_min;
   ^
Waveshare_2_9:400:3: error: 'CurrentSec' was not declared in this scope
   CurrentSec  = timeinfo.tm_sec;
   ^
Waveshare_2_9:406:48: error: 'weekday_D' was not declared in this scope
       sprintf(day_output, "%s, %02u. %s %04u", weekday_D[timeinfo.tm_wday], timeinfo.tm_mday, month_M[timeinfo.tm_mon], (timeinfo.tm_year) + 1900); // day_output >> So., 23. Juni 2019 <<
                                                ^
Waveshare_2_9:406:95: error: 'month_M' was not declared in this scope
       sprintf(day_output, "%s, %02u. %s %04u", weekday_D[timeinfo.tm_wday], timeinfo.tm_mday, month_M[timeinfo.tm_mon], (timeinfo.tm_year) + 1900); // day_output >> So., 23. Juni 2019 <<
                                                                                               ^
Waveshare_2_9:410:46: error: 'weekday_D' was not declared in this scope
       sprintf(day_output, "%s %02u-%s-%04u", weekday_D[timeinfo.tm_wday], timeinfo.tm_mday, month_M[timeinfo.tm_mon], (timeinfo.tm_year) + 1900);
                                              ^
Waveshare_2_9:410:93: error: 'month_M' was not declared in this scope
       sprintf(day_output, "%s %02u-%s-%04u", weekday_D[timeinfo.tm_wday], timeinfo.tm_mday, month_M[timeinfo.tm_mon], (timeinfo.tm_year) + 1900);
                                                                                             ^
Waveshare_2_9:413:35: error: 'TXT_UPDATED' was not declared in this scope
     sprintf(time_output, "%s %s", TXT_UPDATED, update_time);
                                   ^
Waveshare_2_9:419:35: error: 'TXT_UPDATED' was not declared in this scope
     sprintf(time_output, "%s %s", TXT_UPDATED, update_time);
                                   ^
Waveshare_2_9:421:3: error: 'date_str' was not declared in this scope
   date_str = day_output;
   ^

G6EJD, 2019-07-02:
Sorry, I’m in the process of aligning all the versions and converting to the U8g fonts, should be done by tomorrow.

carjufi, 2019-07-02:
Amazing! thanks for the prompt response

G6EJD, 2019-07-03:
OK all done, you'll need to download the REPO again, adjust the credentials file to suit your details, API key and location. Install u8g2 and GxEPD2 all from the IDE sketch manager

closed 2019-07-03 by G6EJD

carjufi, 2019-07-03:
Thanks very much!! I will test it now and let you know

carjufi, 2019-07-03:
It works! I have many "WiFi connection *** FAILED ***" attempts, I am not sure why but it happened with the previous code and with your updated one. Do you have any idea why? Maybe the router is blocking something?

I am using a DOIT Esp32 DevKit v1

G6EJD, 2019-07-03:
This usually due to insufficient WiFi signal try turning it through 90°to see if that makes a difference also is there any electrical equipment running near by? I did not include WiFi signal strength but try printing that out to see if you’ve got a reasonable signal say >65dBm.

carjufi, 2019-07-03:
[image]
[image]

Nice, yeah when I walk nearer the router the connection success is much higher. Thanks for this! 
I have it enclosed in a 3D printed case with a temperature sensor and a small PCB for the voltage divider to read the battery level, do you think this adds noise? Signal I am getting is around -49dBM, worth adding an antenna? And, how to add it?

Now with your new code it looks like it will be more stable, before the wifi function stuck and the board tried to continuously connect and thus draining the battery (3.7V - 550mAh) in a matter of hours. Would you recommend having a bigger battery?

Thanks and awesome work!

G6EJD, 2019-07-03:
It’s interesting to see your project, thanks, the case is neat - good design job on it. The signal strength is high, no problems there. I’ve spent a lot of time on the new software to reduce power consumption in and note it now sleeps over night to save about 1/3 of daily power consumption, plus if when it wakes it can’t make a WiFi connection or get time or read the weather data then it goes back to sleep for 30-mins before it tries again. Also I d added the http client method that has proved to be more reliable together with improved WiFi and time acquisition methods all of which will make for a better solution.

G6EJD, 2019-07-03:
The battery should last about 1-month but with solar support that you have indefinitely. You can use my calculator from the solar TN to get a more accurate battery duration abs simulate how well your solar panels perform to keep it all running.

carjufi, 2019-07-04:
Yes, thanks! I have your solar calculator and yes, that should be indefinitely. I will finish adapting the code to include the temperature and humidity sensor and will test how it goes and let you know. 

One more question, now regarding the voltage divider and the battery level. I’me made a voltage divider with a 100k and 360k resistor and updated the formula as follows (as per your video): analogRead(36)/4096*4.2*(366.667/360). 
Then you have added a new formula to calculate the percentage using a polynomial. To find this formula, how can i do it? I mean, how can i extract the data to find it on excel? Do i have to connect the battery and read continuously until it drains out?