G6EJD/ESP32-e-Paper-Weather-Display #70: Busy Timeout!

schroepfer, 2019-07-04:
Hello,
i use your newest V16.4 and have problems to connect. 16.3 works correct.

_ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1100
load:0x40078000,len:9232
load:0x40080400,len:6400
entry 0x400806a8

Connecting to: FRITZ2
WiFi connected at: 192.168.8.102
Busy Timeout!
Entering 157-secs of sleep time
Awake for : 19.987-secs
Starting deep-sleep period..._

G6EJD, 2019-07-04:
Interesting as there is little difference, in fact no difference in the http fetch functions. I did change the time server to the recommended ‘pool.ntp.org’ rather than a specific ‘0.europe.pool.ntp.org’ or ‘0.de.pool.ntp.org’ but getting time is early in the start-up, so not that.
I’ll do some difference checks and test it this morning. Thanks.

schroepfer, 2019-07-04:
in the WAVESHARE_2_9 example i get the same problem:

Connecting to: FRITZ2
WiFi connected at: 192.168.8.104
Failed to obtain time
Busy Timeout!
Entering 1800-secs of sleep time
Awake for : 18.999-secs
Starting deep-sleep period...

but i changed the 5 sec to 15 and it worked!
   while (!getLocalTime(&timeinfo, 5000)) { // Wait for 5-sec for time to synchronise

G6EJD, 2019-07-04:
Thanks for the feedback. I assume your using the top-level pool NTP server address in the credentials file
default 'pool.ntp.org' ? Which is what they recommend, but I'm thinking that address is taking a long time to be resolved and hence the need to increase connection time-out, it should connect in millis-secs.
Please try '0.de.pool.ntp.org' in the credential file. That is a more direct route to your nearest time server. Or 1.de.pool.ntp.org Or 2.de.pool.ntp.org

schroepfer, 2019-07-04:
I use this one:
configTime(0, 0, "0.de.pool.ntp.org", "3.de.pool.ntp.org");

G6EJD, 2019-07-04:
Yes, those are your local time-servers so best solution. The NTP Pool advice works but takes too long to resolve the address. BTW ideally set the 0,0 to GMToffset, DSTOffset.

I'm changing the credentials text and examples to reflect this, thanks for your help.

closed 2019-07-04 by G6EJD

schroepfer, 2019-07-04:
I changed it to: (in your newest Waveshare_7_5)

const char* ntpServer   = "1.de.pool.ntp.org";
int   gmtOffset_sec     = 0;  
int  daylightOffset_sec = 3600; 

here the result: (it doesn't update)  :-(
....
Icon name: 04d
Icon name: 04d
Voltage = 4.19
Busy Timeout!
_PowerOn : 10001028
Busy Timeout!
_Update_Full : 10001015
Busy Timeout!
_PowerOff : 10001015
Entering 1710-secs of sleep time
Awake for : 40.294-secs
Starting deep-sleep period...

G6EJD, 2019-07-04:
Interesting, actually it got time OK, but that Busy TimeOut error is as a result of getting trying to get Forecast Weather. So your country code is DE and Language DE ?
This could be a OWM server issue at the moment, I've seen it many times, that Weather data is always received, but Forecast data is not.

schroepfer, 2019-07-04:
yes DE and DE. I'l try another coutry.

G6EJD, 2019-07-04:
I’ve just checked my two (7.7 and 4.2”) and they both show no connection to OWM so I think that’s the (OWM) issue. I’m going to report errors on screen so it’s possible to know what’s going wrong as a next update.

schroepfer, 2019-07-04:
This is a realy good idea!

schroepfer, 2019-07-04:
It is runng now!
I must use the LOLIN 32 PRO for correct working with my TTGO ESP-32 WROVER.
But: it doesn't wake up from deepsleep. Only when i press the reset-button!?

AWSW2017, 2019-07-04:
Hi, I think I have the same „too“-deep sleep issue too. I updated yesterday to 16.4 and it seems that it does not wake up after some cycles/hours.

With the before used 16.0 version I think it stopped after about 3 days and needed a reset click. 

I read about this deep sleep problem somewhere else and there a hint was named that after the Serial.begin(115200) command line you should add a at least 500ms delay to fix this. Within my own sketches I followed this and they seem to work... maybe a hint for this too?

schroepfer, 2019-07-04:
Thank you! I will try this tomorrow and tell you if it helped.

G6EJD, 2019-07-04:
@AWSW2017 I've added a delay after Serial.begin to see if that improves reliability, plus I've added an error log facility, I'll upload it soon. Thanks for your help.

schroepfer, 2019-07-05:
https://github.com/AWSW2017 
Hello, i inserted 1sec delay after Serial.begin and it did **not** help! He doesn't wake up.
I addet a 470uF capacitor on the 3.3V of my WROVER and it did not help to wake up!
It definitely has to do something with the power supply:
When i plug in an USB-power supply, it wake up every 1/2 hour!

I will test another ESP32. Can you recommend a good one?

G6EJD, 2019-07-05:
A lot depends on the load placed on the on-board regulator most are designed for 500mA typical maximum with an absolute maximum of 800mA.

A key factor of all regulators is the drop-out voltage v load current, on my Lolin D32 it uses an NE6211 regulator and the data sheet says at 500mA load, the drop-out voltage rises to 700mV. So input voltage to the regulator needs to be 3.3 + 0.7 = 4.0v which can be achieved if a fully charged Li-ion/Lipo battery is used, but a battery at its nominal 3.8v will result in an ESP32 supply voltage at load of 3.8-0.7=3.1v so starting to go out of device specifications.
Given the ESP32 alone requires a peak current of 450mA when the RF stage is working, which occurs in the very early stages of switch on from sleep I understand why delays are being considered.

You could use a capacitor to provide the energy when there is load demand (See: http://g6ejd.dynu.com/index.php/micro-controllers/calculate-smoothing-capacitor-values/) but for a 1A peak lasting for 50mS you would require 250,000uF to give you an idea of the problem.

Ideally remove any loads and see if the bare (monitor the serial port) starts after sleep with no load, if so you know where the problem is. If needs be provide a separate regulator.

If applying USB power solves the problem then it is your regulator dropping out due to excess load and for battery supplies you need a higher voltage, but this is difficult, say 2x18650 in series would give you a nominal 7.6v input BUT most regulators have maximum Vin of 6.5v - so that’s not a solution. Using Nicads or Nmih say 4 in series gives 4.8v which would solve the problem, but at lower capacity of say 2400mAHr unless 8 batteries (2x4 in parallel- series) were used, but a lot of space needed!

Conclusion- it’s not a software issue it’s a voltage regulator problem, but not easy to solve for a development board with surface mount regulators, I also doubt there’s a product that can achieve a high load current LDO solution and the NE6211 is the best available. If you have any devices connected to the ESP32 power them separately via their own regulator.

schroepfer, 2019-07-05:
I could'nt find anything about the NE6211.
But i found the datasheet of my working eBox-Board: [(https://cdn-shop.adafruit.com/product-files/3081/mic5219.pdf](url)
I thing, it is nearly the same and available. [https://lcsc.com/search?q=mic5219](url)

G6EJD, 2019-07-05:
Apple spellchecker problem ME6211 or https://datasheet.lcsc.com/szlcsc/Nanjing-Micro-One-Elec-ME6211C33M5G-N_C82942.pdf

G6EJD, 2019-07-05:
Yes the Adafruit regulator is good one, but only can supply 500mA the ME6211 on the Lolin D32 is 800mA.

schroepfer, 2019-07-05:
This ist not correct. The datasheet of the ME6211C33 says max.500mA. 

I looked in the schematic of the Adafruit Huzzah32. There is a [AP2112-3.3](https://www.diodes.com/assets/Datasheets/AP2112.pdf) 
This has a Vdrop typ=250mV at 600mA. Realy nice!

G6EJD, 2019-07-05:
Yes, sorry about that, I was looking at another device then. I just used the Mouser product selector and used 3.3v, 1A, 50uA quiescent current, SMA/SDO package and nothing matches on the market.

G6EJD, 2019-07-05:
@AWSW2017 I think as we learn more about what's going on, we can conclude it's not a software issue, and you've helped focus on the hardware issues that it now seems to be causing this errant behaviour, yours running for 3-days then needing to be reset could be linked to a slowly discharging battery, so there is mounting evidence that this is all supply voltage linked.

AWSW2017, 2019-07-05:
@G6EJD Thank you for your reply. Seems to be this way. I will watch this. I transferred the sketch 16.4 with no added delay again yesterday with a slower baud rate, because I used 915000 for a try (with a good cable) and now set it to 115200 again. I will tell you if it is fixed. The battery was still at 100% after the reset after about 1.5 weeks👌 and now runs since yesterday nicely without a recharge...  

If it does not react anymore I will add some MQTT to get some log messages to my smart mirror without the need of a serial connection to the Lolin board. With this info the mirror can notify me if the device does not respond as usual by not sending the messages anymore. I have added this so far to all of my projects to have a at a glance look of all batteries levels of all devices around the house. If you are interested I can send you the changes needed to modify the weather station... 

Have a nice evening 😁

G6EJD, 2019-07-05:
Thank you for the feedback and ideas. In the 16.5 version I uploaded yesterday I’ve added the ability to record in an error log albeit transient what went wrong and to display the reason on screen. Thanks to your lead I think we have honed in on what is causing the issues and it’s not software related. For example o very rarely suffer from any issues only when my ISP service is interrupted or as happened this week, that the OWM servers failed, but now in 16.5 it will at least say why on screen until the next scheduled wake up time.

AWSW2017, 2019-07-05:
Thanks a lot! Just uploaded 16.5 and will leave it running on 2 displays to be sure. 

Have a nice weekend! :)

AWSW2017, 2019-07-07:
Good morning,

where can I find the log file? Today morning 1 of the 2 did not wake up after the update at 08:00 o'clock. I kept it in the status. The last shown battery level is still 100%.

Thanks in advance.

AWSW2017, 2019-07-07:
I have used the Reset button now. Before I could not see any output on the serial monitor.
I think it hangs in a loop, because after the reset was done, the battery showed 90% only. So it seemed to be active in the last 2 hours... Maybe this is a help... Thanks in advance.

G6EJD, 2019-07-07:
Well I made it transient, by that I mean when errors are detected by the code they are displayed on the screen until the next update, so you’d see any detected errors on the screen over writing from the bottom up. Now I assume none were displayed, there can be up to 10 errors.
For example if the ESP32 started but could not get a WiFi connection it will display that error, or could not get time, or no weather data receiver, same for forecast data or then JSON deserialisation for weather and forecast, so what is causing the error as it dies not appear to be under program control?

G6EJD, 2019-07-07:
The first error that can be recorded is a failed WiFi connection which will time out trying to get a connection after 15-secs, then getting time another 15-secs maximum, now as they weren’t display it suggests something went wrong before that,  like the processor did not start from sleep back to a known state. 
Interestingly my Lolin D32 never shows any errors, well apart from when the OWM servers go off line.

I’ll investigate this this morning and maybe change the error log to be a series of codes saved in EEPROM or nvram for later interrogation. The alternative is to display the error log and send the cpu to sleep for a long time  so the errors can be viewed.

G6EJD, 2019-07-07:
It now reports why the CPU restarted or was reset, does this on screen if it gets that's far in the code!

G6EJD, 2019-07-08:
16.7 I've now Disabled the brown-out detector, which I think is the problem your having, although this is a not a good option as if it's a PSU or grounding issue, this is not the solution rather a better PSU or grounding setup. Shorted larger cables to handle the >500mA current flow at start-up.

AWSW2017, 2019-07-08:
Hello and thank you. I will try the new version tomorrow.

AWSW2017, 2019-07-09:
Hello, 
just uploaded 16.7 to both screens. I will let you know. 
Both stopped 2x in the last 2 days with 16.5.
Thanks for the update.

PS: When it stops again I will add several MQTT points to track external how far it gets during an update.

G6EJD, 2019-07-09:
Thanks for helping. To be honest I think the CPU never starts and never gets to be under programme control. You’ll see that I’ve disabled the brown-out detector but the documentation does not explain whether that is a persistent or transient change, my best guess is it is a power supply issue. I have 5 boards running this code and have yet to see any of them fail. All Lolin D32 boards.

AWSW2017, 2019-07-10:
Hi 👋 Small update: 1 of the 2 setups stopped yesterday evening at about 23:30 o’clock. Screen was not updated anymore... After about 18.5 hours the battery went down about 60% then. So I think there runs something... in serial monitor I could not see anything... 

I will add the MQTT to log the walkthrough to the program to an external source. I’ll keep you updated. 😊

Besides that I will order some new Lolin D32 boards in China to sort out that these 2 boards might an issue.

G6EJD, 2019-07-10:
Thank you for the update. If you have the sleep time set to 23 then the last update should be 23:30 or occasionally 23:59 if the ESP32 (inaccurate RTC) is running fast then it wakes at say 23:59:30 and still does a valid update conversely at 24:30 when the ESP32 sleeps for 30-mins it next wakes at 0:00 and goes back to sleep not updating the screen, I see that across my samples, but they always wake up and never lock.

It will be interesting to see if your MQTT reports are made as the first thing it does is get a WiFi connection or not recording an error if not then reads the time or records an error if not, then if at 23:30 or beyond would not update the Wx data but if any error was detected it would display the errors on screen. So we can conclude it woke at 00:00 and went straight back to sleep but not waking up again.

Whether switching the brown-out detector off has made s difference is difficult to ascertain, but at first analysis it made no difference. The process restart being anything other than normal would have been recorded too.

Other differences could be the power/up reset sequence and whether all the conditions are being met for normal boot after sleep.

Whether it’s the development board or not is debatable but my Lolin D32 boards don’t exhibit any failures from sleep wake-up. This is why I’ve found it difficult to replicate the problem.

Initially there were clone Wemos D32 boards being sold on AliExpress  then Wemos changed its name to Lolin and I’ve never seen the clone boards since, it could be this difference that is creating the anomaly in operation. 

Maybe a short test programme that wakes,  blinks the LED, then goes to sleep and then see how that performs to eliminate nearly all of the code I’ll programme an old silicon Rev.0 board to do that and see how it performs, I’ll let you know the outcome starting from tonight.

AWSW2017, 2019-07-10:
You’re welcome. I have to say thanks. 😊 
Yes, I recognised that with the “no update during the set night times”. The display that started updating today morning was last updated at 23:59 yesterday. The one that stopped was last updated 23:30 the last time and was kept in this state until today 18:00 until I used the reset button.

What I find really strange is that I have 3 ePaper picture frames displaying randomly downloaded greyscale picture with the same electronic setup that run now for about 4 weeks without no issue. I also put the nightly timeout time to them during the night because I found that really clever so save some energy.

I will update the code with the MQTT and give it a try. 

Thanks a lot and have a nice evening!

G6EJD, 2019-07-10:
It is worth checking what silicon revision your ESP32’s are as I’ve read that restarting from sleep was a problem with the Rev.0 first release, now solved in Rev.1

G6EJD, 2019-07-11:
Out of interest, testing of an early ESP32 Lolin v1.0 board together with a do-it ESP32S shows some encouraging similarity of failures to wake-up, in that my simple sleep flash the-on-board LED then sleep again sketch (actually the same Wx code but it just flashes an LED when it wakes rather than refreshing the screen) has had errors from both boards in the form of the blue LED being stuck on at random times, not waking from sleep. Both are Rev.0 ESP32's. The do-it ESP32S board sometimes does not wake from sleep at all, but the Lolin ESP32 v1.0 board (not a D32) always appears to wake and the on-board blue LED stays illuminated and as it's configured as the EPD-CS this means it was held low (EPD selected) but it never goes on to flash the LED so never gets to be under programme control.

My Rev.1 ESP32 on the Lolin D32 boards work perfectly. Reference to various forums confirms the Rev.1 silicon did indeed fix wake-from-sleep issues and using WiFi after wake-up and more than a few other issues were resolved too.

So-far, it's looking for me like, it is an issue with Rev.0 silicon variants of the ESP32.

I'm leaving the tests running.

AWSW2017, 2019-07-12:
Hello 👋 Thanks for all your help 😁
Just ordered 10 new LOLIN D32 V1.0.0 boards from AliExpress. Should be here within 10 days...
The naming on my actual ones says V1.0.0 too and should be original ones ordered from Amazon...

For the moment I uploaded my still super stable running sketch using the 7.5 ePaper as a random picture frame with the same nightly timeout phase to see if that continues on the circuits not constantly running the weather station... The electronics setup is the same on all of them.... I just want to be sure if the circuit has a problem. I will leave this running until the new D32 boards are here. I added MQTT messages to these all too to see in my central MagicMirror to see if they all update every hour except the night times...

Have a nice weekend 😎

G6EJD, 2019-07-12:
Thank you for the update. My boards still haven’t failed displaying the weather due to a CPU restart error, but today all my 3 trial systems recorded a failed reception of Weather data, but Forecast was OK. This was the first time I’ve seen the error reporting work for real. The processor restart message was a normal wake from sleep status. Hopefully your new boards will improve your reliability. My boards say  V1.0.0 but they are not the same as the V1.0.0 boards I purchased about 2-years ago and which have Rev.0 silicon ESP32 fitted, they are the unreliable types.

AWSW2017, 2019-07-22:
Hi 👋 Just a quick update: The picture frame sketch is still running on all 3 displays without a single error or timeout since 10 days.

I am still waiting for the new D32 boards. I will create a new weather station for them then and keep the picture frames running until the battery dies. Actually they are at 75% with a deep sleep timeout of 30 minutes.

Have a nice day 😎

G6EJD, 2019-07-22:
Thank you for the update. At some point it's worth checking your ESP32's for their silicon revision either 0 or 1.  Rev.0 ESP32 is unreliable and prone to errors after sleep. The compiler gives the revision number during its first few lines (scroll-up the message output to find it).

AWSW2017, 2019-07-28:
Test is running with the new boards:

[image]

G6EJD, 2019-07-28:
Nice set-up and a good idea to remove the long ribbon and use the shorter stub. I'm going to do that too.

Newer boards do run a lot better than older variants, but silicon revision appears to be the same.

ag-ni, 2020-06-08:
I am having the same "busy time out" issue. I have tried increasing the timers as @schroepfer has mentioned (which solved the problem for him ) but the issue remains. I am using a 7.5 inch Waveshare paper display. The sketch uploads and it pulls data etc but the display fails. 
To eliminate the loose ribbon cable problem - I have tried uploading example sketch from Waveshare and that displays perfectly fine but when I upload this sketch, it gets uploaded, fetches data and then throws "busy timeout" error. Can someone help please. 

Regards
Shek

G6EJD, 2020-06-08:
Did you modify the gxepd2 library to shorten the reset pulse?

G6EJD, 2020-06-08:
The reset delay has to be 10x shorter

ag-ni, 2020-06-09:
@G6EJD : Can you give me a hint where can I modify it please ?

ag-ni, 2020-06-09:
I mean can  you tell me the file and parameter that needs to be modified.

G6EJD, 2020-06-09:
The GxEPD2 code that needs to be modified, https://github.com/ZinggJM/GxEPD2/blob/master/src/GxEPD2_EPD.cpp

Scroll down to line 69 you will see the reset function there, the delay used is 20mS not 2mS. You can adjust your library file to try it out, make a back up first then reduce the 20 to 2mS

Until the GxEPD2 code is modified this is a required fix.

ag-ni, 2020-06-09:
@G6EJD : I modified the code GxEPD2_EPD.cpp, reduced it to 2ms, recompiled and pushed, but still getting busy timeout.

G6EJD, 2020-06-10:
All the delay(20) lines become delay(2) when you say ‘pushed’ what do you mean? Your local library copy should now be showing the shorter delays.

Are you sure the cables especially the busy line is connected.

If that doesn’t work you need to contact the GxEPD author, but unfortunately he has removed the ability to report issues, In a somewhat arrogant way!

ag-ni, 2020-06-10:
Hi @G6EJD : Thanks a lot for responding. I have changed all the delays from 20 to 2 . When I say 'pushed', I mean compiled and uploaded the code in the esp32. 
The cables are connected properly. Because when I am checking with the Waveshare example sketch, the display works. But when I am checking with this code, I get 'busy timeout' error. I have changed the file in my local library but to no avail.

G6EJD, 2020-06-10:
So, to recap, your using the new display HAT version? And Waveshare have said the reset pulse duration needs to be reduced which is what they have changed in their example, so the question is, which GxEPD2 library are you using when compiling have you turned on verbose mode to ensure it is coming from your libraries folder (the location of the one the modified) as it’s easy to install two versions of a library.

ag-ni, 2020-06-11:
@G6EJD : Yes, I am using v2 display. 
Indeed its coming from the library folder where I have modified the file. I have made sure that I do not have two versions as it was the first thing in your post :-) 
Any other ideas where it could be going wrong ?

G6EJD, 2020-06-11:
Ask @AWSW2017 how he solved it? Maybe he made other changes in the GxEPD library

AWSW2017, 2020-06-11:
Hello 👋 
Please have a look to this entry: https://github.com/G6EJD/ESP32-e-Paper-Weather-Display/issues/102#issuecomment-630969193

All necessary changes are named there to get the V2 7.5” displays to work. Good luck

ag-ni, 2020-06-11:
@AWSW2017 , @G6EJD : Thanks a lot for comments. Let me try it.

G6EJD, 2020-06-11:
@ag-ni don't forget to select the correct display driver object too.

[19 more replies]