G6EJD/ESP32-e-Paper-Weather-Display #73: WiFi - authentication failure

hoffmakl1961, 2019-07-21:
On the serial monitor I've seen a faw WiFi - authentication failure.  
For testing I reduced the sketch - and changed the deep sleep mechanism 
- wakeup every minute (later for a BME280 extension) and every 30 min download of OWM - data. 

Sketch: 
[WakeUpTimer_WLAN.zip](https://github.com/G6EJD/ESP32-e-Paper-Weather-Display/files/3414626/WakeUpTimer_WLAN.zip)

The StartWiFi() produce a lot of  ...  
	_eventCallback(): Event: 5 - STA_DISCONNECTED
	_eventCallback(): Reason: 202 - AUTH_FAIL. 

I've tested with different boards, but get the same "AUTH_FAIL". 
       - NodeMCU ESP32
       - HIMALAYA ESP32 Dev Board
       - Universal e-Paper Raw Panel Driver Board, ESP32. 
 
Monitor output: 
[Monitor output.zip](https://github.com/G6EJD/ESP32-e-Paper-Weather-Display/files/3414623/Monitor.output.zip)

G6EJD, 2019-07-21:
That’s an ESP32 to Router problem rather than code. Have you considered if there’s any electrical noise around or there’s a Router problem?

hoffmakl1961, 2019-07-22:
The distance between WiFi-AP and ESP32 is 1,5m. In this area I've nevery problems with my mobile phones iPad, Surface and ThinkPad.   
I'm not sure if I'm having really problems with electrical noice in 2,4  or 5 GHz frequency range. But only the ESP32 have the problem no all others devices would have to do. 

An other test was with different fix WiFi channels.
channel 11 = 2,462 GHz and chanel 1 = 2,412 GHZ . 
I reconfigure my router and used only fix Wi-Fi channel 11 and a 2nd test with channel 1. 
For the ESP32 
I used the following command "WiFi.begin(ssid, password,chnl, bssid);"
chnl = WiFi channel, (fix) 
bssid= destination MAC - address (Router - WiFi access point.)

My Router shows also no problems with the used WiFi channel 11 
[image]

Have you really no problems? Can you test the Sketch?

G6EJD, 2019-07-22:
The sketch works OK:
Connecting to: SKYAE43F
6
6
6
6
3
3
WiFi connected at: 192.168.0.20
time: 16:11:00
Entering 60-secs of sleep time
Awake for : 0.383-secs
Starting deep-sleep period...

What silicon Revision is your board? 0 or 1, the compiler tells you that as part of its output, here's one of my Rev.0 boards output:
esptool.py v2.6
Serial port COM4
Connecting........__
Chip is ESP32D0WDQ6 (revision 0)

Rev.1 boards/devices are the reliable ones at startup.

What version of my software has an error?

hoffmakl1961, 2019-07-22:
I use a ESP32 with "Chip is ESP32D0WDQ6 (revision 1)".
How long do you testing the sketch (WakeUpTimer_WLAN.ino)? 1 min. ? 
about 50% of StartWiFi() end with AUTH_FAIL, as you can see in the Monitor output.zip
please let it run about 30 min.

hoffmakl1961, 2019-07-23:
I've tested again, and here is a output filter on AUTH_FAIL and WiFi connected at: 192.168.0.144

20:04:15.239 -> [W][WiFiGeneric.cpp:351] _eventCallback(): Reason: 202 - AUTH_FAIL
20:05:16.548 -> WiFi connected at: 192.168.0.144
20:06:02.169 -> WiFi connected at: 192.168.0.144
20:07:01.487 -> [W][WiFiGeneric.cpp:351] _eventCallback(): Reason: 202 - AUTH_FAIL
20:08:02.949 -> WiFi connected at: 192.168.0.144
20:09:01.312 -> [W][WiFiGeneric.cpp:351] _eventCallback(): Reason: 202 - AUTH_FAIL
20:10:02.765 -> WiFi connected at: 192.168.0.144
20:11:02.112 -> [W][WiFiGeneric.cpp:351] _eventCallback(): Reason: 202 - AUTH_FAIL
20:12:03.547 -> WiFi connected at: 192.168.0.144
20:13:01.902 -> [W][WiFiGeneric.cpp:351] _eventCallback(): Reason: 202 - AUTH_FAIL
20:14:03.345 -> WiFi connected at: 192.168.0.144
20:15:01.705 -> [W][WiFiGeneric.cpp:351] _eventCallback(): Reason: 202 - AUTH_FAIL
20:16:03.189 -> WiFi connected at: 192.168.0.144
20:17:01.564 -> [W][WiFiGeneric.cpp:351] _eventCallback(): Reason: 202 - AUTH_FAIL
20:18:03.003 -> WiFi connected at: 192.168.0.144
20:19:01.299 -> [W][WiFiGeneric.cpp:351] _eventCallback(): Reason: 202 - AUTH_FAIL
20:20:02.718 -> WiFi connected at: 192.168.0.144
20:21:02.091 -> [W][WiFiGeneric.cpp:351] _eventCallback(): Reason: 202 - AUTH_FAIL
20:22:03.549 -> WiFi connected at: 192.168.0.144
20:23:01.937 -> [W][WiFiGeneric.cpp:351] _eventCallback(): Reason: 202 - AUTH_FAIL
20:24:03.326 -> WiFi connected at: 192.168.0.144
20:25:01.719 -> [W][WiFiGeneric.cpp:351] _eventCallback(): Reason: 202 - AUTH_FAIL
20:26:03.206 -> WiFi connected at: 192.168.0.144
20:27:01.516 -> [W][WiFiGeneric.cpp:351] _eventCallback(): Reason: 202 - AUTH_FAIL
20:28:02.969 -> WiFi connected at: 192.168.0.144
20:29:01.280 -> [W][WiFiGeneric.cpp:351] _eventCallback(): Reason: 202 - AUTH_FAIL
20:30:02.734 -> WiFi connected at: 192.168.0.144
20:31:01.173 -> WiFi connected at: 192.168.0.144
20:32:01.578 -> [W][WiFiGeneric.cpp:351] _eventCallback(): Reason: 202 - AUTH_FAIL
20:33:02.953 -> WiFi connected at: 192.168.0.144
20:34:01.313 -> [W][WiFiGeneric.cpp:351] _eventCallback(): Reason: 202 - AUTH_FAIL
20:35:02.750 -> WiFi connected at: 192.168.0.144
20:36:02.111 -> [W][WiFiGeneric.cpp:351] _eventCallback(): Reason: 202 - AUTH_FAIL
20:37:03.442 -> [W][WiFiGeneric.cpp:351] _eventCallback(): Reason: 202 - AUTH_FAIL
20:38:04.922 -> WiFi connected at: 192.168.0.144
20:39:01.296 -> [W][WiFiGeneric.cpp:351] _eventCallback(): Reason: 202 - AUTH_FAIL
20:40:02.750 -> WiFi connected at: 192.168.0.144

G6EJD, 2019-07-23:
Interesting, but this is an ESP32 and compiler issue. I have some ideas on this surrounding the WiFi persistence settings and will post those later today.

G6EJD, 2019-07-23:
Please try adding these to the Wi-Fi Section:

 WiFi.setAutoConnect(true);
 WiFi.setAutoReconnect(true);

And run your experiment again

hoffmakl1961, 2019-07-24:
The Test was with 
`WiFi.setAutoConnect(true);
WiFi.setAutoReconnect(true);`

What is the reson that you declare `IPAddress dns(8, 8, 8, 8);  // Google DNS`

hoffmakl1961, 2019-07-24:
An other subject: 
I've created a BME280 extension based on Waveshare_7_5.
Can I upload the informations and Sketch under pull request?

closed 2019-07-24 by hoffmakl1961

G6EJD, 2019-07-24:
I declare a Domain Name Server (Google’s DNS) because after wake I notice it could not resolve logical name address of OWM and reported server not found, with an explicit DNS address it always work. Usually it would find a DNS via the router base address but I found that route also problematic. Overall there are issues either the ESP32 WiFi implementation. 

I’ll check out the push later.

G6EJD, 2019-07-26:
I forgot about your upload request, please create a pull request with a new file name

hoffmakl1961, 2019-07-27:
I'm not sure, how can I create a new Version with a pull request.   A compare changes make not sense.
here is a picture and a ppt with a short description. 
[image]
[Wetter Vxy.pptx](https://github.com/G6EJD/ESP32-e-Paper-Weather-Display/files/3437909/Wetter.Vxy.pptx)