G6EJD/ESP32-e-Paper-Weather-Display #76: compile error no matching function to call

vemeT5ak, 2019-09-01:
Hello, I'm trying to compile for the LolinD32. Error is  --- no matching function for call to 'GxEPD2_3C<GxEPD2_750, 384u>::getTextBounds(String&, int&, int&, int16_t*, int16_t*, uint16_t*, uint16_t*)'

Any ideas??? It's 2am and I'm fresh out of brain.

G6EJD, 2019-09-01:
Ok so all you need to do is copy my line, comment it out for reference, your compiling for the 3-colour variant and replace:
GxEPD2_BW<GxEPD2_750
With 
GxEPD2_BW<GxEPD2_750_3C

closed 2019-09-01 by G6EJD

vemeT5ak, 2019-09-02:
I tried your suggestion. That didn't work. Just leaving this here in case anyone else has the same issue. Turns out I had an old version of the Adafruit GFX library and the Adruino IDE was pulling from that. Deleting the old library and relying on the fresh download fixed the compile issue. I've got a lot of Arduino projects and some do require a specific version of a library, I tend to keep olds around. 

Thank you again for the help, 
Cheers

G6EJD, 2019-09-02:
Sometimes it’s worth checking the verbose compile option to see which file is being used for compilation, in your case it wasn’t the current version it seems. Thanks.

autodiag, 2020-02-08:
have same problem

test some version but still no luck to compile

how U8g2_for_Adafruit_GFX and Arduino version use ?
thanks

G6EJD, 2020-02-08:
What exactly is the error?

autodiag, 2020-02-08:
Arduino: 1.8.12 Hourly Build 2020/02/07 11:33 (Windows 10), Vývojová deska: "ESP32 Wrover Module, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), QIO, 80MHz, 921600, None"

D:\arduino workspace\ESP32-e-Paper-Weather-Display-master\examples\Waveshare_7_5\Waveshare_7_5.ino: In function 'void drawString(int, int, String, alignment)':

Waveshare_7_5:923:53: error: no matching function for call to 'GxEPD2_BW<GxEPD2_750, 384u>::getTextBounds(String&, int&, int&, int16_t*, int16_t*, uint16_t*, uint16_t*)'

   display.getTextBounds(text, x, y, &x1, &y1, &w, &h);

                                                     ^

In file included from C:\Users\Radio\Documents\Arduino\libraries\arduino_167194\src/GxEPD2_BW.h:15:0,

                 from D:\arduino workspace\ESP32-e-Paper-Weather-Display-master\examples\Waveshare_7_5\Waveshare_7_5.ino:26:

C:\Users\Radio\Documents\Arduino\libraries\Adafruit_GFX_Library/Adafruit_GFX.h:104:5: note: candidate: void Adafruit_GFX::getTextBounds(char*, int16_t, int16_t, int16_t*, int16_t*, uint16_t*, uint16_t*)

     getTextBounds(char *string, int16_t x, int16_t y,

     ^

C:\Users\Radio\Documents\Arduino\libraries\Adafruit_GFX_Library/Adafruit_GFX.h:104:5: note:   no known conversion for argument 1 from 'String' to 'char*'

C:\Users\Radio\Documents\Arduino\libraries\Adafruit_GFX_Library/Adafruit_GFX.h:106:5: note: candidate: void Adafruit_GFX::getTextBounds(const __FlashStringHelper*, int16_t, int16_t, int16_t*, int16_t*, uint16_t*, uint16_t*)

     getTextBounds(const __FlashStringHelper *s, int16_t x, int16_t y,

     ^

C:\Users\Radio\Documents\Arduino\libraries\Adafruit_GFX_Library/Adafruit_GFX.h:106:5: note:   no known conversion for argument 1 from 'String' to 'const __FlashStringHelper*'

D:\arduino workspace\ESP32-e-Paper-Weather-Display-master\examples\Waveshare_7_5\Waveshare_7_5.ino: In function 'void drawStringMaxWidth(int, int, unsigned int, String, alignment)':

Waveshare_7_5:933:53: error: no matching function for call to 'GxEPD2_BW<GxEPD2_750, 384u>::getTextBounds(String&, int&, int&, int16_t*, int16_t*, uint16_t*, uint16_t*)'

   display.getTextBounds(text, x, y, &x1, &y1, &w, &h);

                                                     ^

In file included from C:\Users\Radio\Documents\Arduino\libraries\arduino_167194\src/GxEPD2_BW.h:15:0,

                 from D:\arduino workspace\ESP32-e-Paper-Weather-Display-master\examples\Waveshare_7_5\Waveshare_7_5.ino:26:

C:\Users\Radio\Documents\Arduino\libraries\Adafruit_GFX_Library/Adafruit_GFX.h:104:5: note: candidate: void Adafruit_GFX::getTextBounds(char*, int16_t, int16_t, int16_t*, int16_t*, uint16_t*, uint16_t*)

     getTextBounds(char *string, int16_t x, int16_t y,

     ^

C:\Users\Radio\Documents\Arduino\libraries\Adafruit_GFX_Library/Adafruit_GFX.h:104:5: note:   no known conversion for argument 1 from 'String' to 'char*'

C:\Users\Radio\Documents\Arduino\libraries\Adafruit_GFX_Library/Adafruit_GFX.h:106:5: note: candidate: void Adafruit_GFX::getTextBounds(const __FlashStringHelper*, int16_t, int16_t, int16_t*, int16_t*, uint16_t*, uint16_t*)

     getTextBounds(const __FlashStringHelper *s, int16_t x, int16_t y,

     ^

C:\Users\Radio\Documents\Arduino\libraries\Adafruit_GFX_Library/Adafruit_GFX.h:106:5: note:   no known conversion for argument 1 from 'String' to 'const __FlashStringHelper*'

no matching function for call to 'GxEPD2_BW<GxEPD2_750, 384u>::getTextBounds(String&, int&, int&, int16_t*, int16_t*, uint16_t*, uint16_t*)'

G6EJD, 2020-02-08:
The error suggests the GFX library is not installed or done other libraries are missing, there is nothing wrong with the code, so you need to check your libraries