G6EJD/ESP32-e-Paper-Weather-Display #80: ESP32 freezing after WiFi conection is established

rmoravcik, 2019-11-15:
Hi all,

my TTGO Mini32 is sporadically freezing after it's woke up from deep sleep. The last print on serial console is, that WiFi connection was established.

I was able to trace down, that function call display.init() from InitialiseDisplay() sometimes never return.

I added some trace prints to GxEPD2 library and it looks like GxEPD2 is trying to initialize serial port if the first argument of display.init() is greater than 0. And this Serial.begin() inside of GxEPD2 sometimes never return.

I don't see any reason, why another Serial.begin() should hang ESP32, but there is a probably a bug in esp32-arduino or ESP SDK. I'm using version 1.0.4 of esp32-arduino.

So, after I've change code to:

```display.init(0);```

and my ESP32 started running stable.

What do you think about this modification?

Thank you,
Roman

G6EJD, 2019-11-22:
This looks like the GxEPD2 library has been changed, needs investigating.

rmoravcik, 2019-11-22:
Do you remember which version of GxEPD2 library are you using? I will try to do a test of serial reinitialization during the weekend to see if I'm able to reproduce the issue just with that.

G6EJD, 2019-11-22:
I’m not using the updated version yet

G6EJD, 2019-11-22:
Yes display.init(115200) now needs to be display.init(0); following GxEPD2 update.
Changes made

closed 2019-11-22 by G6EJD