G6EJD/ESP32-e-Paper-Weather-Display #82: Compile error: GxEPD2_BW.h

B0IIE, 2019-11-23:
Hi,

I wanted to flash my SBC-NodeMCU-ESP32 and get the error  GxEPD2_BW.h: No such file or directory
Heres the error message:
`Arduino: 1.8.10 (Windows 10), Board: "DOIT ESP32 DEVKIT V1, 80MHz, 921600, None"

Waveshare_4_2:43:23: error: GxEPD2_BW.h: No such file or directory

Multiple libraries were found for "Carousel.h"
compilation terminated.

 Used: C:\Users\B0IIE\Documents\Arduino\libraries\minigrafx-master
Multiple libraries were found for "FS.h"
 Used: C:\Users\B0IIE\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\FS
Multiple libraries were found for "SPIFFS.h"
 Used: C:\Users\B0IIE\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\SPIFFS
Multiple libraries were found for "SPI.h"
 Used: C:\Users\B0IIE\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\SPI
Multiple libraries were found for "ArduinoJson.h"
 Used: C:\Users\B0IIE\Documents\Arduino\libraries\ArduinoJson-6.x
Multiple libraries were found for "WiFi.h"
 Used: C:\Users\B0IIE\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\WiFi
 Not used: e:\Arduino\libraries\WiFi
exit status 1
GxEPD2_BW.h: No such file or directory

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
`

What am I missing / doing wrong?

G6EJD, 2019-11-23:
Have you updated GxEPD2 yet?

B0IIE, 2019-11-23:
That helped. Thanks.
But now I get:

> Error compiling for board DOIT ESP32 DEVKIT V1.

B0IIE, 2019-11-26:
I just did a clean intall of the Arduino IDE and set everything up according to your instructions but still cant get it to compile.

G6EJD, 2019-11-26:
What is the error, still the same?

B0IIE, 2019-11-26:
Thats the error message:
> > Arduino: 1.8.10 (Windows 10), Board: "ESP32 Dev Module, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, None"
> 
> Multiple libraries were found for "ArduinoJson.h"
>  Used: C:\Users\B0IIE\Documents\Arduino\libraries\ArduinoJson
> Multiple libraries were found for "WiFi.h"
> In file included from C:\Users\B0IIE\Documents\Arduino\libraries\ESP32-e-Paper-Weather-Display-master\examples\Waveshare_2_9\Waveshare_2_9.ino:27:0:
> 
>  Used: C:\Users\B0IIE\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\WiFi
> C:\Users\B0IIE\Documents\Arduino\libraries\GxEPD2\src/GxEPD2_BW.h:15:26: fatal error: Adafruit_GFX.h: No such file or directory
> 
>  Not used: E:\Arduino\libraries\WiFi
> compilation terminated.
> 
> Multiple libraries were found for "SPI.h"
>  Used: C:\Users\B0IIE\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\SPI
> Multiple libraries were found for "GxEPD2_BW.h"
>  Used: C:\Users\B0IIE\Documents\Arduino\libraries\GxEPD2
> exit status 1
> Error compiling for board ESP32 Dev Module.
> 
> This report would have more information with
> "Show verbose output during compilation"
> option enabled in File -> Preferences.
>

G6EJD, 2019-11-26:
You haven’t installed adafruit_gfx

B0IIE, 2019-11-26:
Thanks.
In addition to that I also needed to install U8g2_for_Adafruit.
(Maybe you want to include these libraries in the description)

And I had to remap the Pins to the following NodeMCU-ESP32 but than it worked:

> // Connections for e.g. LOLIN D32 
> static const uint8_t EPD_BUSY = 15;  // to EPD BUSY
> static const uint8_t EPD_CS   = 2;  // to EPD CS
> static const uint8_t EPD_RST  = 4; // to EPD RST
> static const uint8_t EPD_DC   = 5; // to EPD DC
> static const uint8_t EPD_SCK  = 18; // to EPD CLK
> static const uint8_t EPD_MISO = 19; // Master-In Slave-Out not used, as no data from display
> static const uint8_t EPD_MOSI = 23; // to EPD DIN

Thank you very much for this great project and for the help for my beginner questions.

closed 2019-11-26 by B0IIE