G6EJD/ESP32-e-Paper-Weather-Display #85: [4.2" v12.2] Temperature: numeric forecast does not match forecast graph?

PieBru, 2019-12-28:
Hi David,
I didn't dig jet into OWM json or your Arduino code, but as you can see in the attached picture the numeric temperature forecast (+9°|-4°) does not match the graph below, which shows circa +7°|+3° .
Also, the rainfall graph at bottom doesn't show the rain expected for the next 3 hours on top.
Is this intentional? Did I miss something?
Thank you,
Piero

[image]

G6EJD, 2019-12-28:
Hi Piero, it’s not my programming it’s an issue with OWM, the current temperature or pressure never matches the forecast values, clearly one (actual) is fact and the forecast data is derived from a forecast algorithm. Often they forecast rain but don’t provide any rain data to be displayed.
I’m not aware of any code issues.

PieBru, 2019-12-28:
Btw, in the picture above the current temperature and the forecast graphs correspond to other local forecast services, i.e.: [1](https://www.arpa.veneto.it/previsioni/it/html/index.php), [2](3bmeteo.com), [3](ilmeteo.it).
The min/max temperature and the 3 hours forecast are wrong, as today here it's a beautiful sunny day.
Happy new year,
Piero

G6EJD, 2019-12-28:
Yes I have observed the same here in the UK, but on average the data tends to be correct, it’s the forecast algorithms that need improving. My impression is that unlike the now closed to new users Weather Underground service there is a lack of data feeds to ensure accuracy, for example my weather station reports to the Weather Underground server system and their forecasts are always good and data is consistent. The only other free access data is from Darksky.com and I have a 7.5” version able to use that source, but I have noticed it too lacks any consistency.

PieBru, 2019-12-28:
Good advice.
Did you try [WeatherBit](https://www.weatherbit.io/pricing)?

PieBru, 2019-12-29:
Home Assistant switched to met.no , it's free and it seems reliable and coherent.
https://www.met.no/en/free-meteorological-data
https://www.home-assistant.io/integrations/met

In my house the weather display is used a lot so coherent and hopefully reliable info is a must.

Weather forecasting data depends on the data sources and model(s) they apply. This summer I used the the free [windy.com](https://www.windy.com/) android app which has been pretty good with just the default model, plus real-time lightning, wind and other goodies. It has been invaluable to plan my RV outdoor activities. You can choose between a bunch of models, each one's peculiarity is described in the list. Unfortunately, while windy.com web site and android app are free, their API has a free trial period but it isn't free.

I like a lot the ePaper+ESP32 cable-less concept, but for the same cost of a 7.5" ePaper display and an ESP32 I can dedicate a 10" cheap android tablet as a permanent windy.com display, with energy-savings enabled when the display isn't active. And meanwhile the tablet can do a lot of other things.

Just my 2 cents...
Piero

PieBru, 2019-12-29:
I just noticed a free API service, the phone number and DNS looks UK:
https://developer.weatherunlocked.com/localweather/pricing

PieBru, 2019-12-30:
Here is a working example of location forecast API call to met.no for London
$ curl 'https://api.met.no/weatherapi/locationforecast/1.9/?lat=59.2735&lon=-0.1275&msl=0'

They have other free API services: https://api.met.no/weatherapi/documentation
I found met.no data is one of the most reliable services, at least for Europe.

G6EJD, 2020-02-03:
Ive tried Darksky and their data is more coherent than OWM

closed 2020-02-03 by G6EJD