G6EJD/ESP32-e-Paper-Weather-Display #89: Confusing wiring jpg

systeembeheerder, 2020-02-04:
https://raw.githubusercontent.com/G6EJD/ESP32-e-Paper-Weather-Display/master/Schematic.jpg

The diagram shows ESP pin 19 connected to ePaper busy. The table also says 19. The text on the left however shows pin 4 for busy and 19 as not used.

https://raw.githubusercontent.com/G6EJD/ESP32-e-Paper-Weather-Display/master/Lolin32_Wiring.JPG
and
https://raw.githubusercontent.com/G6EJD/ESP32-e-Paper-Weather-Display/master/Lolin32_Lite_Wiring.JPG

Both use pin 4 for busy.

Please fix Schematic or remove it in favour of the both Lolin32 jpg's.

G6EJD, 2020-02-04:
First of all the now obsolete ESP32 lite is demonstrating that you can use any pins and in that example they are grouped together for ease of wiring, there is no standard to conform to. In the other wiring diagrams using the current D32 boards I have opted for a different arrangement. I don’t see any benefit in re-doing the ESP32 lite version?

systeembeheerder, 2020-02-04:
Correct me if i am wrong; but shouldn't https://raw.githubusercontent.com/G6EJD/ESP32-e-Paper-Weather-Display/master/Schematic.jpg say

`//static const uint8_t BUSY = 19`

instead of 

`//static const uint8_t BUSY = 4`

to match the drawing/scematic?

G6EJD, 2020-02-04:
I've deleted the image as there is a correct wiring diagram for the ESP32 Lite

closed 2020-02-04 by G6EJD