G6EJD/ESP32-e-Paper-Weather-Display #91: 800x480 update?

systeembeheerder, 2020-02-07:
Waveshare updated the 7.5" inch e-Paper from 640x384 pixels to 800x480. The current code does not work for this version. Changing:

`GxEPD2_BW<GxEPD2_750, GxEPD2_750::HEIGHT> display(GxEPD2_750(/*CS=*/ EPD_CS, /*DC=*/ EPD_DC, /*RST=*/ EPD_RST, /*BUSY=*/ EPD_BUSY));   // B/W display`
to 
`GxEPD2_BW<GxEPD2_750_T7, GxEPD2_750_T7::HEIGHT> display(GxEPD2_750_T7(/*CS=*/ EPD_CS, /*DC=*/ EPD_DC, /*RST=*/ EPD_RST, /*BUSY=*/ EPD_BUSY));   // B/W display`
makes it works, but does not use the extra pixels. Is seems all the dimensions are hard coded.

G6EJD, 2020-02-07:
You need to post this issue on the GxEPD driver library page, but don’t expect an instant response as my experience is unless the author has a display to test it, he won’t make the changes. If you provide him with a display then the changes are likely to be made and if Waveshare releases the design data to enable this. Good luck.

closed 2020-02-07 by G6EJD

systeembeheerder, 2020-02-07:
GxEPD supports this just fine. I have a working display and code. Should I create a pull request?

G6EJD, 2020-02-07:
Most of the current x,y coordinates would need to be scaled by 800/640 for the x axis and 600/384 for the y-axis, so if you feel confident go ahead.