G6EJD/ESP32-e-Paper-Weather-Display #94: Updated Graps: Some code

charno, 2020-03-12:
Hi,

I've used your code to write a weather station that does not only display the forecast, but also prints some data from (homemade) temp/humidity sensors. Most of the code is pretty specific, but I think one piece of code you might be interested in is an updated graph-drawing method.

[image]

Improvements:
- Allow multiple datasets per graph (one `struct reading` per dataset)
- Allow to set color per dataset (`reading.color`)
- Show midnight-spacers (the small lines coming from the bottom into the graph)
- Print the actual date on the bottom (instead of 0, 1, 2).  `bool historicalData` can be set to print dates from the past.
- Make the dotted lines finer

Since the integration of this changes need some changes on the calling functions, I don't want to simply open a pull request. I put the code there: https://gist.github.com/charno/cc8dcf5443a5760c63d0384358b395b4

Best regards
Christof

G6EJD, 2020-03-12:
Hi Christof, how scaleable are the resultant graphs, can you reduce their sizes, and I don’t clearly see how to select the x-axis units that can always be record counts but not always time based, so I’m wondering how you manage that, did you add time to the struct parsed? What I try to do is make all the code as easy to understand for everyone of varying skills and right now I’m struggling to see how it’s being done, I think there are some wider changes that need to be explained.

charno, 2020-03-12:
Hi G6EJD,

You are right, the code got slightly more complicated.

I started off with the code from your 7.5 inch example, and I did keep a lot of it. 

Scaleability didn't change. You can see two examples of different sizes in the picture above.
The upper graphs have width = 290, height = 145.
The lower graphs have width = 120, height = 58.

I'm simply assuming a time-base of 3 days, either from now or until now. This is what I did assume was already done, since in the old code there was "0 1 2" printed below the graph. 
The different data rows don't have the same amount of entries, but the timespan of the data is (currently) fixed to three days in my code.

To draw the humidity graph on the bottom, I simply changed the call 
```
DrawGraph(gx + 2 * gap, gy, gwidth, gheight, 0, 100,   TXT_HUMIDITY_PERCENT, humidity_readings, max_readings, autoscale_off, barchart_off);
```
to
```
DrawGraph(gx + 2 * gap, gy, gwidth, gheight, 0, 100,   TXT_HUMIDITY_PERCENT, autoscale_off, barchart_off, {{humidity_readings, max_readings, GxEPD_RED}});
```

And a (not so senseful, since the values have different units) example to call for multiple data rows would be
```
DrawGraph(gx + 2 * gap, gy, gwidth, gheight, 0, 100,   TXT_HUMIDITY_PERCENT, autoscale_off, barchart_off, {{humidity_readings, max_readings, GxEPD_RED}, {temperature_readings, max_readings, GxEPD_BLACK}});
```

MorbidSerenity, 2020-03-18:
Hi Charno, can you give some details on how you used your (homemade) temp/humidity sensors? Would like to do this for my self.

charno, 2020-03-18:
Hi @MorbidSerenity 

Yes of course.

The sensors are esp32 boards with a dht22 temperature sensor. They make a measurement every 5 mins, and send these to a mqtt broker every 30 minutes. In between they are in deep sleep mode, the 150ma battery lasts about 1.5 to two weeks depending on temperature. I wrote this code myself, it got quite (unnecessarily) complex so it's not something I think is fit for sharing. But there are lots of tutorials out there showing how to publish values to a mqtt broker.

The server setup is VERY similar to this: https://dzone.com/articles/playing-with-docker-mqtt-grafana-influxdb-python-a, with grafana to show the results in a browser.

For the display I added some code which uses the ArduinoJson library to directly query the influxdb.
On arduinojson webpage there is a very handy tool where you can enter some json, and it will give you the parsing code. To get the json you can enter there, you can use the grafana query editor to output the result.

Here is some (shortened) code I use:
```
String request = "SELECT mean(\"temperature\") FROM \"weather\" WHERE (\"sensor\" =~ /^(SENSORNAME)$/) AND time >= now() - 3d GROUP BY time(30m), \"sensor\" fill(previous)";
      HTTPClient http;
      String uri = "/query?db=sensors&q=" + url_encode(std::string(request.c_str()));

      http.begin(client, "<server ip>", <port>, uri);
      int httpCode = http.GET();
      if(httpCode == HTTP_CODE_OK) {
        Serial.println("connection successful");
        const size_t capacity = 2*JSON_ARRAY_SIZE(1) + 146*JSON_ARRAY_SIZE(2) + JSON_ARRAY_SIZE(145) + 2*JSON_OBJECT_SIZE(1) + JSON_OBJECT_SIZE(2) + JSON_OBJECT_SIZE(4) + 3480; // this can be calculated using the arduinojson webpage
        DynamicJsonDocument doc(capacity);
        DeserializationError error = deserializeJson(doc, http.getString().c_str());
        // Test if parsing succeeds.
        if (error) {
          Serial.print(F("deserializeJson() failed: "));
          Serial.println(error.c_str());
          return false;
        }

          for(int i = 0; i < max_readings_sensor; i++)
          {
              tempInside[i] = doc["results"][0]["series"][0]["values"][i][1];
          }
        client.stop();
        http.end();
      }
      else
      {
        Serial.printf("connection failed, error: %s", http.errorToString(httpCode).c_str());
        client.stop();
        http.end();
        return false;
      }
```

Sorry for the formatting, I deleted some more branches (other sensors, other measurements etc.) so the intendation is a bit broken.

closed 2020-04-03 by G6EJD