G6EJD/ESP32-e-Paper-Weather-Display #95: ESP32 Arduino compile issue

Courty40, 2020-03-15:
Not had a chace to have a good look but im getting this error.

"no matching function for call to 'GxEPD2_BW<GxEPD2_420, 300u>::getTextBounds(String&, int&, int&, int16_t*, int16_t*, uint16_t*, uint16_t*)'"

Im using the 4.2 Waveshare display version

G6EJD, 2020-03-16:
Not all libraries are installed

Courty40, 2020-03-16:
Got to the bottom of the issue. it seems that earlier Adafruit GFX libs have a different name to newer ones. 
I had installed 1.7.5 (newest) but looking at the detailed info on the compiler, it was using 1.2.0 ?!
Found the Adafruit GFX folder in the library and deleted it (the new one installs as arduino_148920 on my box) and all works fine..
Courty

G6EJD, 2020-03-16:
Good to hear it’s resolved, that’s a GFX function so I knew it was the issue.

closed 2020-03-16 by G6EJD