G6EJD/ESP32-e-Paper-Weather-Display #96: error when compiling

rgot-org, 2020-03-18:
hello,
i get an error when compiling : 
`Waveshare_7_5.ino: 92:44: error: 'alignment' has not been declared` 
`Waveshare_7_5.ino: 93:77: error: 'alignment' has not been declared`

but when i edit around the line 92 i find this code :
```c {.line-numbers}
#define max_readings 24  // line 88
 //89
Forecast_record_type  WxConditions[1];   // 90
Forecast_record_type  WxForecast[max_readings]; //91
                 //92
#include "common.h"    //93
                                   //94
#define autoscale_on  true
#define autoscale_off false
#define barchart_on   true
#define barchart_off  false

```
and the alignment enum is declared at line 42
```c
enum alignment {LEFT, RIGHT,CENTER};
```

please some help...

G6EJD, 2020-03-18:
There are no code issues.
How did you copy the code? Browsers often replace “ with a backwards version, did you use the zip file?

G6EJD, 2020-03-19:
Update on this issue?

rgot-org, 2020-03-20:
i am using  the plugin visual micro for visual studio .
the compilation crashes because the compilator don't support `//#######` comment lines
i've replaced all the # with * then it compiles like a charme.
also i've add ` #include <Arduino.h>` at the top of `owm_credentials.h`.
you can close this issue

closed 2020-03-20 by rgot-org