G6EJD/ESP32-e-Paper-Weather-Display #97: Multiple Refreshes

Courty40, 2020-04-02:
I have my Weather station running perfectly now on the 4.2in display and a custom ESP32 board.
While its not missed a beat since I got it working, I have now noticed some odd behaviour and was wondering if anyone else has the same..
I've noticed on the hour that the first refresh will run at for example 11:59:30 and then the refresh will run again at 12:00:05. While this isn't a problem for the use of the device, it must be using extra power and resource ?
Thanks for a great project, loving having it on my desk especially while having to work from home !

Paul
G1ROK

G6EJD, 2020-04-02:
Hi Paul, the update timing relies on a fairly accurate ESP32 timer to start the refresh after the designated Wait time, usually 30-minutes or an hour. But, the RC timer (150khz) used by the ESP32 is quite inaccurate and on some boards this leads to the ESP32 duly waking up after the time-out, then the first thing it does is check the time from NTP and it’s usually early! Then it runs through the data reception and finally calculates the sleep time to the next boundary say every 30-mins or hour, but if early say by 10-secs it calculates a sleep time of 10-secs and so wakes again a short time (10-secs) later, and hence you get this odd behaviour. The solution is (and it does depend on the individual ESP32) to add a further delay to the final sleep time value so that a fast running RC clock is delay further until after the designated time boundary, this means that the final update time will not be near the designated boundary, maybe 30-secs later.

I’ve noticed some ESP32 exhibit this behaviour, some not and it is ambient temperature related. Me being slightly ocd, could usefully drop the display of seconds, so if the update time was say 23:00:30 who cares! but I have trouble with that 😀 most of my ESP32 cpus are quite accurate so this rarely happens. Some development boards have a 32khz xtal fitted to improve RTC accuracy but this can’t be enabled in the Arduino IDE. The SparkFun and DFRobot boards are two that come to mind.

A bit long, but I hope it explains what’s going on, 73 Dave

G6EJD, 2020-04-02:
esp_sleep_enable_timer_wakeup((SleepTimer+30) * 1000000LL);

Would add 30-secs to the overall timing used. It looks like your board in certain conditions may need to be delayed even more, as it has a very fast RC timer running.

closed 2020-04-03 by G6EJD

Courty40, 2020-04-06:
I can confirm its been running for a couple of days and its working fine. may be worth just putting  20 secs on the timer by default ?

Paul

G6EJD, 2020-04-06:
Yes, I agree, I’ll look at the code tomorrow.

G6EJD, 2020-04-07:
Done that now.