G6EJD/ESP32-e-Paper-Weather-Display #98: Compilation of Waveshare_7_5_T7.ino fails for GxEPD2_750c_Z08 (7.5" b/w/r 800x480)

ristomatti, 2020-04-15:
I'm having issues compiling for Waveshare 7.5" color display using PlatformIO. The compilation fails with the below message:

```
|-- <WiFi> 1.0
|-- <Wire> 1.0.1
Building in release mode
Compiling .pio/build/esp32dev/src/main.cpp.o
Linking .pio/build/esp32dev/firmware.elf
/home/ristomatti/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/esp32dev/firmware.elf section `.dram0.bss' will not fit in region `dram0_0_seg'
/home/ristomatti/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: DRAM segment data does not fit.
/home/ristomatti/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: region `dram0_0_seg' overflowed by 17200 bytes
collect2: error: ld returned 1 exit status
*** [.pio/build/esp32dev/firmware.elf] Error 1
========================================== [FAILED] Took 7.75 seconds ==========================================
```

So it appears the code doesn't fit to DRAM. Any tips on reducing the memory consumption?

I was able to compile it with `#include <GxEPD2_BW.h>` and using the `GxEPD2_3C<GxEPD2_750c_Z08, GxEPD2_750c_Z08::HEIGHT> display` constructor. However using the graphics/text end up being dim red. When the screen refreshes it goes through a nice crisp black which it then replaces with dark red and eventually a pale red color.

To compile I have done the following changes:

1. Added library dependencies to `platformio.ini`:

```ini
lib_deps =
  ArduinoJson@6.15.1
  GxEPD2@1.2.7
  U8g2_for_Adafruit_GFX@1.7.0
  https://github.com/G6EJD/ESP32-e-Paper-Weather-Display.git
```

2. Added forward declarations to functions (PIO compiles as `.cpp`)

3. Included `GxEPD2_3C.h`:
```cpp
// #include <GxEPD2_BW.h>
#include <GxEPD2_3C.h>
```
4. Changed display initialization:
```cpp
// GxEPD2_BW<GxEPD2_750_T7, GxEPD2_750_T7::HEIGHT> display(GxEPD2_750_T7(/*CS=*/ EPD_CS, /*DC=*/ EPD_DC, /*RST=*/ EPD_RST, /*BUSY=*/ EPD_BUSY));   // B/W display
GxEPD2_3C<GxEPD2_750c_Z08, GxEPD2_750c_Z08::HEIGHT> display(GxEPD2_750c_Z08(/*CS=*/ EPD_CS, /*DC=*/ EPD_DC, /*RST=*/ EPD_RST, /*BUSY=*/ EPD_BUSY)); // 3-colour displays
```

G6EJD, 2020-04-15:
This is not a code issue with this software, it's a hardware issue when trying to drive a 3-colour display. There has to be enough RAM to hold programme variables and 3xscreen buffers rather than 2 for B/W. You can use the half-size buffer options of the GxEPD2, otherwise there is little than can be done to solve the problem. I have already moved as many constants and little used variables to flash, and in any event adding more won't save a screen full of RAM required to drive the display.

closed 2020-04-15 by G6EJD

ristomatti, 2020-04-15:
It wasn't my intention to indicate this as a code problem but rather ask a question. I will look into the half-size buffer options, thanks for the tip and the quick response!

G6EJD, 2020-04-15:
OK, I understand, try the half-size buffer option, it works. Each screen (B/W/R) requires about 30K so ~90K total, so actually you should have enough RAM. The JSON decoding should drop it's buffer after use, so before drawing, maybe try display the heap and stack size on the serial port to see what the issue might be, may not be memory.

ristomatti, 2020-04-15:
@G6EJD I admit this is a more than a little over my head but the code compiles after changing the constructor to:

```cpp
GxEPD2_3C<GxEPD2_750c_Z08, GxEPD2_750c_Z08::HEIGHT / 2> display(GxEPD2_750c_Z08(/*CS=*/ EPD_CS, /*DC=*/ EPD_DC, /*RST=*/ EPD_RST, /*BUSY=*/ EPD_BUSY)); // 3-colour displays
```

The display also updates to black color but (unsurprisingly) draws only the top side of the display. Maybe this was not what you meant? :grin:

ristomatti, 2020-04-15:
Nevermind, I found the link to https://github.com/olikraus/u8glib/wiki/tpictureloop from GxEPD2 documentation. I managed to get the display to draw after replacing `display(false);` in `void setup()` with the below snippet. I suppose this is what you were after? It seems to work pretty much just as fast.

```cpp
      if (RxWeather && RxForecast)
      {             // Only if received both Weather or Forecast proceed
        StopWiFi(); // Reduces power consumption
        display.firstPage();
        do {
          DisplayWeather();
        } while (display.nextPage());
      }
```

G6EJD, 2020-04-15:
OK, glad you go it running OK.