G6EJD/ESP32-e-Paper-Weather-Display #103: platformio unable to compile

loulou4cv, 2020-04-30:
Hello,

the program is working fine in ArduinoIDE, however I'm unable to compile it in platformio, do you have same problem please ?

.platformio/lib/Adafruit STMPE610_ID377/Adafruit_STMPE610.cpp:32:18: fatal error: Wire.h: No such file or directory

Thank you.

G6EJD, 2020-04-30:
You have to include all libraries eg Wire or its equivalent in platformio, it may not be called wire and therefore all Adafruit libraries are likely to require modifications.

closed 2020-05-01 by G6EJD

loulou4cv, 2020-05-02:
Hello, 
finally found the solution. The #include <Wire.h> should work but it doesn't, you have to modify the platformio.ini to force it (no need to force other libraries). Then platformio.ini should look like :

[env:esp32doit-devkit-v1]
platform = espressif32
board = esp32doit-devkit-v1
framework = arduino
upload_speed = 921600
monitor_speed = 115200

lib_deps =
  Wire@~1.0.1

Now it works.

G6EJD, 2020-05-02:
Good to hear you solved it.