G6EJD/ESP32-e-Paper-Weather-Display #104: 2.9" epaper display

d0ugak, 2020-05-01:
Is this actually confirmed to work on the 2.9" e-paper displays? I have successfully compiled this and uploaded to my ESP32 with the display attached. On power up nothing is happening. I only see the following in the serial output. I have triple checked the connections between the epaper display and esp32 and they are correct per the jpg you have posted.

I notice there are no screen shots of the 2.9 display, so made me wonder if it was actually tested.

`rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5816
entry 0x400806ac

Connecting to: AndroidAP
WiFi connected at: 192.168.43.11
Entering 921-secs of sleep time
Awake for : 0.685-secs
Starting deep-sleep period...
`

d0ugak, 2020-05-01:
I'll also add that I have configured and compiled the GxEPD2 example for the 2.9" and this is working. So nothing wrong with the display or wiring.

G6EJD, 2020-05-01:
Even if the display was disconnected it would still receive and report all the weather data on the serial monitor port, and it’s getting nothing, so do you have a working API key?
Yes, the code is tested.

d0ugak, 2020-05-01:
I do have a working key.

I tested with their example API call http://api.openweathermap.org/data/2.5/forecast?id=524901&APPID={APIKEY} and get a data return

d0ugak, 2020-05-01:
I don't know why, but it is working now. Perhaps openweather was having some kind of an outage yesterday at the time I was playing around with this. This issue can be closed

G6EJD, 2020-05-01:
It was OWM server outage, I had the same

closed 2020-05-01 by G6EJD