G6EJD/ESP32-e-Paper-Weather-Display #105: Some cosmetic changes for the 7.5"-T7 version (800x480px) plus small fix for DE language

AWSW2017, 2020-05-21:
Hello, 
i did some cosmetic changes for the T7 version of the 7.5" displays with 800x480px and some small fixes for the DE language to avoid english/german text mixtures.

In version 16.8 the section boxes and some texts are not all arranged all the same and show some gaps:
[image]

So i had a look and rearranged them to look a bit cleaner (to me):
[image]

In addition i made some small changes listed below and tested them in german and english language:
[image]

[image]

I allowed me to call it "Version 16.9".  Changes done:

1. Some smaller cosmetic changes regarding the position of the linees of the section boxes, e.g. there were some small gaps between the lines and the outer bottom lines were missing at all
2. Smaller position changes for the shown texts, e.g. the "RH" values was 10px off the label
3. Small changes in the lang_gr.h because of false translations, e.g. for TXT_PRECIPITATION_SOON was set "in Kürze" which means "comming soon" and needs to be "Niederschlag"
4. Modified void DisplayForecastTextSection to avoid german+english text mixtures in the string. Tested with DE and EN language setting

Here are the files:
- [lang_gr.h.zip](https://github.com/G6EJD/ESP32-e-Paper-Weather-Display/files/4661623/lang_gr.h.zip)
- [Waveshare_7_5_T7.ino.zip](https://github.com/G6EJD/ESP32-e-Paper-Weather-Display/files/4661624/Waveshare_7_5_T7.ino.zip)

Maybe you want to use these changes?

Kind regards :)

AWSW2017, 2020-05-22:
Minor Bugfix for the TXT_RAINFALL_MM text y-position:

[image]

[Waveshare_7_5_T7.ino.zip](https://github.com/G6EJD/ESP32-e-Paper-Weather-Display/files/4667877/Waveshare_7_5_T7.ino.zip)

G6EJD, 2020-05-22:
There was a proposed change way back, that removed any rainfall icons if there was no rain, I myself prefer it to show the icon.

One change worth considering is the sleep delay at night, as currently it still sleeps for ‘sleep time’ wakes up and if in the sleep period goes back to sleep, it would be better to sleep for say 8-hours then check the time, saves 1/3 of a days power consumption. Just not had the time to do it. Really it’s duration = Sleeptime - Waketime with some tests around the timing. E.g 2300-0700 = sleep time

AWSW2017, 2020-05-22:
Hello, the text "x.yz mm" was shown in the topic text of the box header. Just set the text value to y 40px like the raindrop icon y value. There simply were not in line... Yesterday was sun all day here so i could not check this text position...

Good point. I thought of the for my actual calendar project too, because this is a huge amount of time it could stay in deep sleep for each night... For my calendar i changed the setup void so that it  checks the time via wifi only and then falls to sleep again. I think it is similar here. A while ago i read an entry in a EPS32 forum that the max time of deep sleep would be 4 hours. That can't be true, because my random picture frames only wake up every 8 hours to download a 400-800 kByte picture. During night times they will check the time only if they wake up and recalculate then the time until the set wakeup time again and go to deep sleep. That saves much power... I will think about that too... 

Actually i am struggling with Arduino that you seem not to be able to count with variables within a for loop. I need to fill up to 135 text variables and all loop actions do not seem to be supported... :(

G6EJD, 2020-05-22:
The ESP32 can sleep for 2^64 uS or 213,503,982 days! The timer variable has to be defined as a Long. Thanks for the other corrections.

You should have no problems with using variables with a for loop. Ideally define them before entering the for loop. What is going wrong? Try a while x < value loop and say x++ as your loop code counter to eliminate the for loop as an issue.

AWSW2017, 2020-05-23:
Good to know... I think this amount of deep sleep should last long enough. 😂 Thanks. You are welcome. Let me know if I can help.

I think I’ll use a 2-dimensions array. I thought I could simply up a variable like „field(i)text(y) = Text“ to set up to 135 variables for the calendar project in a for loop, but that does not seem to be possible in Arduino to do. I‘ll the the array these days. Thanks a lot.

G6EJD, 2020-05-23:
Looks like you need to do:
typedef struct record
  {
      String Day;
      int DaysInMonth;  
  } record;

record Month[12]];

void loop(){
  for (int m=1; m<13;m++){
    Month[m].Day = "January";
    Month[m].DaysInMonth = 31;
  }
}

G6EJD, 2020-07-20:
It will do if you have the correct pin definitions as listed.

closed 2020-07-20 by G6EJD