G6EJD/ESP32-e-Paper-Weather-Display #106: Some cosmetic changes for the 7.5" version (640x384px) plus small fix for DE language

AWSW2017, 2020-05-21:
Hello,
i did some cosmetic changes for the classic version of the 7.5" displays with 640x384px and some small fixes for the DE language to avoid english/german text mixtures.

In version 16.8 the section boxes and some texts are not all arranged all the same and show some gaps:
[image]

So i had a look and rearranged them to look a bit cleaner (to me):
[image]

In addition i made some small changes listed below and tested them in german and english language:
[image]

[image]

I allowed me to call it "Version 16.9". Changes done:
1. Some smaller cosmetic changes regarding the position of the lines of the section boxes, e.g. there were some small gaps between the lines.
2. Smaller position changes for the shown texts
3. Small changes in the lang_gr.h because of false translations, e.g. for TXT_PRECIPITATION_SOON was set "in Kürze" which means "coming soon" and needs to be "Niederschlag". It is the same file like reported in #105 
4. Modified void DisplayForecastTextSection to avoid german+english text mixtures in the string. Tested with DE and EN language setting 

Here are the files:
- [lang_gr.h.zip](https://github.com/G6EJD/ESP32-e-Paper-Weather-Display/files/4663382/lang_gr.h.zip)
- [Waveshare_7_5.ino.zip](https://github.com/G6EJD/ESP32-e-Paper-Weather-Display/files/4663385/Waveshare_7_5.ino.zip)

Maybe you want to use these changes?

Kind regards :)

G6EJD, 2020-05-21:
Yes, but before I publish them, I want to suggest a removal of the forecast text that OWM issue.  They have the main forecast text c and then include a forecast text for upto two more local areas, the forecast in (), plus there can be some really long text that does not wrap that well at present, I’ll upload a revised function to do that soon.

AWSW2017, 2020-05-21:
Thanks a lot in advance. Take your time. I’m happy with this so far.

BTW: I used the 2ms delay setting done in the GxEPD2 file for the old HATs and old display here too and seems to work (#102).

closed 2020-07-20 by G6EJD