G6EJD/ESP32-e-Paper-Weather-Display #107: Hardcoded Language

bobybobybob, 2020-05-24:
Hello
the XXXXXXXX   is Hardcoded Language. It should also connected with the right language File in translation File.h 

Happy Coding and Thak you. 

drawString(SCREEN_WIDTH / 2, y + 180, "XXXXXXXXXX", CENTER);
  u8g2Fonts.setFont(u8g2_font_helvB10_tf);
  DrawGraph(SCREEN_WIDTH / 400 * 30,  SCREEN_HEIGHT / 300 * 221, SCREEN_WIDTH / 4, SCREEN_HEIGHT / 5, 900, 1050, "XXXXXXX", pressure_readings, max_readings, autoscale_on, barchart_off);
  DrawGraph(SCREEN_WIDTH / 400 * 158, SCREEN_HEIGHT / 300 * 221, SCREEN_WIDTH / 4, SCREEN_HEIGHT / 5, 10, 30, "XXXXXXX", temperature_readings, max_readings, autoscale_on, barchart_off);
  DrawGraph(SCREEN_WIDTH / 400 * 288, SCREEN_HEIGHT / 300 * 221, SCREEN_WIDTH / 4, SCREEN_HEIGHT / 5, 0, 30, "XXXXXX", rain_readings, max_readings, autoscale_on, barchart_on);

bobybobybob, 2020-05-24:
Yes, i like E-Paper Weater Display very much and my brother too.

G6EJD, 2020-05-24:
I don’t understand you, the code already uses a language file just select the file you need and edit it’s contents if required?

There is no code issue to report.

closed 2020-05-24 by G6EJD

bobybobybob, 2020-05-24:
No no, that three are hardcoded Text where language file not linked to translation.

bobybobybob, 2020-05-24:
https://www.bilder-upload.eu/upload/756f49-1590307267.jpg

G6EJD, 2020-05-24:
Updated now.

bobybobybob, 2020-05-24:
We say all thank you very much and have a happy day :-)