G6EJD/ESP32-e-Paper-Weather-Display #110: Rasing the deep sleep timer value when the SleepTime hour is reached to save more battery power

AWSW2017, 2020-05-30:
Hello,

quick and dirty solution with some extra serial output lines (for testing) to extend the deep sleep timer during night times until the WakeupTime hour is reached to save more battery power.

The SleepTimer value is raised for the amount of hours + minutes to the next full hour between SleepTime and WakeupTime. This will be done only when the exact SleepTime hour value is reached to keep it simple:

`//#########################################################################################
void BeginSleep() {
  display.powerOff();
  long SleepTimer = (SleepDuration * 60 - ((CurrentMin % SleepDuration) * 60 + CurrentSec)); //Some ESP32 are too fast to maintain accurate time

  // Calculate the sleep timer for night time to save more battery power. This will be done only when the exact SleepTime hour value is reached!
  // CurrentHour = SleepTime; // Test switch
  if (CurrentHour == SleepTime)  {
    Serial.println("Night time hour " + String(SleepTime) + " o'clock reached... I was told to sleep until " + String(WakeupTime) + " o'clock to save some battery power... Good night... zZzZzZ");

    Serial.print("Current hour:   ");
    Serial.println(CurrentHour);
    Serial.print("Current minute: ");
    Serial.println(CurrentMin);

    int SleepTimeUntilMidnight = 24 - SleepTime;
    Serial.print("SleepTimeUntilMidnight: ");
    Serial.println(SleepTimeUntilMidnight);

    int SleepTimeAfterMidnight = WakeupTime;
    Serial.print("SleepTimeAfterMidnight: ");
    Serial.println(SleepTimeAfterMidnight);

    int SleepTimerNightTimeHours = SleepTimeUntilMidnight + SleepTimeAfterMidnight;
    Serial.print("SleepTimerNightTimeHours: ");
    Serial.println(SleepTimerNightTimeHours);

    SleepTimer = ((60 * 60) * SleepTimerNightTimeHours) + (60 * (60 - CurrentMin));
    Serial.print("SleepTimer during night time: ");
    Serial.println(SleepTimer);
  }

  esp_sleep_enable_timer_wakeup((SleepTimer + 20) * 1000000LL); // Added extra 20-secs of sleep to allow for slow ESP32 RTC timers
#ifdef BUILTIN_LED
  pinMode(BUILTIN_LED, INPUT); // If it's On, turn it off and some boards use GPIO-5 for SPI-SS, which remains low after screen use
  digitalWrite(BUILTIN_LED, HIGH);
#endif
  Serial.println("Entering " + String(SleepTimer) + " seconds or " + String(SleepTimer / 60 ) + " minutes or " + String((SleepTimer / 60) / 60) + " hours of sleep time");
  Serial.println("Awake for : " + String((millis() - StartTime) / 1000.0, 3) + "-secs");
  Serial.println("Starting deep-sleep period...");
  esp_deep_sleep_start();      // Sleep for e.g. 30 minutes
}
//#########################################################################################`

[image]

HTH & enjoy your weekend

G6EJD, 2020-05-30:
Yes, good, that’s exactly how I did it. There are other methods like subtracting wake up time from sleep time modulo 24 but this approach is far simpler and I like that, saves a 1/3 (8/24) of a days running power extending battery life by nearly the same extent.

AWSW2017, 2020-05-30:
Glad you like it. 😊 Seems to run well. I use this similar in my new ePaper calendar project and seems to perform well.

closed 2020-06-09 by G6EJD

Jarod96, 2023-04-21:
@AWSW-de Thank you for sharing this, I was looking for a solution to that issue too but I'm just still a beginner in all of this!
I've managed to had it calculate the SleepTimer differently for someone how use more than an hour for the Timer.

int SleepTimerNightTimeHours = SleepTimeUntilMidnight + SleepTimeAfterMidnight;
            Serial.print("SleepTimerNightTimeHours: ");
            Serial.println(SleepTimerNightTimeHours);

            if (CurrentHour << SleepTime){
              SleepTimer = ((60 * 60) * (SleepTimerNightTimeHours - CurrentHour - 2)) + (60 * (60 - CurrentMin));
              Serial.print("SleepTimer during night time: ");
              Serial.println(SleepTimer);
              Serial.println("seconds.");
            }
            if (CurrentHour >= SleepTime){
              SleepTimer = ((60 * 60) * SleepTimerNightTimeHours) + (60 * (60 - CurrentMin));
              Serial.print("SleepTimer during night time: ");
              Serial.println(SleepTimer);
              Serial.println("seconds.");
            }

AWSW-de, 2023-04-21:
Hi, glad I could help a bit =)