G6EJD/ESP32-e-Paper-Weather-Display #112: multiple library error

AhsanChuadhry, 2020-07-12:
Arduino: 1.8.10 (Windows 10), Board: "ESP32 Dev Module, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 115200, None"

In file included from C:\Users\Ahsan\Documents\Arduino\libraries\DHTLib\dht.h:18:0,

                 from C:\Users\Ahsan\Documents\Arduino\libraries\DHTLib\dht.cpp:30:

C:\Users\Ahsan\Documents\Arduino\libraries\DHTLib\dht.cpp: In member function 'int dht::_readSensor(uint8_t, uint8_t)':

C:\Users\Ahsan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32/Arduino.h:106:91: error: cannot convert 'volatile uint32_t* {aka volatile unsigned int*}' to 'volatile uint8_t* {aka volatile unsigned char*}' in initialization

 #define portInputRegister(port)     ((volatile uint32_t*)((port)?GPIO_IN1_REG:GPIO_IN_REG))

                                                                                           ^

C:\Users\Ahsan\Documents\Arduino\libraries\DHTLib\dht.cpp:116:29: note: in expansion of macro 'portInputRegister'

     volatile uint8_t *PIR = portInputRegister(port);

                             ^

Multiple libraries were found for "FS.h"
 Used: C:\Users\Ahsan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\FS
Multiple libraries were found for "Wire.h"
 Used: C:\Users\Ahsan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\Wire
Multiple libraries were found for "WebServer.h"
 Used: C:\Users\Ahsan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\WebServer
Multiple libraries were found for "WiFi.h"
 Used: C:\Users\Ahsan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\WiFi
 Not used: C:\Program
Multiple libraries were found for "MAX30100_PulseOximeter.h"
 Used: C:\Users\Ahsan\Documents\Arduino\libraries\Arduino-MAX30100-master
Multiple libraries were found for "OneWire.h"
 Used: C:\Users\Ahsan\Documents\Arduino\libraries\OneWire-master
Multiple libraries were found for "DallasTemperature.h"
 Used: C:\Users\Ahsan\Documents\Arduino\libraries\Arduino-Temperature-Control-Library-master
Multiple libraries were found for "dht.h"
 Used: C:\Users\Ahsan\Documents\Arduino\libraries\DHTLib
exit status 1
Error compiling for board ESP32 Dev Module.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

AhsanChuadhry, 2020-07-12:
can anyone help me with this?

AWSW2017, 2020-07-12:
Try to temporarily remove this folder:

C:\Users\Ahsan\Documents\Arduino\libraries\DHTLib

closed 2020-07-20 by G6EJD