G6EJD/ESP32-e-Paper-Weather-Display #113: 2.9inch e-Paper Module (B) and 3 line SPI or 4 line SPI

echelon75, 2020-07-18:
Hi,

I am using a 2.9inch e-Paper Module (B): [https://www.waveshare.com/wiki/2.9inch_e-Paper_Module_(B)](https://www.waveshare.com/wiki/2.9inch_e-Paper_Module_(B))
This is a three-color version

When I try this code it return me a Busy Timeout!:
Icon name: 01n
Icon name: 01n
Icon name: 01d
Busy Timeout!
Busy Timeout!
Busy Timeout!
Busy Timeout!
Entering 1740-secs of sleep time
and nothing move on my screen

1) First question , there are a 0ohm resistor on display . 
By default position of BS resistor is  0 = 4 line SPI
EPD_MOSI = 14 is not use, in this case BS resistor must be 1= 3 lines SPI because EPD_MISO = 19 is not implemented ?

2) do you think this code is compatible 2.9inch e-Paper Module (B) three-color

3) Wiring :
I am using a ESP32 Devkit
static const uint8_t EPD_BUSY = 4;  // GPIO4  to EPD BUSY
static const uint8_t EPD_RST  = 16; // GPIO16 to EPD RST
static const uint8_t EPD_DC   = 17; // GPIO17  to EPD DC
static const uint8_t EPD_CS   = 5;  // GPIO5  to EPD CS
static const uint8_t EPD_SCK  = 18; // GPIO18  to EPD CLK
static const uint8_t EPD_MOSI = 23; // GPIO23  to EPD DIN

[image]

Regards

Richard

G6EJD, 2020-07-18:
Yes the code will work on a 3C display. You have to ensure the display is wired as per your pin definitions but you can’t deviate from the standard SPI wiring for MOSI, SCK, Etc.
You have to select the correct display driver for a colour display, if you look at the GxEPD2 examples you will see an ESP32 example line fir the 2.9” 3C display copy that and replace the current one with that, just one line amended. To get other colours you will need to add new colour options in place of black. Note the 3C displays are very slow typically 12-secs to update

G6EJD, 2020-07-18:
You need the 4-line SPI switch position, are you using a v2 HAT ?

G6EJD, 2020-07-18:
No your not I should have checked, so assuming the wiring is correct just one line to change

echelon75, 2020-07-18:
**You need the 4-line SPI switch position**
OK I solder RS resistor on 0 position (4lines)

 **are you using a v2 HAT ?**
How can know version ? Here is rear of display:
[image]

**No your not I should have checked, so assuming the wiring is correct just one line to change**
What do you mean by "Just one line to change"
In your code wiring is : 
static const uint8_t EPD_BUSY = 4;  // GPIO4  to EPD BUSY
static const uint8_t EPD_RST  = 16; // GPIO16 to EPD RST
static const uint8_t EPD_DC   = 17; // GPIO17  to EPD DC
static const uint8_t EPD_CS   = 5;  // GPIO5  to EPD CS
static const uint8_t EPD_SCK  = 18; // GPIO18  to EPD CLK
static const uint8_t EPD_MOSI = 23; // GPIO23  to EPD DIN
Here my wiring : 
[image]
[image]
[image]

In GxEPD2_WS_ESP32_Driver wiring is :
GxEPD2_3C<GxEPD2_290c, GxEPD2_290c::HEIGHT> display(GxEPD2_290c(/*CS=*/ 5, /*DC=*/ 17, /*RST=*/ 16, /*BUSY=*/ 4));
Here is modified source code : 
**`// Display Library example for SPI e-paper panels from Dalian Good Display and boards from Waveshare.
// Requires HW SPI and Adafruit_GFX. Caution: the e-paper panels require 3.3V supply AND data lines!
//
// Display Library based on Demo Example from Good Display: http://www.e-paper-display.com/download_list/downloadcategoryid=34&isMode=false.html
//
// Author: Jean-Marc Zingg
//
// Version: see library.properties
//
// Library: https://github.com/ZinggJM/GxEPD2

// GxEPD2_WS_ESP32_Driver : GxEPD2_Example variant for Universal e-Paper Raw Panel Driver Board, ESP32 WiFi / Bluetooth Wireless
// https://www.waveshare.com/product/e-paper-esp32-driver-board.htm

// Supporting Arduino Forum Topics:
// Waveshare e-paper displays with SPI: http://forum.arduino.cc/index.php?topic=487007.0
// Good Display ePaper for Arduino: https://forum.arduino.cc/index.php?topic=436411.0

// mapping of Waveshare ESP32 Driver Board
// BUSY -> 25, RST -> 26, DC -> 27, CS-> 15, CLK -> 13, DIN -> 14

// NOTE: this board uses "unusual" SPI pins and requires re-mapping of HW SPI to these pins in SPIClass
//       this example shows how this can be done easily

// base class GxEPD2_GFX can be used to pass references or pointers to the display instance as parameter, uses ~1.2k more code
// enable or disable GxEPD2_GFX base class
#define ENABLE_GxEPD2_GFX 0

// uncomment next line to use class GFX of library GFX_Root instead of Adafruit_GFX
//#include <GFX.h>
// Note: if you use this with ENABLE_GxEPD2_GFX 1:
//       uncomment it in GxEPD2_GFX.h too, or add #include <GFX.h> before any #include <GxEPD2_GFX.h>

#include <GxEPD2_BW.h>
#include <GxEPD2_3C.h>
#include <Fonts/FreeMonoBold9pt7b.h>

#if defined(ESP32)
// ***** for mapping of Waveshare ESP32 Driver Board *****
// use Board "ESP32 Dev Module" to build with Arduino IDE
// select one, can use full buffer size (full HEIGHT)
//GxEPD2_BW<GxEPD2_154, GxEPD2_154::HEIGHT> display(GxEPD2_154(/*CS=*/ 15, /*DC=*/ 27, /*RST=*/ 26, /*BUSY=*/ 25)); // GDEP015OC1 no longer available
//GxEPD2_BW<GxEPD2_154_D67, GxEPD2_154_D67::HEIGHT> display(GxEPD2_154_D67(/*CS=*/ 15, /*DC=*/ 27, /*RST=*/ 26, /*BUSY=*/ 25)); // GDEH0154D67
//GxEPD2_BW<GxEPD2_154_T8, GxEPD2_154_T8::HEIGHT> display(GxEPD2_154_T8(/*CS=*/ 15, /*DC=*/ 27, /*RST=*/ 26, /*BUSY=*/ 25)); // GDEW0154T8 152x152
//GxEPD2_BW<GxEPD2_154_M09, GxEPD2_154_M09::HEIGHT> display(GxEPD2_154_M09(/*CS=*/ 15, /*DC=*/ 27, /*RST=*/ 26, /*BUSY=*/ 25)); // GDEW0154M09 200x200
//GxEPD2_BW<GxEPD2_154_M10, GxEPD2_154_M10::HEIGHT> display(GxEPD2_154_M10(/*CS=*/ 15, /*DC=*/ 27, /*RST=*/ 26, /*BUSY=*/ 25)); // GDEW0154M10 152x152
[... 1027 lines cut ...]
#endif
  if (display.epd2.panel == GxEPD2::GDEW042Z15)
  {
    for (uint16_t i = 0; i < sizeof(bitmap_pairs) / sizeof(bitmap_pair); i++)
    {
      display.firstPage();
      do
      {
        display.fillScreen(GxEPD_WHITE);
        display.drawInvertedBitmap(0, 0, bitmap_pairs[i].black, display.epd2.WIDTH, display.epd2.HEIGHT, GxEPD_BLACK);
        display.drawInvertedBitmap(0, 0, bitmap_pairs[i].red, display.epd2.WIDTH, display.epd2.HEIGHT, GxEPD_RED);
      }
      while (display.nextPage());
      delay(2000);
    }
  }
}
#endif**`
But display doesn't display anything.
I order a new monochrome display in case of mine is dead . I will receive it on 23/07
I keep you in touch...

Regards

Richard

G6EJD, 2020-07-20:
Hello Richard, your pin setup and wiring looks OK, as does the 3C display object setting, so the code should be working. I can see it is receiving weather data OK, so it's down to the display.

Have you tried any of the GxEPD2 3C examples, they should all work as I use the default wiring in my code too.

You do-not need to change the display to a 4-line SPI, mine all work OK on 3-line setting.

This is a display issue (hardware) or wiring, be sure those du-pont cables are actually making contact end-to-end they are notoriously unreliable for connections.

closed 2020-07-20 by G6EJD