G6EJD/ESP32-e-Paper-Weather-Display #115: GxEPD2_750c_Z08 epd not worked because out of RAM

esonhon, 2020-09-13:
I think we should use GxEPD2's paged drawing feature to reduce the RAM size (or Heap size?)
Now, if I defined my screen with
GxEPD2_3C < GxEPD2_750c_Z08, GxEPD2_750c_Z08::HEIGHT> display(GxEPD2_750c_Z08(/*CS=*/ EPD_CS, /*DC=*/ EPD_DC, /*RST=*/ EPD_RST, /*BUSY=*/ EPD_BUSY)); // GDEW075Z08 800x480

Got the link error:
/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\esp32dev\firmware.elf section `.dram0.bss' will not fit in region `dram0_0_seg'
/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: DRAM segment data does not fit.
/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: region `dram0_0_seg' overflowed by 17464 bytes

G6EJD, 2020-09-13:
Well you could propose a solution... I doubt the savings will amount to much(~17K)

clusterice, 2020-10-30:
Hi esonhon

This is not an issue. You have two possibilities to fix this:
- Use external RAM
- Or do this small channges:

Change the constructor to:
```cpp
GxEPD2_3C<GxEPD2_750c_Z08, GxEPD2_750c_Z08::HEIGHT / 2> display(GxEPD2_750c_Z08(/*CS=*/ EPD_CS, /*DC=*/ EPD_DC, /*RST=*/ EPD_RST, /*BUSY=*/ EPD_BUSY));
```

Witch means you only use the half buffer at one time and change the drawing routine to pagewise drawing:
```cpp
      if (RxWeather && RxForecast) { // Only if received both Weather or Forecast proceed
        StopWiFi(); // Reduces power consumption.
        display.firstPage();
        do {
        DisplayWeather();
        } while (display.nextPage()); // Full screen update mode
      }
```

Clusterice

G6EJD, 2020-11-15:
Thank you @clusterice for suggesting this solution.

closed 2020-11-15 by G6EJD