G6EJD/ESP32-e-Paper-Weather-Display #116: This is not exactly a problem! Cyrillic alignment does not work!

baivas, 2020-09-13:
Thanks for the idea and the excellent realization.
I allowed myself to make it look like this in Cyrillic!
[https://drive.google.com/file/d/1XGxCTo-OJcLdnXo4vVQTQbeBcgCxhcIL/view?usp=sharing](url)

This happened to me!

Greetings!

G6EJD, 2020-09-13:
Looks good, we’ll done.

closed 2020-09-13 by G6EJD

baivas, 2020-09-14:
I made him choose from two different networks

baivas, 2020-09-14:
but thanks for the perfect code

baivas, 2020-09-14:
there remains the problem of alignment in different fonts

G6EJD, 2020-09-14:
You could edit your character set to make it uniform spacing, then it will be aligned in columns.

baivas, 2020-09-14:
https://github.com/olikraus/u8g2/wiki/fntlistall#8-pixel-height
i use these fonts

I was looking for a font editor but there is only a Mac I found

G6EJD, 2020-09-14:
You should be able to find a font editor there are plenty around

baivas, 2020-09-14:
you use - u8g2_font_helvB08_tf
i use this one - u8g2_font_6x13B_t_cyrillic

what is the difference

t | Transparent font,
f | The font includes up to 256 glyphs.

shouldn't the program calculate the difference between the two