G6EJD/ESP32-e-Paper-Weather-Display #118: Waveshare_4_3

ojaksch, 2020-10-19:
I'd love to see this working on a Waveshare_4_3 with a resoultuion of 800x600 pixels.

G6EJD, 2020-10-19:
To be honest I’ve long since given up on that display as it was so unreliable and used a (relative) lot of power in sleep mode. The concept is good but they (WS) failed to release a firmware update that addressed the issues with the display driver.
There would be a lot of work to make the conversion and done of it won’t translate, sorry.

ojaksch, 2020-10-19:
Okay, no problem at all. Good to know that's a unreliable device, at least for this project. Will gather a 4.2 and retry.
Thanks for listenig.

closed 2020-10-19 by ojaksch

G6EJD, 2020-10-19:
4.2” is amongst the best, it takes just 2-secs to refresh and of course takes no power in sleep mode.