G6EJD/ESP32-e-Paper-Weather-Display #119: Blue LED of Lolin D32 stays on after update of 4.2” display

AWSW2017, 2020-11-07:
Hello,

I just noticed that the blue LED of the Lolin D32 stays on after update of 4.2” display sketch.

I could not see this behaviour with the usually from me used 7.5” displays also using Lolin D32 boards.

I already played around with the build-in-led section without success.

Maybe you have a hint for me to save some more power of the battery by turning the led off while the ESP32 is in deep sleep mode. 

Thanks in advance. 😊

G6EJD, 2020-11-07:
On some boards the LED pin is used by the SPI bus usually CS and after the screen is used the CS line is low leading to the LED being on. In the 7.5” version there is this code to switch it off:
#ifdef BUILTIN_LED
  pinMode(BUILTIN_LED, INPUT); // If it's On, turn it off and some boards use GPIO-5 for SPI-SS, which remains low after screen use
  digitalWrite(BUILTIN_LED, HIGH);
#endif

But if the correct board is not selected or a different one then the LED pin is not switched OFF (on usually!) it’s a question of finding out what pin your board uses for the LED and then whether it needs to be high or low to turn it off. Then add a line to set the pin accordingly just before sleep.
This works but has always confused me as when the ESP32 enters sleep it releases all the GPIO pins to a high impedance state, well that’s what the data sheet says, but in reality it doesn’t and they continue to operate. I believe it is connected with setting the pins to sleep mode too which is possible when the NVRAM is still active.

closed 2020-11-07 by G6EJD

AWSW2017, 2020-11-08:
Thanks 😊 You are right. I just took 2 new Lolin D32 boards and they do not show this behaviour. Strange, because i used the same wiring and the came in the same delivery... Thanks again and stay safe. 😊

G6EJD, 2020-11-08:
There the LED and non-LED boards they supply, I've never really found out what the difference is, but this could be it.
Best regards and yes do stay safe