G6EJD/ESP32-e-Paper-Weather-Display #120: [FYI] 5 months on single charge

marekorok, 2020-11-16:
Hi, seems no better place so would like to share here my results with the project.

Elements: LolinD32, Waveshare 2.7" 3C (although used as 2C, refreshing was annoying), some random Li-Ion battery 2330mAh

I have added mqtt reporting for battery voltage and percentage to get nice graphs (mqtt -> Home Assistant -> Influxdb -> Grafana), also calibrated battery factors with a multimeter to get more accurate measurements and finally 3D printed case.

Refresh period 20 minutes with 7 hours sleep time.
Whole set was working straight 5 months and few days on single battery charge.

[image]

[image]

Thank you  @G6EJD for this great project!

AWSW2017, 2020-11-16:
Hi! 
Interesting to hear some values to that. I’m using a 7.5” ePaper with a 5000mAh battery and charged it on 1st of January 2020. I am refreshing the data every 30 minutes between 07:00 and 23:30 o’clock. At 23:30 it takes a longer deep sleep timeout until 07:00 the next day. The battery actually has 61% of charge after 11 and 1/2 months now. 
I also thought of sending the data via MQTT to a server but this might drain the battery to much I think, but would be a nice try. 😁
Kind regards and from me also thanks again for this nice project.

G6EJD, 2020-11-16:
Thank you both for sharing your experiences, your results are very similar to mine, some days I have to check that the battery really is doing well after many months of operation, there’s no doubt that sleeping over night makes a huge difference to battery life. I’ve also switch off all serial printing and gone for an event actuated start up of WiFi to save as much time awake as possible.

Yes colour displays are just too slow. The 7.5” version does well as it’s relatively quick.

Great to see the pictures too.

closed 2020-11-16 by G6EJD

ElwoodSC, 2021-07-05:
How on earth are you able to reach such a long runtime?! :-)
I'm failling miserably with my setup, using a Wemos Lolin D32, a Waveshare 4.2 ePaper display and a 3000mAh LiPo battery. The longest time I was able to make was ~3 days with wakeup every 30 Minutes and sleeping from 23:00 until 6:00. Baiscally I use the 4.2 example with my own redentials. 
Is there anything I'm missing? 
I would be highly interested in @marekorok 's code with MQTT reporting as I failed to implement this by myself...
Sorry for digging up that old issue but this here seems to be my best chance to finally also reach some better battery runtime... :)

G6EJD, 2021-07-05:
Suspect a battery that is not actually 3000mAh but more likely 300mAh, it's a typical Chinese trick to add a 0.
The 4.2" display is about the best you can get for speed and power efficiency so you should be getting really long battery life.
First thing to check is the serial port and see how long it is awake for?

ElwoodSC, 2021-07-05:
The batteries in deed charging aroun ~2500 mAh according to my USB power-meter. They might not have 3000mAh but certainly more than just 300. 
I just connected to my serial monitor and rebooted the D32, that is the result:

_PowerOn : 39001
_Update_Full : 1548001
_PowerOff : 20001
_PowerOff : 1
Entering 118-secs of sleep time
Awake for : 3.923-secs
Starting deep-sleep period...

Looks reasonable to me.

G6EJD, 2021-07-05:
Yes all looks correct. 
I've just used my mAh calculator and with that size battery it shows a de-rated value of 2450mAhr, your getting 2500, so that's OK.
With a 30-min update rate it predicts battery life of 261-days assuming 145mA awake current and 0.07uA sleep current, a value to be expected.
If I change the sleep current to 44mA to simulate an ESP32 that is not sleeping and I get 2.6-days, so that is the problem. For some reason the code is not entering sleep mode, has the code been changed in anyway?

ElwoodSC, 2021-07-05:
Nope, I just flashed the D32 with the 4.2" example code and did only edit the owm-crentials.h with my data.

G6EJD, 2021-07-05:
Do you have any means to monitor power drawn via the battery terminal? DVM?

ElwoodSC, 2021-07-05:
Unfortunately not, I can just use my multimeter in serial with the battery... this hopefully gives at least an indication whether deepsleep is reached or not...
BTW, I just recognized that I compiled with ESP32 1.04 - I'm currently updating and will recompile and flash with the 1.06 environment. 
But beforehand I will measure power consumtion to have a comparison.

ElwoodSC, 2021-07-12:
Power consumption & battery life is much better now - unfortunately now the ESP32 dies around every 1,5 days. When I reboot the ESP32 it'll run for another 1,5 days until again it sleeps forever. But at least the battery does not drain as fast anymore.
Still, it is plain the code from the 4.2 example with no additions at all... Ah, and it is still compiled with ESP32 1.04, I will try to compile with 1.06 next.

G6EJD, 2021-07-12:
You should if you can monitor the battery voltage with a DVM to confirm it is not battery related. It is possible you have a faulty ESP32 or e-paper module as with the 4.2” variant you should be getting battery life of much greater than 1-year. I have 3 x 4.2” Displays running and many ESP32 devices in sleep mode and they have never failed. 
One thing you should check is your WiFi service as this may be creating the problem.

dreamy1, 2023-02-27:
...I’m using a 7.5” ePaper with a Lolin D32 too...

Just a question: what kind of driver board for the 7.5" e-paper do you use?

Thanks in advance,
Stefan

G6EJD, 2023-02-27:
The one provided by Waveshare. Some early variants caused the display to fail, so they added a buffer on later variants.

dreamy1, 2023-02-27:
Addendum: the board DESPI-C02 seems to be the best solution so far, but it is very difficult to get here in germany:

https://de.aliexpress.com/item/1005003600416933.html

dreamy1, 2023-02-27:
Sure, you are right - the wake-up-time is the most significant part. I will do some tests when the hardware is here, maybe there is some kind of optimization with varying the transmit level of the ESP32 and using an external antenna (with "free" sight to the router, the internal antenna onto the pcb is not ideal when using the lolin behind the e-paper because of shielding).

My goal is to get the best out of the hardware with minimizing standby current as a start point. 

I will also implement a hardware switch to manually activate wifi and start a webserver to place new firmware OTA. This would be a great goodie to load new firmware into the station...