G6EJD/ESP32-e-Paper-Weather-Display #123: DRAM segment data does not fit

AndreasMainz, 2020-12-01:
Hello,

i setup your sketch with 800x480 for 3 color display with an Esp32. 
It looks like the RAM runs out of limits with 
`dram0_0_seg' overflowed by 13144 bytes
Any idea what to change in setup?
The display already worked with example sketch without problems.
I checked also several modules like WROVER chip, but still the same issue.
Maybe any special library must be used?

Many thanks for any help..

G6EJD, 2020-12-01:
This is a problem with the display and 3 colour as the GxEPD2 driver builds an image in ESP32 memory for transfer to the display  buffer, so the solution is to use a half-size buffer and modify the screen update lines to keep refreshing the data until all has been transferred. There is a hint in the GxEPD2 examples for the half buffer example in essence the change required is the half size buffer display driver object if you go down the list of options you will find the half buffer version, then instead of display.update there is another method of a while loop I can’t remember the exact details but you’ll find the method in the examples for the library, I’ll try and provide an update soon. Although I haven’t checked recently have you looked through the display options for the larger screens someone may have already done it. About 3 lines of code to change no more than that.

G6EJD, 2020-12-01:
See here: https://github.com/G6EJD/ESP32-e-Paper-Weather-Display/issues/115

G6EJD, 2020-12-02:
Did you get it to work?

AndreasMainz, 2020-12-03:
Yes you can close the issue. I didnt see the already closed issue115 about the same issue, sorry.
But i decided to go to a python solution since the esp32 is at its limits for the 7in5 display.
Maybe if you have time you could port your project to python ;-) make it run on raspbery pi..
Many thanks for your great project.

closed 2020-12-03 by AndreasMainz

G6EJD, 2020-12-03:
Ok good to hear you have a solution now.