G6EJD/ESP32-e-Paper-Weather-Display #124: HTTPClient Version

m1geo, 2020-12-03:
Which version of HTTPClient are you expecting to use?

> C:\Users\geosma01\OneDrive` - ARM\Documents\Arduino\libraries\ESP32-e-Paper-Weather-Display\src/common.h:6:24: fatal error: HTTPClient.h: No such file or directory

https://github.com/G6EJD/ESP32-e-Paper-Weather-Display/blob/c9903cad67ca45b5ba24c101c27a378885bb9b96/src/common.h#L6

I have tried the standard [ArduinoHttpClient](https://github.com/arduino-libraries/ArduinoHttpClient) and [Adrian McEwen's HttpClient](https://github.com/amcewen/HttpClient).

I didn't see any info the in readme files. Thanks for a fantastic project!

Thanks in advance,
George M1GEO

G6EJD, 2020-12-03:
Hi George, I just tried to compile with <HTTPClient.h> as listed, its OK, Did you select the ESP board type and have you installed the ESP32 extension in your IDE?

m1geo, 2020-12-03:
Hi David.I took a clean Arduino 1.8.5 (and the latest, too), and used ESP core 2.4.1 (and the latest, again).Selected NodeMCU 1.0 ESP12E (using a Waveshare epaper driver board, NodeMCU was what their docs say to use). The waveshare example code builds/works fine.I hit every issue trying to get it built! It complains about HTTPClient.h (using ESP8266HTTPClient.h stops errors).It complains about SPI.begin() not taking arguments, and a million things else. None of the deep sleep stuff appears to be present. I'm sure it's me, but I can't find what I've missed!ThanksGeorge. M1GEO.

G6EJD, 2020-12-03:
Hi George, it’s because the code is for an ESP32 but you’ve already made most of the changes with the ESP8266 prefix for httpclient and yes sleep is a different instruction, plus you can only sleep for about 71 minutes maximum.
The SPI.begin with parameters only works with the ESP32 libraries because the pins are fixed on the ESP8266, the ESP32 can use nearly any pin and so allows the SPI statement to make changes you can find the sleep command for the ESP8266 easily. All else’s will compile ok.
If you have any issues let me know.

m1geo, 2020-12-03:
Understood. I hadn't actually noticed ESP32 vs ESP8266. Should have. 

I saw the Internet awash with folks using the same hardware as me, and figured it'd be straight forward. 

I'll make a fork and tweak my changes. I'd like to add a little area for news headlines, say from the BBC RSS or something. 

Thanks
73, George M1GEO. 

⁣-- 
Please excuse my brevity; this was sent from my mobile phone.​

closed 2020-12-04 by m1geo

G6EJD, 2020-12-04:
Good idea, if you do make a new version I’d be interested to see it, 73 Dave