G6EJD/ESP32-e-Paper-Weather-Display #128: 4.2" example has wrong data

danradom, 2020-12-11:
The Humidity / Precipitation / Rain / Snow seems in the 3-day forecast values to not work.  It's been snowing all day here, and it still shows nothing.  I've gone so far as to search out showing or raining locations and test, but it all seems way off.

DrawGraph(SCREEN_WIDTH / 400 * 158, SCREEN_HEIGHT / 300 * 221, SCREEN_WIDTH / 4, SCREEN_HEIGHT / 5, 10, 30, TXT_RAINFALL_IN, rain_readings, max_readings, autoscale_on, barchart_on);
DrawGraph(SCREEN_WIDTH / 400 * 288, SCREEN_HEIGHT / 300 * 221, SCREEN_WIDTH / 4, SCREEN_HEIGHT / 5, 0, 30, TXT_SNOWFALL_IN, snow_readings, max_readings, autoscale_on, barchart_on);

G6EJD, 2020-12-12:
What does the serial port data show, it prints out all the data received?

G6EJD, 2020-12-12:
To test this for you I'd need your location. As far as I'm aware there are no known issues for the software. Sometimes OWM don't report actuality. Sometimes OWM servers are off-line for forecast data too.

danradom, 2020-12-12:
I slightly modified the code to use city IDs instead of names as we have multiple cities with the same name.

String City             = "Windsor, CO";
String CityID.         = "5583509";

# common.h URL change
String uri = "/data/2.5/" + RequestType + "?id=" + CityID + "&APPID=" + apikey + "&mode=json&units=" + units + "&lang=" + Language;

Receiving Forecast period - 
Period-0--------------
DTim: 1607796000
Temp: 22.53
TLow: 22.53
THig: 24.19
Pres: 1015.00
Humi: 93.00
For0: Clouds

Icon: 04d
Desc: overcast clouds
CCov: 94
WSpd: 2.77
WDir: 61.00
Rain: 0.00
Snow: 0.00
Pop:  0.00
Peri: 2020-12-12 18:00:00

Period-1--------------
DTim: 1607806800
Temp: 26.20
TLow: 26.20
THig: 27.66
Pres: 1014.00
Humi: 93.00
For0: Clouds

Icon: 04d
Desc: overcast clouds
CCov: 98
WSpd: 4.43
WDir: 30.00
Rain: 0.00
Snow: 0.00
Pop:  0.10
Peri: 2020-12-12 21:00:00

Period-2--------------
DTim: 1607817600
Temp: 25.07
TLow: 25.07
THig: 25.36
Pres: 1018.00
Humi: 98.00
For0: Snow

Icon: 13n
Desc: light snow
CCov: 99
WSpd: 6.49
WDir: 356.00
Rain: 0.00
Snow: 0.41
Pop:  0.43
Peri: 2020-12-13 00:00:00

G6EJD, 2020-12-12:
There is snow in period 2 as a forecast but without reviewing the code in detail that maybe is being skipped for 3-hour periods for the next 48-hours, the only reflects what’s broadcasted. I find one does not reflect current conditions.

danradom, 2020-12-12:
This is the three day forecast area not the three hour forecast

G6EJD, 2020-12-12:
Ok, so is there still an issue?

danradom, 2020-12-12:
Well the three day forecast doesn't show any real precipitation or humidity readings still.

G6EJD, 2020-12-12:
The serial port shows the received data issued, probably all zero for snow unless forecasted.

danradom, 2020-12-12:
OK I'll continue to mess with it and change locations so I better understand what its doing.  Thanks.

Very nice work!  I really appreciate it.

G6EJD, 2020-12-12:
You’ll find all of the weather data is let’s call it odd. Sometimes the current weather and forecast data are not contiguous which I often find just weird, it’s probably machine generated. For example temperatures now can suddenly jump in a 3-hour forecast which can’t be, is there still an issue here?

danradom, 2020-12-12:
Nope.  All good.  Thanks for your time.

closed 2020-12-12 by danradom